fastmachinelearning/hls4ml #438: HLS4ML Vivado Synthesis report not found. Co-simulation report not found.

Pythagoras21, 2021-11-04:
HI!
I ran hls_model.build() and hls4ml.report.read_vivado_report('my-hls-test')
However after 5-10 mins, i got the output as  Synthesis report not found. Co-simulation report not found.

- I ran this on Jupyter notebook
- i use Vivado 2019.2
- My OS is Ubuntu 20.04.3 LTS
- Python version- Python 3.8.10
===================================================================================
## What i did:
===================================================================================
```
import hls4ml

#Fetch a keras model from our example repository
#This will download our example model to your working directory and return an example configuration file
config = hls4ml.utils.fetch_example_model('KERAS_3layer.json')

print(config) #You can print the configuration to see some default parameters

#Convert it to a hls project
hls_model = hls4ml.converters.keras_to_hls(config)

# Print full list of example models if you want to explore more
hls4ml.utils.fetch_example_list()
```
Downloading example model files ...
{'OutputDir': 'my-hls-test', 'ProjectName': 'myproject', 'XilinxPart': 'xcku115-flvb2104-2-i', 'ClockPeriod': 5, 'Backend': 'Vivado', 'IOType': 'io_parallel', 'HLSConfig': {'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': '1'}}, 'KerasJson': 'KERAS_3layer.json', 'KerasH5': 'KERAS_3layer_weights.h5'}
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 16]]
Layer name: fc1_relu, layer type: Dense, current shape: [[None, 16]]
Layer name: fc2_relu, layer type: Dense, current shape: [[None, 64]]
Layer name: fc3_relu, layer type: Dense, current shape: [[None, 32]]
Layer name: output_softmax, layer type: Dense, current shape: [[None, 32]]
Creating HLS model
{   'keras': [   'KERAS_3layer.json',
                 'garnet_1layer.json',
                 'KERAS_dense_16x500x500x500x500x500x5.json',
                 'KERAS_conv1d_small.json',
                 'qkeras_mnist_cnn.json',
                 'KERAS_1layer.json',
                 'KERAS_3layer_binarydense_relu_max.json',
                 'KERAS_3layer_binary_smaller.json',
                 'KERAS_conv1d.json',
                 'KERAS_conv2d_model.json',
                 'jetTagger_Conv2D_Small_NoBatchNorm.json',
                 'keras_bnn.json',
                 'KERAS_3layer_ternary_small.json',
                 'garnet_3layer.json',
                 'KERAS_dense_16x100x100x100x100x100x5.json',
                 'KERAS_dense_16x200x200x200x200x200x5.json',
                 'qkeras_mnist_dense.json',
                 'qkeras_3layer.json',
                 'jetTagger_Conv2D_Small.json',
                 'KERAS_3layer_batch_norm.json'],
    'onnx': [   'conv1d_small_keras.onnx',
                'three_layer_bn_pytorch.onnx',
                'three_layer_bn_keras.onnx',
                'dense_big_keras.onnx',
                'three_layer_pytorch.onnx',
                'two_layer_pytorch.onnx',
                'conv2d_small_mp_keras.onnx',
                'two_layer_keras.onnx',
                'three_layer_keras.onnx',
                'conv2d_small_keras.onnx'],
    'pytorch': ['two_layer_model.pt', 'three_layer_model.pt'],
    'tensorflow': ['3layer.pb']}

`#Use Vivado HLS to synthesize the model
[... 181 lines cut ...]
===================================================================================
Here, we can see that "Synthesis report not found."
After this, if i run: `hls4ml.report.read_vivado_report('my-hls-test')`
i get
 
Found 1 solution(s) in my-hls-test/myproject_prj.
Reports for solution "solution1":

C SIMULATION RESULT:
INFO: [SIM 2] *************** CSIM start ***************
INFO: [SIM 4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
INFO: Unable to open input/predictions file, using default input.
0.0292969 0.756836 0.0546875 0.139648 0.0371094 
INFO: Saved inference results to file: tb_data/csim_results.log
INFO: [SIM 1] CSim done with 0 errors.
INFO: [SIM 3] *************** CSIM finish ***************

Synthesis report not found.
Co-simulation report not found.
=====================================================================================

Please help me resolve this issue

thesps, 2021-11-04:
`/home/gegerin/Vivado_HLS/Vivado/2019.2/bin/rdiArgs.sh: line 280: 2555 Killed "$RDI_PROG" "$@"`
This error usually means Vivado HLS has used all of the available memory and died. The memory requirement depends on the model size, but something in the 16 GB - 64 GB range is usually advised.

thesps, 2021-11-04:
Your config looks like this:

```
{'OutputDir': 'my-hls-test', 
'ProjectName': 'myproject', '
XilinxPart': 'xcku115-flvb2104-2-i', 
'ClockPeriod': 5, 
'Backend': 'Vivado', 
'IOType': 'io_parallel', 
'HLSConfig': {'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': '1'}}, 
'KerasJson': 'KERAS_3layer.json', 
'KerasH5': 'KERAS_3layer_weights.h5'}
```

I'd suggest updating it to:

```
{'OutputDir': 'my-hls-test', 
'ProjectName': 'myproject', 
'XilinxPart': 'xcku115-flvb2104-2-i', 
'ClockPeriod': 5, 
'Backend': 'Vivado', 
'IOType': 'io_parallel', 
'HLSConfig': {'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': '64', 'Strategy':'Resource'}}, 
'KerasJson': 'KERAS_3layer.json', 
'KerasH5': 'KERAS_3layer_weights.h5'}
```

The documentation on configuration is [here](https://fastmachinelearning.org/hls4ml/api/configuration.html).

thesps, 2021-11-04:
Also, could you please close any issues that are now resolved, thanks 🙏

Pythagoras21, 2021-11-07:
I tried your suggestion and also replaced IOtype tonio_serial but still it returned "$RDI_PROG" "$@".
I am running this on 4GB of RAM.

How can I tackle this problem

Is there any way to make vivado consume less RAM or avail free service to run vivado on a server .

jmduarte, 2021-11-08:
AWS has F1 servers that allow you to run Vivado, and should have enough RAM, but it's not free.

In general, it seems 8 GB is better, and maybe even 16+ GB is needed for UltraScale(+) devices (like the Kintex UltraScale 115 that you're targeting). Maybe it'll work better if you target a smaller device like a PYNQ-Z2 (`xc7z020clg400-1`), but even for this 4 GB may not be enough. Ultimately, you may just need access to a computer with more RAM.

https://www.reddit.com/r/FPGA/comments/ne8al6/comment/gyf7abu/?utm_source=share&utm_medium=web2x&context=3

Pythagoras21, 2021-11-16:
Update: Synthesis achieved with RAM upgrade to 8GB

closed 2021-11-16 by Pythagoras21