fastmachinelearning/hls4ml #442: Can't synthesis Qkeras CNN model

HenningCode, 2021-11-05:
Hello Guys,
I successfully synthesized some FC neural networks, now I tried CNN's, but I cant seem to make them work. I used the master branch and also the branch form PR #220. 
I tried both `io_stream` as well as `io_parallel`, with both `Resource` and `Latency`. 
Do you guys have an idea why it doesn't work? 

The synthesis either fails with being stuck at the unrolling process being stuck or with this error log:

```
source /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'henning' on host 'henningubuntu' (Linux_x86_64 version 5.4.0-89-generic) on Fri Nov 05 17:11:21 CET 2021
INFO: [HLS 200-10] On os Ubuntu 18.04.6 LTS
INFO: [HLS 200-10] In directory '/home/henning/Schreibtisch/Masterarbeit/Hls4Ml/Output/my_model_qkeras'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/home/henning/Schreibtisch/Masterarbeit/Hls4Ml/Output/my_model_qkeras/mnist_classifier_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/mnist_classifier.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'mnist_classifier_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/henning/Schreibtisch/Masterarbeit/Hls4Ml/Output/my_model_qkeras/mnist_classifier_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020-clg400-1'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../mnist_classifier_test.cpp in debug mode
   Compiling ../../../../firmware/mnist_classifier.cpp in debug mode
   Generating csim.exe
INFO: Unable to open input/predictions file, using default input.
WARNING: Hls::stream 'layer11_cpy1' contains leftover data, which may result in RTL simulation hanging.
0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 0.0996094 
INFO: Saved inference results to file: tb_data/csim_results.log
INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
***** C SIMULATION COMPLETED IN 0h0m6s *****
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/mnist_classifier.cpp' ... 
WARNING: [HLS 200-40] In file included from firmware/mnist_classifier.cpp:1:
In file included from firmware/mnist_classifier.cpp:22:
In file included from firmware/parameters.h:12:
In file included from firmware/nnet_utils/nnet_conv2d_stream.h:5:
firmware/nnet_utils/nnet_conv_stream.h:13:10: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
 if (idx < K - S) {
     ~~~ ^ ~~~~~
firmware/nnet_utils/nnet_conv_stream.h:58:16: note: in instantiation of function template specialization 'nnet::scale_index_K_gte_S<2, 2, 28>' requested here
        return scale_index_K_gte_S<K, S, W>(idx);
               ^
firmware/nnet_utils/nnet_conv2d_stream.h:16:29: note: in instantiation of function template specialization 'nnet::scale_index<2, 2, 28>' requested here
    const unsigned sh_idx = scale_index<CONFIG_T::filt_height, CONFIG_T::stride_height, CONFIG_T::in_height>(h_idx);
                            ^
firmware/nnet_utils/nnet_conv2d_stream.h:58:13: note: in instantiation of function template specialization 'nnet::compute_scaled_indices_2d<nnet::array<ap_fixed<16, 6, 5, 3, 0>, 1>, config2>' requested here
            compute_scaled_indices_2d<data_T, CONFIG_T>(i_ih, i_iw, pixel_idx);
            ^
firmware/mnist_classifier.cpp:69:2: note: in instantiation of function template specialization 'nnet::conv_2d_cl<nnet::array<ap_fixed<16, 6, 5, 3, 0>, 1>, nnet::array<ap_fixed<16, 6, 5, 3, 0>, 16>, config2>' requested here
 nnet::conv_2d_cl<layer11_t, layer2_t, config2>(layer11_cpy2, layer2_out, w2, b2);
 ^
In file included from firmware/mnist_classifier.cpp:1:
In file included from firmware/mnist_classifier.cpp:22:
In file included from firmware/parameters.h:12:
In file included from firmware/nnet_utils/nnet_conv2d_stream.h:5:
firmware/nnet_utils/nnet_conv_stream.h:35:10: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
 if (idx < S - K) {
     ~~~ ^ ~~~~~
firmware/nnet_utils/nnet_conv_stream.h:60:16: note: in instantiation of function template specialization 'nnet::scale_index_K_lt_S<2, 2, 28>' requested here
        return scale_index_K_lt_S<K, S, W>(idx);
[... 258 lines cut ...]
      Trace: true
    dense:
      Precision:
        bias: ap_fixed<5,1>
        weight: ap_fixed<5,1>
      ReuseFactor: 1000000
      Trace: true
    input:
      Precision:
        result: ap_fixed<16,6>
      Trace: true
    softmax:
      Precision: ap_fixed<16,6>
      ReuseFactor: 1000000
      exp_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
      inv_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
      table_size: 1024
      Trace: true
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 1000000
    Strategy: Resource
```
This is my Model. 
[image]

jmduarte, 2021-11-08:
Can you send us a script to reproduce your model and error?

HenningCode, 2021-11-08:
This Code is used to convert the model:
```
print('-------------------------------------------------')
print('INFO: Starting Converter')
print('-------------------------------------------------')

QKERAS = 1
LOAD_CONFIG = 1
BUILDING = 1
CONVNET = 1

model_dir = 'model/qkeras_mymodel_smallconv.h5'

if QKERAS:
    co = {}
    utils._add_supported_quantized_objects(co)
    model = utils.load_qmodel(model_dir,custom_objects=co)

model.summary()

print('-------------------------------------------------')
print('INFO: Starting Conversion of Config')
print('-------------------------------------------------')

if LOAD_CONFIG:
    args = command_line_chk()
    param = yaml_load(args.config)
    config = param
    config['KerasModel'] = model
    hls_model = hls4ml.converters.keras_to_hls(config)
else:
    config = hls4ml.utils.config_from_keras_model(model,granularity='name')
    yaml_save(config=config,filepath='qkeras_mymode_conv_config.yml')
    print_dict(config)
    hls_model = hls4ml.converters.convert_from_keras_model(model,
                                            hls_config=config,
                                            output_dir='../output/model_std/hls4ml_prj',
                                            fpga_part='xc7z020clg400-1')

_ = hls_model.compile()

if BUILDING:
    print('-------------------------------------------------')
    print('INFO: Starting Building')
    print('-------------------------------------------------')
    hls_model.build(validation=True,export=True)
    hls4ml.report.read_vivado_report('../Output/my_model_qkeras')
```
This Code is used to create the Model: 
```
def count_errors(x, y):
    error = 0
    for i in range(len(x)):
        if(not np.array_equal(x[i], y[i])):
            error += 1

    return error

def yaml_save(config, filepath):
    with open(filepath, 'w') as outfile:
        yaml.dump(config, outfile, default_flow_style=False)

print("============================================================================\n\n\n")

model_dir = 'model/qkeras_mymodel_smallconv.h5'
config_dir ='config/qkeras_mymodel_smallconv_config.yml'
picture_dir = 'picture/qkeras_mymodel_smallconv.png'

NB_EPOCH = 5
BATCH_SIZE = 64
VERBOSE = 1
NB_CLASSES = 10
OPTIMIZER = Adam(learning_rate=0.0001, decay=0.000025)
VALIDATION_SPLIT = 0.1

SAFE_CONFIG = 1
COVNET = 1

(x_train, y_train), (x_test, y_test) = mnist.load_data()

RESHAPED = 784

x_test_orig = x_test

x_train = x_train.astype("float32")
x_test = x_test.astype("float32")

if COVNET:
    x_train = x_train[..., np.newaxis]
    x_test = x_test[..., np.newaxis]
else:
    x_train = x_train.reshape(x_train.shape[0], RESHAPED)
    x_test = x_test.reshape(x_test.shape[0], RESHAPED)

# Here only tested with, without the accuracy is bad even without QDense
x_train /= 256
x_test /= 256

print('Train shape: ', x_train.shape)
print('Test shape: ', x_test.shape)

y_train = to_categorical(y_train, NB_CLASSES)
y_test = to_categorical(y_test, NB_CLASSES)

if COVNET:
    x = x_in = Input(
        x_train.shape[1:-1] + (1,), name="input")
    x = QConv2D(
        16, (2, 2), strides=(2, 2),
        kernel_quantizer=quantized_bits(4, 0, alpha=1),
        bias_quantizer=quantized_bits(4, 0, alpha=1),
        name="conv2d_0_m")(x)
    x = QActivation("quantized_relu(4,2)", name="act0_m")(x)
    x = QConv2D(
        32, (3, 3), strides=(2, 2),
        kernel_quantizer=quantized_bits(4, 0, alpha=1),
        bias_quantizer=quantized_bits(4, 0, alpha=1),
        name="conv2d_1_m")(x)
    x = QActivation("quantized_relu(4,2)", name="act1_m")(x)
    x = Flatten()(x)
    x = QDense(NB_CLASSES, kernel_quantizer=quantized_bits(4, 0, alpha=1),
               bias_quantizer=quantized_bits(4, 0, 1, alpha=1),
               name="dense")(x)
    x = Activation("softmax", name="softmax")(x)

else:
    x = x_in = Input((RESHAPED,), name="input")
    #x = Dense(64,name="dense0")(x)
    x = QDense(64, kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1), name="dense0")(x)
    x = QBatchNormalization()(x)
    x = QActivation("quantized_relu(4,2)", name="act0")(x)
    x = QDense(NB_CLASSES, kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1), name="dense2")(x)
    x = Activation("softmax", name="softmax")(x)

model = Model(inputs=[x_in], outputs=[x])
model.summary()
model.compile(
    loss="categorical_crossentropy", optimizer=OPTIMIZER, metrics=["accuracy"])

history = model.fit(
    x_train, y_train, batch_size=BATCH_SIZE,
    epochs=NB_EPOCH, initial_epoch=1, verbose=VERBOSE,
    validation_split=VALIDATION_SPLIT)

model.save(model_dir)

config = hls4ml.utils.config_from_keras_model(model, granularity='name')
config['Model']['Strategy'] = 'Resource'

if SAFE_CONFIG:
    yaml_save(config=config, filepath=config_dir)

print_dict(config)
hls_model = hls4ml.converters.convert_from_keras_model(model,
                                                       hls_config=config,
                                                       output_dir='../output/model_std/hls4ml_prj',
                                                       fpga_part='xc7z020clg400-1')
_ = hls_model.compile()

TEST_CASES = 1000

hls4ml.utils.plot_model(hls_model, show_shapes=True,
                        show_precision=True, to_file='pictures/batchdense_model.png')
plt.show()

out_model = model.predict(x_test[0:TEST_CASES])
out_model_change = np.zeros_like(out_model)
out_model_change[np.arange(len(out_model)), out_model.argmax(1)] = 1

out_hls = hls_model.predict(x_test[0:TEST_CASES])
out_hls_change = np.zeros_like(out_hls)
out_hls_change[np.arange(len(out_hls)), out_hls.argmax(1)] = 1

print('Error Normal: ', count_errors(out_model_change, y_test[0:TEST_CASES]))
print('Error HLS: ', count_errors(out_hls_change, y_test[0:TEST_CASES]))
```

HenningCode, 2021-11-12:
I also found out, that I can synthesize the example `qkeras_mnist_cnn` just fine. I then tried to replicate the model for my self, train it and synthesize it. 

I does not work... 
I am on the latest master branch as well as the `version 0.6.0` from pip
This is the script that works just fine for the example model.
```
import hls4ml
from tensorflow.keras.utils import to_categorical
from tensorflow.keras.datasets import mnist
import numpy as np
from sklearn.metrics import accuracy_score
import tensorflow as tf
from qkeras import *
import yaml
import json

def print_dict(d, indent=0):
    align = 20
    for key, value in d.items():
        print('  ' * indent + str(key), end='')
        if isinstance(value, dict):
            print()
            print_dict(value, indent+1)
        else:
            print(':' + ' ' * (20 - len(key) - 2 * indent) + str(value))

def yaml_save(config, filepath):
    with open(filepath, 'w') as outfile:
        yaml.dump(config, outfile, default_flow_style=False)

print('----------------------------------------------------------------')

output_dir = '../Output/example_model'

with open('qkeras_mnist_cnn.json') as f:
  data = json.load(f)

data = json.dumps(data)

co = {}
utils._add_supported_quantized_objects(co)
model = tf.keras.models.model_from_json(data,custom_objects=co)
model.load_weights('qkeras_mnist_cnn_weights.h5')
model.summary()

config = hls4ml.utils.fetch_example_model('qkeras_mnist_cnn.json')
config.pop('KerasJson')
config.pop('KerasH5')
config['XilinxPart'] = 'xc7z020clg400-1'
config['OutputDir'] = output_dir

#config['KerasJson'] = 'qkeras_mnist_cnn.json'
config['HLSConfig']['Model']['Strategy'] = 'Resource'
config['HLSConfig']['Model']['ReuseFactor'] = 1024
config['IOType'] = 'io_stream'
config['Backend'] = 'Vivado'
config['HLSConfig']['LayerName']['q_conv2d']['ReuseFactor'] = 10000
config['HLSConfig']['LayerName']['q_conv2d_1']['ReuseFactor'] = 10000000
config['HLSConfig']['LayerName']['q_dense']['ReuseFactor'] = 100000000
yaml_save(config=config, filepath='config/example_config.yml')
config['KerasModel'] = model
print_dict(config)
hls_model = hls4ml.converters.keras_to_hls(config)
hls4ml.utils.plot_model(hls_model, show_shapes=True,
                        show_precision=True, to_file='pictures/example_model.png')

(x_train, y_train), (x_test, y_test) = mnist.load_data()

TEST_CASES = 1000

[... 173 lines cut ...]
    yaml_save(config=backend_config, filepath=config_dir)

backend_config['KerasModel'] = model

print_dict(backend_config)
hls_model = hls4ml.converters.keras_to_hls(backend_config)
_ = hls_model.compile()

TEST_CASES = 1000

hls4ml.utils.plot_model(hls_model, show_shapes=True,
                        show_precision=True, to_file=picture_dir)

y_model = model.predict(x_test[0:TEST_CASES])
y_hls = hls_model.predict(x_test[0:TEST_CASES])
acc_qkeras = accuracy_score(np.argmax(y_test[0:TEST_CASES], axis=1), np.argmax(y_model, axis=1))
acc_hls4ml = accuracy_score(np.argmax(y_test[0:TEST_CASES], axis=1), np.argmax(y_hls, axis=1))
rel_diff = abs(acc_qkeras - acc_hls4ml) / acc_qkeras
print('\n')
print('Accuracy qkeras:     {}'.format(acc_qkeras))
print('Accuracy hls4ml:     {}'.format(acc_hls4ml))
print('Relative difference: {}'.format(rel_diff))

hls_model.build()
hls4ml.report.read_vivado_report(output_dir)

thesps, 2021-11-12:
I think this is the key difference:
`config['IOType'] = 'io_stream'` in the working one
`backend_config['IOType'] = 'io_parallel'` in the not working one. 

Try switching to `io_stream`

HenningCode, 2021-11-12:
Yes I am trying to target a `pynq-z2`

If I just change to `io_stream` it doenst take ages for the `Unrolling loop` but still fails, with this Error:

`ERROR: [XFORM 203-711] Internal stream 'layer14_cpy1.V.data[0].V' failed dataflow checking: it must be produced and consumed exactly once.`

I am also getting a ton of these warnings after writing the HLS Project is done, but before my predictions.
`WARNING: Hls::stream 'layer14_cpy1' contains leftover data, which may result in RTL simulation hanging.`

If I change to the `VivadoAccelerator` backend it fails with compilation errors, because some template arguments are worng.

```
In file included from ../../../../MyProject_test.cpp:29:0:
../../../../firmware/nnet_utils/nnet_helpers.h:304:6: Anmerkung: candidate: template<class src_T, class dst_T, long unsigned int OFFSET, long unsigned int SIZE> void nnet::copy_data_axi(std::vector<_Tp>, dst_T*)
 void copy_data_axi(std::vector<src_T> src, dst_T dst[SIZE]) {
      ^~~~~~~~~~~~~
../../../../firmware/nnet_utils/nnet_helpers.h:304:6: Anmerkung:   Herleitung/Ersetzung von Templateargument gescheitert:
../../../../MyProject_test.cpp:80:63: Fehler: Templateargument 2 ist ungültig
       nnet::copy_data_axi<float, inputs_t, 0, N_IN>(in, inputs);
                                                                                               ^
```

If I only change `io_stream` and `axi_stream` its the same as with only `io_stream` changed.

HenningCode, 2021-11-15:
I tested some more and this model somehow worked. 

```
def create_QConv2(input_shape,output_shape):  
        model = Sequential()
        model.add(QConv2D(16,kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1), kernel_size=(3, 3),
                        activation='relu',
                        input_shape=input_shape))
        model.add(QConv2D(16, (3, 3),kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1), activation='relu'))
        model.add(MaxPooling2D(pool_size=(2, 2)))
        model.add(Flatten())
        model.add(QDense(128, kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1),activation='relu'))
        model.add(QDense(output_shape, kernel_quantizer=quantized_bits(5, 0, alpha=1),
               bias_quantizer=quantized_bits(5, 0, alpha=1),activation='softmax'))

        model.summary()
        return model
```
So it seems that there are still issues with activation functions?

jmduarte, 2021-12-26:
Hi @HenningCode,

I think the only problem is one of your layers is named `input`, which is a reserved name in `HLSModel`. I've opened a PR to raise a runtime error when this is detected

Here's a version of your script that works (basically just renaming your layer `input1`: https://gist.github.com/jmduarte/c211ca7d8e43b5b4e11328e97fc20dde

PR #482 (2022-05-31): Raise runtime error when a layer is named `input`
- Raise runtime error when a layer is named `input` because it is a reserved layer name in `HLSModel`

Resolves #442 and many other similar situations
closed 2022-01-07 by HenningCode