fastmachinelearning/hls4ml #446: Allowed ReuseFactor per layer

AlexMontgomerie, 2021-11-09:
I'm curious what the set of allowed reuse factors are for a given network? I've been trying out this tool for your simple 4-layer network with only `Dense` nodes and modifying the reuse factor so that at first it's less than `channels_in * channels_out`, and now so that it's a factor of this instead. However when I've run the example below, I do not get what I expect.

| Layer | Channels In | Channels Out | ReuseFactor |
|---------|------------------|--------------------|-------------------|
| ip1     | 16      | 64 | 1024 |
| ip2     | 64 | 32 | 128 |
| ip3 | 32 | 32 | 64 |
| ip4 | 32 | 5 | 16 |

In the above table, I expect that `ip1` is the slowest layer and so should take around 1024 clock cycles. However, it takes 1523 cycles instead. I noticed that I was getting `INFO: [SCHED 204-61] Unable to satisfy pipeline directive: Unable to pipeline the region.` for the first 3 layers, and the last layer had an II of 160.

So I was wondering if there are any constraints on the reuse factor? And whether there's any validation that can be done on the configurations to see if they will generate the expected hardware?

vloncar, 2021-11-09:
There is a function that will give you a list of valid reuse factors for a given layer. See [here](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado_template.py#L424). Note that even if you put some wrong RF, you'll still end up with the valid one closest to it.

What device are you trying this on? We've seen that on some older FPGA families Vivado HLS has scheduling issues like yours, and then gets a much higher II.

AlexMontgomerie, 2021-11-09:
Ah great, I'll have a look at that!

I'm running it for `xcku115-flvb2104-2-i` (just the default device), with vivado hls 2019.1.