fastmachinelearning/hls4ml #452: How to use profiling with master branch

HenningCode, 2021-11-11:
Hello Guys,

I want to debug my HLS Model with profiling. I also want to use the latest master branch, but if I try to call eg. `hls4ml.model.profiling.numerical` I get this error `module 'hls4ml.model' has no attribute 'profiling'` is there a way to use profiling with the latest version ?

Greetings.

vloncar, 2021-11-11:
You need to install hls4ml with `profiling` extra, like `pip install hls4ml[profiling]`. This will install the additional packages required by the profiling tools and `hls4ml.model.profiling` will be available.

HenningCode, 2021-11-11:
Yeah I know, but then I am stuck with `0.5.0` and not with the newest version available right?

vloncar, 2021-11-11:
`pip install git+https://github.com/fastmachinelearning/hls4ml.git@master#egg=hls4ml` :wink:

HenningCode, 2021-11-11:
Thanks! I learned something new today :)

closed 2021-11-11 by HenningCode