fastmachinelearning/hls4ml #458: Resource Modelling

AlexMontgomerie, 2021-11-22:
Hi, I was curious if there is any resource modelling in the repo? I noticed that there is a `get_bram_size(self, layer)` method for a HLS configuration, however I'm not clear as to whether this is for a layer or for the whole bitstream.

https://github.com/fastmachinelearning/hls4ml/blob/0ef2766cc0da5705e7f93906b2e489df56baeba7/hls4ml/model/hls_model.py#L121-L123

vloncar, 2021-11-22:
Currently, `bram_size` (controlled by the `BramFactor` in the config) is the threshold above which weights will be implemented as BRAM ports (basically the `#pragma HLS INTERFACE bram port=weight_name`)

HenningCode, 2021-12-01:
@vloncar could you give me an example how to use the `BramFactor` in the config ?

vloncar, 2021-12-01:
For example:
```
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 42
    BramFactor: 1337
```
Now every layer whose number of weights is larger than 1337 will be implemented as BRAM ports. (The other parameters here don't matter, I just put them so you see at what level are you supposed to put `BramFactor`)

closed 2026-10-23 by github-actions[bot]