fastmachinelearning/hls4ml #463: HLS4ML Model and QKeras Model prediction not equal

constiDisch, 2021-11-30:
Hi,
i am very new to hls4ml and thus was trying to run an adapted [tutorial](https://github.com/fastmachinelearning/hls4ml-tutorial/blob/master/part6_cnns.ipynb). I wanted to compare the output of the qkeras model with the hls4ml model.
The code looks like this: 
```python
from keras.layers import Input, Flatten, MaxPool2D
from qkeras import QConv2D, QActivation, QDense, quantized_bits, quantized_relu
from keras.models import Model, Sequential
import numpy as np
import hls4ml

NB_CLASSES = 10
SHAPE = (16, 16, 3)

qmodel = Sequential()
qmodel.add(
    QActivation(quantized_bits(8, 0, 1, alpha=1))
)
qmodel.add(
    QConv2D(12, (3, 3),
        kernel_quantizer=quantized_bits(8, 0, 1, alpha=1),
        bias_quantizer=quantized_bits(8, 0, 1, alpha=1)
    )
)
qmodel.add(
    QActivation(quantized_relu(8, 0, 1))
)
qmodel.add(
    MaxPool2D()
)
qmodel.add(
    Flatten()
)
qmodel.add(
    QDense(NB_CLASSES,
        kernel_quantizer=quantized_bits(8, 0, 1, alpha=1),
        bias_quantizer=quantized_bits(8, 0, 1, alpha=1)
    )
)
qmodel.add(
    QActivation(quantized_bits(8, 0, 1, alpha=1))
)
qmodel.build(input_shape=(None, ) + SHAPE)

## now we convert the model to hls4ml
hls4ml.model.optimizer.OutputRoundingSaturationMode.layers = ['Activation']
hls4ml.model.optimizer.OutputRoundingSaturationMode.rounding_mode = 'AP_RND'
hls4ml.model.optimizer.OutputRoundingSaturationMode.saturation_mode = 'AP_SAT'

hls_config_q = hls4ml.utils.config_from_keras_model(qmodel, granularity='name', default_precision='ap_fixed<8,1>')
hls_config_q['Model']['ReuseFactor'] = 1
  
cfg_q = hls4ml.converters.create_config(backend='Vivado')
cfg_q['IOType']     = 'io_stream' # Must set this if using CNNs!
cfg_q['HLSConfig']  = hls_config_q
cfg_q['KerasModel'] = qmodel
cfg_q['OutputDir']  = 'quantized_pruned_cnn/'
cfg_q['XilinxPart'] = 'xcu250-figd2104-2L-e'
  
hls_model_q = hls4ml.converters.keras_to_hls(cfg_q)
hls_model_q.compile()

N = 10
for i in range(N):
    example_image = np.random.randn(1, 16,16,3).astype(np.float32)
    hls_model_out = hls_model_q.predict(np.ascontiguousarray(example_image))
    qkeras_model_out =  qmodel.predict(example_image)
    assert np.max(np.abs(hls_model_out - qkeras_model_out)) < 1e-6
```
However, the assertion error is raised, because `hls_model_out` and `qkeras_model_out` are not equal.
I think, I am missing something easy, so any help is appreciated.
Thanks.
`hls4ml` has version `0.6.0`.

HenningCode, 2021-12-01:
Could you share the results of the Outputs ? I would say the boundary of 1e^-6 is a bit low.

constiDisch, 2021-12-01:
Thanks, sure:
```python
print(hls_model_out)
[-0.8671875  0.390625   0.1796875 -0.109375   0.5078125  0.421875
  0.453125   0.0546875  0.375      0.6953125]
print(qkeras_model_out)
[[-0.0703125  0.3125     0.9921875  0.9921875  0.015625   0.8515625
  -0.9921875 -0.9921875 -0.0625     0.9921875]]
```
So the boundary is not the issue here.

constiDisch, 2022-02-21:
I tried to use a modified tests now, but without success. Could anybody give me a hint how to configure the hls4ml model, such that outputs are equal to the qkeras model?
```python
def test_single_dense():
  '''
  Test a single Dense -> Activation layer topology for
  bit exactness with number of bits parameter
  '''
  bits = 8
  X = ((np.random.randint(0, 255, size=(100, 16)) - 128)/128).astype(np.float32)
  model = Sequential()
  model.add(QDense(16, input_shape=(16,), name='fc1',
                  kernel_quantizer=quantized_bits(bits,0,alpha=1), bias_quantizer=quantized_bits(bits,0,alpha=1),
                  kernel_initializer='lecun_uniform'))
  model.add(QActivation(activation=quantized_relu(bits,0), name='relu1'))
  model.compile()
  with tempfile.TemporaryDirectory() as tmp:
    hls4ml.model.optimizer.OutputRoundingSaturationMode.layers = ['fc1', 'fc1_input', 'fc1_linear', 'relu1']
    hls4ml.model.optimizer.OutputRoundingSaturationMode.rounding_mode = 'AP_RND_CONV'
    hls4ml.model.optimizer.OutputRoundingSaturationMode.saturation_mode = 'AP_SAT'
    hls_config = hls4ml.utils.config_from_keras_model(model, granularity='name')
    hls_config['Model']['ReuseFactor'] = 1
    hls_config['Model']['Precision'] = f'ap_fixed<{bits},1>'
    hls_config['Model']['Strategy'] = 'Latency'
    hls_model = hls4ml.converters.convert_from_keras_model(model,
                                                        hls_config=hls_config,
                                                        output_dir=str(tmp),
                                                        part='xcu250-figd2104-2L-e')
    hls4ml.model.optimizer.OutputRoundingSaturationMode.layers = []
    hls_model.compile()
    hls4ml.utils.plot_model(hls_model, show_shapes=True, show_precision=True, to_file="model.png")
    y_qkeras = quantized_bits(bits,0,alpha=1)(model.predict(X)).numpy()
    y_hls4ml = hls_model.predict(np.ascontiguousarray(X))
    np.testing.assert_allclose(y_qkeras.ravel(), y_hls4ml.ravel(), atol=2**-bits, rtol=1.0)
```

closed 2022-02-25 by constiDisch