fastmachinelearning/hls4ml #465: Compilation Termination

rtspk, 2021-12-05:
Hi Guys, 
Some months ago, I worked with hls4ml ( older versions). and everything was fine. recently I moved to new hls4ml verion. 
I setup the environment using  `conda env create -f environment.yml` as mentioned in the hls_tutorials page . I am getting this compilation error, and after exploring it alot, i am not able to solve this. 

- The error occurs after i run the `hls_model.compile()`. ( console output below)
- It did generate the vivado hls files. even after the failed compilation. if i run `hls_model.build()` . it does run the csim and build the project. 
- As the model is not compiled successfully, i cant use further functions like, `y_hls = hls_model.predict(X_test)` etc.

Could anyone please help me out with this error 

Thansk alot 

****************************************************************************
**Console Output After executing `hls_model.compile()`**
****************************************************************************

Writing HLS project
Done
In file included from firmware/parameters.h:7:0,
                 from firmware/myproject.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::load_compressed_weights_from_txt(T*, const char*)’:
firmware/nnet_utils/nnet_helpers.h:92:26: error: range-based ‘for’ loops are not allowed in C++98 mode
             for (char c: extra_chars) {
                          ^
firmware/nnet_utils/nnet_helpers.h:95:23: error: ‘std::string’ has no member named ‘back’
             if (token.back() == ',') {
                       ^
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::load_exponent_weights_from_txt(T*, const char*)’:
firmware/nnet_utils/nnet_helpers.h:138:26: error: range-based ‘for’ loops are not allowed in C++98 mode
             for (char c: extra_chars) {
                          ^
firmware/nnet_utils/nnet_helpers.h:141:23: error: ‘std::string’ has no member named ‘back’
             if (token.back() == ',') {
                       ^
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::save_layer_output(data_T*, const char*, size_t)’:
firmware/nnet_utils/nnet_helpers.h:233:47: error: no matching function for call to ‘std::basic_fstream<char>::open(std::basic_ostringstream<char>::__string_type&, const openmode&)’
         out.open(filename.str(), std::ios::app);
                                               ^
firmware/nnet_utils/nnet_helpers.h:233:47: note: candidate is:
In file included from firmware/nnet_utils/nnet_helpers.h:26:0,
                 from firmware/parameters.h:7,
                 from firmware/myproject.cpp:22:
/usr/include/c++/4.8.2/fstream:886:7: note: void std::basic_fstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
       open(const char* __s,
       ^
/usr/include/c++/4.8.2/fstream:886:7: note:   no known conversion for argument 1 from ‘std::basic_ostringstream<char>::__string_type {aka std::basic_string<char>}’ to ‘const char*’
In file included from firmware/parameters.h:7:0,
                 from firmware/myproject.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::save_layer_output(hls::stream<srcType>&, const char*, size_t)’:
firmware/nnet_utils/nnet_helpers.h:263:47: error: no matching function for call to ‘std::basic_fstream<char>::open(std::basic_ostringstream<char>::__string_type&, const openmode&)’
         out.open(filename.str(), std::ios::app);
                                               ^
firmware/nnet_utils/nnet_helpers.h:263:47: note: candidate is:
In file included from firmware/nnet_utils/nnet_helpers.h:26:0,
                 from firmware/parameters.h:7,
                 from firmware/myproject.cpp:22:
/usr/include/c++/4.8.2/fstream:886:7: note: void std::basic_fstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
       open(const char* __s,
       ^
/usr/include/c++/4.8.2/fstream:886:7: note:   no known conversion for argument 1 from ‘std::basic_ostringstream<char>::__string_type {aka std::basic_string<char>}’ to ‘const char*’
In file included from firmware/parameters.h:7:0,
                 from firmware/myproject.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::copy_data_axi(std::vector<_Tp>, dst_T*)’:
firmware/nnet_utils/nnet_helpers.h:305:14: error: ‘i’ does not name a type
     for(auto i = 0; i < SIZE; i++)
              ^
firmware/nnet_utils/nnet_helpers.h:305:21: error: expected ‘;’ before ‘i’
     for(auto i = 0; i < SIZE; i++)
                     ^
firmware/nnet_utils/nnet_helpers.h:305:21: error: ‘i’ was not declared in this scope
[... 646 lines cut ...]

****************************************************************************
**Console output after `hls_model.build()`**
****************************************************************************

{'CSimResults': [['0.0253906',
   '0.0917969',
   '0.161133',
   '0.638672',
   '0.103516']],
 'EstimatedClockPeriod': '4.371',
 'BestLatency': '10',
 'WorstLatency': '10',
 'IntervalMin': '1',
 'IntervalMax': '1',
 'BRAM_18K': '4',
 'DSP48E': '2784',
 'FF': '13182',
 'LUT': '108621',
 'URAM': '0',
 'AvailableBRAM_18K': '5376',
 'AvailableDSP48E': '12288',
 'AvailableFF': '3456000',
 'AvailableLUT': '1728000',
 'AvailableURAM': '1280'}

vloncar, 2021-12-05:
If `build` ran successfully and `compile` failed that would suggest the issue is with your host compiler (`g++`), but it is hard to tell without more context. Can you provide a minimal example that demonstrates the issue?

rtspk, 2021-12-05:
I have a miniconda on Linux (Red Hat Enterprise Linux Server release 7.9 (Maipo)).  
- I have created an environment., using the  `conda env create -f environment.yml`. 
- I am running the examples from the hls4ml_tutorial repo. I ran the first tutorial `part1_getting_started.py`. the console output after the compilation step is shown in my first message. 
- In my miniconda environment, i tried to check whether i have g++ installed or not. I get the following output

[image]

rtspk, 2021-12-05:
it worked. It seems that my g++ compiler version was old ( or not installed at all). I use the following command. and now my g++ version is 11. 
command was: `conda install -c conda-forge cxx-compiler`

Thanks alot @vloncar for narrowing down the cause of the error. :)

closed 2021-12-05 by rtspk