fastmachinelearning/hls4ml #473: Cant launch csim and cosim on tutorial scripts

samiBENALI, 2021-12-13:
Greetings,

I was trying to test my own model implementation using the hls4ml server, i tried using cosim and csim simulations with the build function: 

`hls_model.build(csim=True,cosim=True)`

however i get the following output with an error:

`
****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'jovyan' on host 'jupyter-samibenali' (Linux_x86_64 version 5.4.129+) on Sun Dec 12 15:36:35 UTC 2021
INFO: [HLS 200-10] In directory '/home/jovyan/test_implementation_accurate'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/home/jovyan/test_implementation_accurate/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/jovyan/test_implementation_accurate/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 100ns.
INFO: [HLS 200-10] Setting target device to 'xc7a100t-ftg256-2'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
csim.mk:78: recipe for target 'obj/myproject_test.o' failed
In file included from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:0,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/c++config.h:495,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/iosfwd:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/ios:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/istream:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/fstream:38,
                 from ../../../../myproject_test.cpp:19:
/usr/include/features.h:424:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
make: *** [obj/myproject_test.o] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Sun Dec 12 15:36:37 2021...

{'EstimatedClockPeriod': '87.248',
 'BestLatency': '9732',
 'WorstLatency': '9733',
 'IntervalMin': '1569',
 'IntervalMax': '9409',
 'BRAM_18K': '14',
 'DSP48E': '22',
 'FF': '34595',
 'LUT': '50913',
 'URAM': '0',
 'AvailableBRAM_18K': '270',
 'AvailableDSP48E': '240',
 'AvailableFF': '126800',
 'AvailableLUT': '63400',
 'AvailableURAM': '0'}`

I attempted to try the simulation with the "part1_getting_started" tutorial (without modifying anything in the provided code) but i it fails to do so and i get a similar output:
`****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'jovyan' on host 'jupyter-sbasam' (Linux_x86_64 version 5.4.129+) on Mon Dec 13 12:24:29 UTC 2021
INFO: [HLS 200-10] In directory '/home/jovyan/model_1/hls4ml_prj'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Creating and opening project '/home/jovyan/model_1/hls4ml_prj/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/jovyan/model_1/hls4ml_prj/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-10] Setting target device to 'xcu250-figd2104-2L-e'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
csim.mk:78: recipe for target 'obj/myproject_test.o' failed
In file included from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:0,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/c++config.h:495,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/iosfwd:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/ios:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/istream:38,
                 from /opt/Xilinx/Vivado/2019.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/fstream:38,
                 from ../../../../myproject_test.cpp:19:
/usr/include/features.h:424:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
make: *** [obj/myproject_test.o] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Mon Dec 13 12:24:33 2021...
Synthesis report not found.`

vloncar, 2021-12-13:
Your Vivado installation is the issue. You're missing some libraries (`multilib` probably).

samiBENALI, 2021-12-13:
Thank you for your reply, im using the gitlab server for bothe the tutorial and my own model. Is there a way to add the library on the server  ?

rtspk, 2021-12-13:
Hey @samiBENALI  just chipping in to answer your doubt related to Vivado 2019.1 . I also use Vivado 2019.1 and it works fine at my end.

samiBENALI, 2021-12-14:
Thank you for your reply @rtspk  , i tried adding all missing libraries required by vivado and vivado hls on my 2019.1 installation, and i attepted to run the cosimulation using the project of the first tutorial (part1_getting_started), the synthesis and c simulation bothe work, however the cosimulation doesnt work and i get the following errors:

```
***** C/RTL SYNTHESIS COMPLETED IN 0h9m17s *****
***** C/RTL SIMULATION *****
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [COSIM 212-47] Using XSIM for RTL simulation.
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/opt/Xilinx/Vivado/2019.1/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling apatb_myproject.cpp
   Compiling myproject.cpp_pre.cpp.tb.cpp
cosim.tv.mk:65: recipe for target 'obj/myproject.cpp_pre.cpp.tb.o' failed
In file included from /opt/Xilinx/Vivado/2019.1/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /opt/Xilinx/Vivado/2019.1/include/hls_fpo.h:186,
                 from /opt/Xilinx/Vivado/2019.1/include/hls_half.h:44,
                 from /opt/Xilinx/Vivado/2019.1/include/etc/ap_private.h:90,
                 from /opt/Xilinx/Vivado/2019.1/include/ap_common.h:641,
                 from /opt/Xilinx/Vivado/2019.1/include/ap_int.h:54,
                 from /home/user/Desktop/zipped_model_1/hls4ml_prj/firmware/myproject.h:23,
                 from /home/user/Desktop/zipped_model_1/hls4ml_prj/firmware/myproject.cpp:21:
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:181:9: error: ‘__gmp_const’ does not name a type
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
         ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:181:48: error: expected constructor, destructor, or type conversion before ‘;’ token
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
                                                ^
In file included from /opt/Xilinx/Vivado/2019.1/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /opt/Xilinx/Vivado/2019.1/include/hls_fpo.h:186,
                 from /opt/Xilinx/Vivado/2019.1/include/hls_half.h:44,
                 from /opt/Xilinx/Vivado/2019.1/include/etc/ap_private.h:90,
                 from /opt/Xilinx/Vivado/2019.1/include/ap_common.h:641,
                 from /opt/Xilinx/Vivado/2019.1/include/ap_int.h:54,
                 from /home/user/Desktop/zipped_model_1/hls4ml_prj/firmware/myproject.h:23,
                 from /home/user/Desktop/zipped_model_1/hls4ml_prj/firmware/myproject.cpp:21:
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:244:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_version _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:245:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_patches _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:260:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char *
  ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:19: error: ‘mpfr_srcptr’ was not declared in this scope
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                   ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:42: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                          ^
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:54: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                      ^
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:66: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                                  ^
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:79: error: expected primary-expression before ‘)’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                                               ^
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:299:79: error: expression list treated as compound expression in initializer [-fpermissive]
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:329:24: error: ‘mpfr_srcptr’ has not been declared
   mpfr_get_f _MPFR_PROTO ((mpf_ptr, mpfr_srcptr, mpfr_rnd_t));
                        ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:340:27: error: ‘__gmp_const’ has not been declared
   mpfr_set_str _MPFR_PROTO ((mpfr_ptr, __gmp_const char *, int, mpfr_rnd_t));
                           ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:340:44: error: two or more data types in declaration of ‘parameter’
   mpfr_set_str _MPFR_PROTO ((mpfr_ptr, __gmp_const char *, int, mpfr_rnd_t));
                                            ^
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:340:45: error: expected ‘)’ before ‘,’ token
   mpfr_set_str _MPFR_PROTO ((mpfr_ptr, __gmp_const char *, int, mpfr_rnd_t));
[... 832 lines cut ...]
 __MPFR_DECLSPEC mpfr_exp_t mpfr_custom_get_exp  _MPFR_PROTO ((mpfr_srcptr));
                                  ^~~~~~~~~~~
/opt/Xilinx/Vivado/2019.1/include/mpfr.h:724:28: error: ‘mpfr_srcptr’ was not declared in this scope
 __MPFR_DECLSPEC int    mpfr_custom_get_kind   _MPFR_PROTO ((mpfr_srcptr));
                            ^~~~~~~~~~~
[... cut ...]
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total elapsed time: 578.3 seconds; peak allocated memory: 1.226 GB.
INFO: [Common 17-206] Exiting vivado_hls at Tue Dec 14 15:45:32 2021...
child process exited 
```

Even when trying with another installation on a different machine, i still get the same error, which makes me think that the issue might be with the generated c code itself.

rtspk, 2021-12-16:
@samiBENALI  sorry bro, not sure about this error. may this link can help you ? https://support.xilinx.com/s/question/0D52E00006iHkfp/vivado-20153-hls-bug-gmph?language=en_US

khizar-anjum, 2022-04-06:
This link here is a good resource, which tackles and cites the exact same issue: [https://chhzh123.github.io/blogs/2020-03-11-vivado-hls/](https://chhzh123.github.io/blogs/2020-03-11-vivado-hls/)

Translating into English, the main thing is:   
This is caused by the conflict between the header files used internally by Xilinx and the system header files, which can be includefixed by modifying the files inside. Modify /tools/Xilinx/Vivado/20XX.X/include/mpfr.h, import the system into the header file <gmp.h> and change it to the current folder to import "gmp.h"

```
#ifndef __GMP_H__
#include "gmp.h"
#endif
```

You may have to do it in files other than `mpfr.h` if the error persists. Just search the instances of `#include <gmp.h>` and replace with `#include "gmp.h"`

holoLens8, 2022-11-06:
Hi, I've got the same error as @samiBENALI and can't seem to find a workaround. I have tried the solution proposed by @khizar-anjum, however it did not work. I used Vivado HLS 18.2.1.
Did anyone find a solution?
Thanks

vloncar, 2022-11-06:
Y'all need to look into the HLS installation documentation. Make sure you're using a supported version of Linux and have installed the required libraries (like gcc multilib). Vivado installation won't resolve dependencies like apt/yum does, so you need to ensure system packages are installed before the main installation. If you use a more modern distro than what the Vivado installation supports, you can try installing in docker/singularity/apptainer.

closed 2022-12-05 by jmduarte