fastmachinelearning/hls4ml #476: Why am I getting this error : ´AttributeError: 'NoneType' object has no attribute 'shape'´ ?

AnouarITI, 2021-12-20:
Hello, I trained a ResNet model on Keras, then I followed the getting started a notebook to use hls4ml for research purposes. First, please check the config output using the following code:

```
config = hls4ml.utils.config_from_keras_model(model, granularity='model')
print_dict(config)
```
```
Model
  Precision:         ap_fixed<16,6>
  ReuseFactor:       1
  Strategy:          Latency
-----------------------------------
Interpreting Model
Topology:
Layer name: input_2, layer type: InputLayer, input shapes: [[None, 1024, 2]], output shape: [None, 1024, 2]
Layer name: reshape_1, layer type: Reshape, input shapes: [[None, 1024, 2]], output shape: [None, 1, 1024, 2]
Layer name: ReStk1_conv1, layer type: Conv2D, input shapes: [[None, 1, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: ReStk1_conv2, layer type: Conv2D, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: ReStk1_conv3, layer type: Conv2D, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: add_12, layer type: Merge, input shapes: [[None, 32, 1024, 2], [None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: activation_13, layer type: Activation, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: ReStk1_conv4, layer type: Conv2D, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: ReStk1_conv5, layer type: Conv2D, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: add_13, layer type: Merge, input shapes: [[None, 32, 1024, 2], [None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: activation_14, layer type: Activation, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 1024, 2]
Layer name: max_pooling2d_6, layer type: MaxPooling2D, input shapes: [[None, 32, 1024, 2]], output shape: [None, 32, 512, 1]
Layer name: ReStk2_conv1, layer type: Conv2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: ReStk2_conv2, layer type: Conv2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: ReStk2_conv3, layer type: Conv2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: add_14, layer type: Merge, input shapes: [[None, 32, 512, 1], [None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: activation_15, layer type: Activation, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: ReStk2_conv4, layer type: Conv2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: ReStk2_conv5, layer type: Conv2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: add_15, layer type: Merge, input shapes: [[None, 32, 512, 1], [None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: activation_16, layer type: Activation, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 512, 1]
Layer name: max_pooling2d_7, layer type: MaxPooling2D, input shapes: [[None, 32, 512, 1]], output shape: [None, 32, 256, 1]
Layer name: ReStk3_conv1, layer type: Conv2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: ReStk3_conv2, layer type: Conv2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: ReStk3_conv3, layer type: Conv2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: add_16, layer type: Merge, input shapes: [[None, 32, 256, 1], [None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: activation_17, layer type: Activation, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: ReStk3_conv4, layer type: Conv2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: ReStk3_conv5, layer type: Conv2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: add_17, layer type: Merge, input shapes: [[None, 32, 256, 1], [None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: activation_18, layer type: Activation, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 256, 1]
Layer name: max_pooling2d_8, layer type: MaxPooling2D, input shapes: [[None, 32, 256, 1]], output shape: [None, 32, 128, 1]
Layer name: ReStk4_conv1, layer type: Conv2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: ReStk4_conv2, layer type: Conv2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: ReStk4_conv3, layer type: Conv2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: add_18, layer type: Merge, input shapes: [[None, 32, 128, 1], [None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: activation_19, layer type: Activation, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: ReStk4_conv4, layer type: Conv2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: ReStk4_conv5, layer type: Conv2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: add_19, layer type: Merge, input shapes: [[None, 32, 128, 1], [None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: activation_20, layer type: Activation, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 128, 1]
Layer name: max_pooling2d_9, layer type: MaxPooling2D, input shapes: [[None, 32, 128, 1]], output shape: [None, 32, 64, 1]
Layer name: ReStk5_conv1, layer type: Conv2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: ReStk5_conv2, layer type: Conv2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: ReStk5_conv3, layer type: Conv2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: add_20, layer type: Merge, input shapes: [[None, 32, 64, 1], [None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: activation_21, layer type: Activation, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: ReStk5_conv4, layer type: Conv2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: ReStk5_conv5, layer type: Conv2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: add_21, layer type: Merge, input shapes: [[None, 32, 64, 1], [None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: activation_22, layer type: Activation, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 64, 1]
Layer name: max_pooling2d_10, layer type: MaxPooling2D, input shapes: [[None, 32, 64, 1]], output shape: [None, 32, 32, 1]
Layer name: ReStk6_conv1, layer type: Conv2D, input shapes: [[None, 32, 32, 1]], output shape: [None, 32, 32, 1]
Layer name: ReStk6_conv2, layer type: Conv2D, input shapes: [[None, 32, 32, 1]], output shape: [None, 32, 32, 1]
Layer name: ReStk6_conv3, layer type: Conv2D, input shapes: [[None, 32, 32, 1]], output shape: [None, 32, 32, 1]
Layer name: add_22, layer type: Merge, input shapes: [[None, 32, 32, 1], [None, 32, 32, 1]], output shape: [None, 32, 32, 1]
[... 51 lines cut ...]

/usr/local/lib/python3.8/dist-packages/hls4ml/model/hls_model.py in _optimize_model(self, optimizers)
    334 
    335     def _optimize_model(self, optimizers):
--> 336         optimize_model(self, optimizers)
    337 
    338     def make_node(self, kind, name, attributes, inputs, outputs=None):

/usr/local/lib/python3.8/dist-packages/hls4ml/model/optimizer/optimizer.py in optimize_model(model, passes)
     36         for opt in optimizers:
     37             for node in model.graph.values():
---> 38                 if opt.match(node):
     39                     res = opt.transform(model, node)
     40                     if res:

/usr/local/lib/python3.8/dist-packages/hls4ml/model/optimizer/passes/repack_stream.py in match(self, node)
     95             inp1 = node.get_input_variable(node.inputs[0])
     96             inp2 = node.get_input_variable(node.inputs[1])
---> 97             return inp1.shape != inp2.shape
     98         else:
     99             return False

AttributeError: 'NoneType' object has no attribute 'shape'
```
Could someone explain to me what is the problem here and how to fix it? Thank you

jmduarte, 2021-12-20:
Hello @AnouarITI 
What version of hls4ml are you using?
Some recent updates may help this, e.g. https://github.com/fastmachinelearning/hls4ml/pull/443

Thanks!

AnouarITI, 2022-01-09:
Hello @jmduarte 
please find attached a notebook with the script to reproduce the error. Sorry, I can not share the dataset but at least it can give you more visibility of the error.
[script.txt](https://github.com/fastmachinelearning/hls4ml/files/7836058/script.txt)

SteCla0, 2022-06-22:
I have the same error when I use "add layers" in my model
https://github.com/fastmachinelearning/hls4ml/issues/578

wilfredkisku, 2022-07-05:
@SteCla0 did you find a resolution?

jmduarte, 2023-04-09:
This should be resolved in the current `main` branch, but reopen if not.

closed 2023-04-09 by jmduarte