fastmachinelearning/hls4ml #485: Keras 2D ConV model HLS synthesis warning Generated.

Pythagoras21, 2021-12-31:
HI! 
I am using keras sequential model :

i trained and save this model in h5 format and loaded it using 
`model=load_model('CNN_Test2.h5')`
and used 
```
config= hls4ml.utils.config_from_keras_model(model,granularity='model')
hls_model=hls4ml.converters.convert_from_keras_model(model,hls_config=config,output_dir='/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/Generated_C',
                                                                                 part='xcu250-figd2104-2L-e')
```

the model is:
```
model.add(keras.Input(shape=(224, 224, 3))) 
model.add(layers.Conv2D(32, 3, activation="relu"))
model.add(layers.MaxPooling2D(3))
model.add(layers.Conv2D(32, 3, activation="relu"))
model.add(layers.MaxPooling2D(5))
model.add(layers.Flatten())
model.add(layers.Dense(128,activation='relu',kernel_regularizer=keras.regularizers.l2(0.001)))
and other layers 
```

Model Config is : `{'Model':` {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': 64, 'Strategy': 'Resource', 'IOType': `'io_serial'}}`

upon creating the model it showed: 
```
Creating HLS model
WARNING: Invalid ReuseFactor=64 with "Resource" strategy in layer "conv2d". Using ReuseFactor=54 instead. Valid ReuseFactor(s): 3,9,27,54,108,216,432,864.
WARNING: Invalid ReuseFactor=64 with "Resource" strategy in layer "conv2d_1". Using ReuseFactor=72 instead. Valid ReuseFactor(s): 2,3,4,6,8,9,12,16,18,24,32,36,48,72,96,144,288,576,1152,2304,4608,9216.
```

and upon building the model: hls_model.build()

I have :

```
****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /media/formal/E016E9BA16E991B8/NECBH/Xilinx_Installed_Dec/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/media/formal/E016E9BA16E991B8/NECBH/Xilinx_Installed_Dec/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'formal' on host 'formal-CELSIUS-R940power' (Linux_x86_64 version 5.4.0-87-generic) on Fri Dec 31 11:09:31 IST 2021
INFO: [HLS 200-10] On os Ubuntu 18.04.6 LTS
INFO: [HLS 200-10] In directory '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/31Dec_Generated_C'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Creating and opening project '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/31Dec_Generated_C/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/31Dec_Generated_C/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-10] Setting target device to 'xcu250-figd2104-2L-e'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
[I 11:09:54.744 NotebookApp] Saving file at /HLS_Train_Files_25Dec/Synthesis_test/SynthesisTest1.ipynb
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
../../../../firmware/myproject.cpp: In function ‘void myproject(input_t*, result_t*, short unsigned int&, short unsigned int&)’:
../../../../firmware/myproject.cpp:37:46: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     const_size_in_1 = N_INPUT_1_1*N_INPUT_2_1*N_INPUT_3_1;
                                               
[... 89 lines cut ...]
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/nnet_utils/nnet_dense_latency.h:79:2
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense_latency.h:76:9
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:66:68
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:66:72
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:78:72
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:78:76
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:90:67
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:90:71
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:98:72
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:98:77
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:106:72
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:106:77
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/myproject.cpp:37:2
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/myproject.cpp:38:5
WARNING: [HLS 200-471] Dataflow form checks found 15 issue(s) in file firmware/myproject.cpp
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:45:22 ; elapsed = 00:46:51 . Memory (MB): peak = 1123.949 ; gain = 675.168 ; free physical = 18817 ; free virtual = 109908
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:45:22 ; elapsed = 00:46:51 . Memory (MB): peak = 1123.949 ; gain = 675.168 ; free physical = 18817 ; free virtual = 109908
INFO: [HLS 200-10] Starting code transformations ...
```

Why are these warnings coming, is there something wrong i did. what can i do to fix this ?

Can we not use keras 2D ConV sequential model for synthesis?

Please help.

vloncar, 2021-12-31:
Try `convert_from_keras_model(..., io_type='io_stream')`

Pythagoras21, 2022-01-03:
Hi, i tried your suggestion, stil pre-synthesis is not working.
i have attached the output: please have a look

`tail: vivado_hls.log: file truncated`

```
****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /media/formal/E016E9BA16E991B8/NECBH/Xilinx_Installed_Dec/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/media/formal/E016E9BA16E991B8/NECBH/Xilinx_Installed_Dec/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'formal' on host 'formal-CELSIUS-R940power' (Linux_x86_64 version 5.4.0-87-generic) on Mon Jan 03 13:47:21 IST 2022
INFO: [HLS 200-10] On os Ubuntu 18.04.6 LTS
INFO: [HLS 200-10] In directory '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/Small_test_3rd_jan/model_1/hls4ml_prj'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/Small_test_3rd_jan/model_1/hls4ml_prj/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/formal/Desktop/NECBH/HLS/HLS_Train_Files_25Dec/Synthesis_test/Small_test_3rd_jan/model_1/hls4ml_prj/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xcu250-figd2104-2L-e'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 200-40] In file included from firmware/myproject.cpp:1:
firmware/myproject.cpp:37:27: warning: implicit conversion from 'int' to 'unsigned short' changes value from 150528 to 19456 [-Wconstant-conversion]
 const_size_in_1 = 224*224*3;
                 ~ ~~~~~~~^~
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:22:
In file included from firmware/parameters.h:11:
In file included from firmware/nnet_utils/nnet_conv2d.h:24:
firmware/nnet_utils/nnet_conv2d_latency.h:214:69: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
                                if ((oh*CONFIG_T::stride_height+fh) < CONFIG_T::pad_top
                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv2d.h:83:9: note: in instantiation of function template specialization 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config2>' requested here
        conv_2d_latency_cl<data_T, res_T, CONFIG_T>(data, res, weights, biases);
        ^
firmware/myproject.cpp:68:2: note: in instantiation of function template specialization 'nnet::conv_2d_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config2>' requested here
 nnet::conv_2d_cl<input_t, layer2_t, config2>(input_9, layer2_out, w2, b2);
 ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:22:
In file included from firmware/parameters.h:11:
In file included from firmware/nnet_utils/nnet_conv2d.h:24:
firmware/nnet_utils/nnet_conv2d_latency.h:216:67: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
                                || (ow*CONFIG_T::stride_width+fw) < CONFIG_T::pad_left
                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv2d_latency.h:30:69: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
                                if ((oh*CONFIG_T::stride_height+fh) < CONFIG_T::pad_top
                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv2d_latency.h:191:31: note: in instantiation of function template specialization 'nnet::compute_multiplier_limit_conv2d<config2>' requested here
 const int multiplier_limit = compute_multiplier_limit_conv2d<CONFIG_T>(weights);
                              ^
firmware/nnet_utils/nnet_conv2d.h:83:9: note: in instantiation of function template specialization 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config2>' requested here
        conv_2d_latency_cl<data_T, res_T, CONFIG_T>(data, res, weights, biases);
        ^
firmware/myproject.cpp:68:2: note: in instantiation of function template specialization 'nnet::conv_2d_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config2>' requested here
 nnet::conv_2d_cl<input_t, layer2_t, config2>(input_9, layer2_out, w2, b2);
 ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:22:
[... 188 lines cut ...]
[... cut ...] ig12>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Mon Jan  3 14:15:10 2022...
```

Please help me solve this issue

Pythagoras21, 2022-01-04:
I have uploaded the model in h5 format here on github in the link below
[](https://github.com/Pythagoras21/HLS_CNN_files)

Please find the model along with the weights and yml config file. It is not synthesizing at my end with error :
`ERROR: [XFORM 203-504] Stop unrolling loop 'ConvOutHeight' (firmware/nnet_utils/nnet_conv2d_latency.h:195) in function 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config12>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "`

Pythagoras21, 2022-01-08:
Solved after changing tensor shape of input and using numpy arrays instead of flow from  directory.
Reduced model size and pruned before synthesis.

closed 2022-01-08 by Pythagoras21