fastmachinelearning/hls4ml #486: Methods for Post Training Quantization

HenningCode, 2022-01-05:
Hello Guys, 

as the title suggest I am looking for methods to use PTQ. I tried to convert a Keras to QKeras but the drop of in accuracy is pretty significant. Are there any other tensorflow based quantization methods compatible with HLS4ML ? 

Greetings

hepkosmos, 2022-01-05:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2022-01-05T17:26:26
          
          
            * 메일 제목 : 
[fastmachinelearning/hls4ml] Methods for Post Training Quantization (Issue #486)
          
          
            
              
                * 실패 사유 : 
              
              
                
받는 사람이 회원님의 메일을 수신차단 하였습니다.
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

jmduarte, 2022-06-22:
Hi @HenningCode, QKeras is the main way we support of doing quantization-aware training (QAT). 

For Post-training quantization (PTQ), it is not necessary to go through QKeras. You can directly convert the model using hls4ml, you should just make sure to use a wide enough bit width.

closed 2022-06-22 by jmduarte