fastmachinelearning/hls4ml #488: Vivado HLS Error

adhulsv, 2022-01-06:
Generating csim.exe
obj/main.o: In function `floatPrototypec':
E:\HLSnew\try11\solution1\csim\build/../../../floatPrototype.h:20: undefined reference to `floatPrototypec::multiply()'
obj/main.o: In function `floatDriverc':
E:\HLSnew\try11\solution1\csim\build/../../../floatDriver.h:23: undefined reference to `floatDriverc::test()'
collect2: ld returned 1 exit status
make: *** [csim.exe] Error 1
ERROR: [SIM 211-100] CSim file generation failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
5
    while executing
"source E:/HLSnew/try11/solution1/csim.tcl"
    invoked from within
"hls::main E:/HLSnew/try11/solution1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C simulation.

hepkosmos, 2022-01-06:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2022-01-06T18:39:10
          
          
            * 메일 제목 : 
[fastmachinelearning/hls4ml] Vivado HLS Error (Issue #488)
          
          
            
              
                * 실패 사유 : 
              
              
                
받는 사람이 회원님의 메일을 수신차단 하였습니다.
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

jmduarte, 2022-06-19:
hi @adhulsv, I think we would need more information to help. What type of model / version of Vivado / platform / etc. are you using? Could you send us the code to reproduce the error? 

Closing for now, but re-open if you want to send us more information.

closed 2022-06-19 by jmduarte