fastmachinelearning/hls4ml #491: Create and deploy block design

HenningCode, 2022-01-18:
Hey Guys,

I worked with hls4ml the past few months and now I finally wanted to deploy a model I transformed with hls4ml, but I dont really know how I have to initialize my IP core created with hls4ml. 
I attached my Block design I tried to get work, but again I struggle to init the ip core. 
[image]

To explore a little bit further I looked into an example of @jmduarte tiny-0.5 example. The Ip cores look completely I know that the version of hls4ml used for this example is a lot older but I feel like it should look at least a little bit the same.  I found two differences in the config, I used `io_stream` and @jmduarte used `io_parallel` and I am trying to target a zcu104 and not a pynq-z2.

[image]

Any tips on how I should try to go forward would be appreciated. 

I could also attach the code I got so far in the sdk, but I dont think that would help, but this is my hls4ml config.

``` 
    backend_config = {}
    backend_config['HLSConfig'] = config
    backend_config['Backend'] = 'Vivado'
    backend_config['ClockPeriod'] = 1
    backend_config['IOType'] = 'io_stream'
    backend_config['OutputDir'] = output_dir+'/cnn2_hls'
    backend_config['ProjectName'] = 'mnist_test'
    backend_config['XilinxPart']= 'xczu7ev-ffvc1156-2-e'
    backend_config['Interface'] = 'm_axi'

```
And this is my config:

```
LayerName:
  input_1_m:
    Precision:
      result: ap_fixed<16,6>
  qact_1_m:
    Precision:
      result: ap_ufixed<4,2>
    ReuseFactor: 300000
  qact_2_m:
    Precision:
      result: ap_ufixed<4,2>
    ReuseFactor: 300000
  qdense_1_m:
    Precision:
      bias: ap_fixed<5,1>
      weight: ap_fixed<5,1>
    ReuseFactor: 300000
  qdense_2_m:
    Precision:
      bias: ap_fixed<5,1>
      weight: ap_fixed<5,1>
    ReuseFactor: 300000
  qdense_3_m:
    Precision:
      bias: ap_fixed<5,1>
      weight: ap_fixed<5,1>
    ReuseFactor: 300000
  sact_1_m:
    Precision: ap_fixed<16,6>
    ReuseFactor: 300000
    exp_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
    inv_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
    table_size: 1024
Model:
  Precision: ap_fixed<16,6>
  ReuseFactor: 300000
  Strategy: Resource

```

thesps, 2022-01-19:
Hi, try targeting "VivadoAccelerator" backend instead, like:
`    backend_config['Backend'] = 'VivadoAccelerator'`
That will add the AXI interfaces you're expecting. When you write the project, there will also then be a `design.tcl` file that creates a block design in Vivado IP Integrator that you can either use to make a bitfile, or use as a reference if you want to do something custom.

You may also need to use `backend_config['Interface'] = 'axi_stream'`.

HenningCode, 2022-01-19:
I tried that already, but than this happens: `WARNING: [HLS 200-40] Resetting target device to 'xc7z020-clg400-1`, but I want to target an ZCU104 or is this not a Problem if I create the Block design by myself?

thesps, 2022-01-19:
> I tried that already, but than this happens: WARNING: [HLS 200-40] Resetting target device to 'xc7z020-clg400-1, but I want to target an ZCU104 or is this not a Problem if I create the Block design by myself?

Right, there is the `'Board'` config field which overrides the `'Part'` field, and which defaults to pynq-z2. We don't have ZCU104 support yet, but we do have full scripts for ZCU102 which may be a better starting point than the pynq-z2 (I think the PS IP block is different, for example). 

I think the approach of choosing a supported board and modifying what you need is the right one for now. You can then export the HLS NN IP with AXI wrapper and integrate it into a block diagram yourself, or use our design.tcl.

> And I cant run the design.tcl script. This is the returned error:

Did you execute it with Vivado or Vivado HLS? The command would be `vivado -mode batch -source design.tcl`. That would create a Vivado project, add the NN IP and other IPs (PS, DMA), and run synthesis and implementation. You can comment out [these lines](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado_accelerator/zcu102/tcl_scripts/axi_stream_design.tcl#L52-L58) if you just want to inspect the block diagram.

HenningCode, 2022-01-26:
[image]

I managed to create the Block Design how I would like it to look like, but it still doesn't really work. I can't just use the FIFO IP to create Inputs to my custom IP. 

I think I initialized it correctly with:
```
    Xil_Out32(XPAR_MNIST_TEST_AXI_0_S_AXI_AXILITES_BASEADDR,0);
    Xil_Out32(XPAR_MNIST_TEST_AXI_0_S_AXI_AXILITES_BASEADDR,0x81);
```
But my Output looks always the same and if I just use basic send and receive functions it doesn't even work. 

These are my basic functions:

```
 int sendData(){
	    for(u32 i = 0x00; i < INPUT_LEN; i++){
	        SourceBuffer[i] = i;
	    }
	    xil_printf("[INFO]: Sending Data\n");

		for(u32 j=0;j<INPUT_LEN;j++){
			if(XLlFifo_iTxVacancy(&Fifo)){
			// writes data into the FIFO buffer
			XLlFifo_TxPutWord(&Fifo, *(SourceBuffer+j));
			}
		}

	    XLlFifo_iTxSetLen(&Fifo, INPUT_LEN * sizeof(u32));
	    while(!(XLlFifo_IsTxDone(&Fifo)));

	    return XST_SUCCESS;
 }

 int receiveData(){
	 xil_printf("[INFO]: Waiting for Output\n");
	 while(!XLlFifo_iRxOccupancy(&Fifo));

	 u32 ReceiveLength = (XLlFifo_iRxGetLen(&Fifo)) / sizeof(u32);

	 for(u32 i = 0x00; i < ReceiveLength; i++){
	     u32 RxWord = XLlFifo_RxGetWord(&Fifo);
	     DestinationBuffer[i] = RxWord;
	     xil_printf("        Data %lu: %x\n", i, RxWord);
	 }

	 return XST_SUCCESS;
 }
```
I am trying to create an MNIST FC test, so my input is 784 long, but thats not enough for the custom IP. He needs an Input of 1568 so twice the expected amount of data. and I have to call my SendData method twice before I get an generic Output, thats always the same regardless if I change the input. 

I cant even just change the `INPUT_LEN = 1568`, than it doesnt work again. The IP needs to receive Information twice. 

I found out about this with this Method:

```
int sendTillFull(){
	u8 data = 0x56;

	while(!XLlFifo_iRxOccupancy(&Fifo)){

		for(u32 i=0x00;i<INPUT_LEN;i++){
			if(XLlFifo_iTxVacancy(&Fifo)){
				// writes data into the FIFO buffer
				XLlFifo_TxPutWord(&Fifo,data);
			}
		}

		XLlFifo_iTxSetLen(&Fifo, INPUT_LEN * sizeof(u8));
		while(!(XLlFifo_IsTxDone(&Fifo)));
		xil_printf("[INFO]: Count:%d\n",count);
		count++;
	}

	//xil_printf("[INFO]: Count:%d\n",count);

	u32 ReceiveLength = (XLlFifo_iRxGetLen(&Fifo)) / sizeof(u32);

	for(u32 i = 0x00; i < ReceiveLength; i++){
		u32 RxWord = XLlFifo_RxGetWord(&Fifo);
		DestinationBuffer[i] = RxWord;
		xil_printf("        Data %lu: %x\n", i, RxWord);
	}

	return XST_SUCCESS;
}
```

That just sends data until the IP core return something. 

Is my approach to use an FIFO Ip wrong ? 
This is still my hls4ml config for a fully connected network:
 ```
    backend_config = {}
    backend_config['HLSConfig'] = config
    backend_config['Backend'] = 'VivadoAccelerator'
    backend_config['ClockPeriod'] = 5
    backend_config['IOType'] = 'io_stream'
    backend_config['OutputDir'] = output_dir+'/io_para_ACC'
    backend_config['ProjectName'] = 'mnist_test'
    backend_config['XilinxPart'] = 'xczu7ev-ffvc1156-2-e'
    backend_config['Interface'] = 'axi_stream'
    backend_config['InputData'] = 'input_features.dat'
    backend_config['OutputPredictions'] = 'output_predictions.dat'
```

thesps, 2022-01-26:
Did you try modifying our block diagram? We use an AXI DMA IP for the AXI4 to AXI-Stream step. Personally, I don't have experience with the AXI-Stream FIFO IP for that purpose.

I made a quick attempt at adding support for ZCU104 to hls4ml based on our existing ZCU102 scripts. I was able to build a bitfile, but I don't have a board to test on. I modified [these lines](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado_accelerator/zcu102/tcl_scripts/axi_stream_design.tcl#L5-L7) of our ZCU102 script to point to the relevant part. I've pushed it to [this branch](https://github.com/thesps/hls4ml/tree/zcu104), and you can try it by adding the following to your config:
```
backend_config['Board'] = 'zcu104'
```
With that, our block design looks like this:
[image]

HenningCode, 2022-01-26:
Thank you first of all!

Why does your IP core not have an x_axi_AxiLiteS Port, which is connect to the AXI Interconnect?  

Now because my IP doesnt behave like I want it to, I tried to use the `build(cosim=True)` option, but this fails. Is there a reason why this fails or does the RTL simulation never work?
```
INFO: [HLS 200-10] Adding test bench file 'mnist_test_test.cpp' to the project
INFO: [COSIM 212-47] Using XSIM for RTL simulation.
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling apatb_mnist_test_axi.cpp
   Compiling mnist_test_test.cpp_pre.cpp.tb.cpp
cosim.tv.mk:61: recipe for target 'obj/mnist_test_test.cpp_pre.cpp.tb.o' failed
In file included from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/hls_fpo.h:186,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/hls_half.h:44,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/etc/ap_private.h:90,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/ap_common.h:641,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/ap_int.h:54,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/firmware/mnist_test.h:23,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/firmware/mnist_test_axi.h:5,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/mnist_test_test.cpp:28:
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:181:9: error: ‘__gmp_const’ does not name a type
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
         ^~~~~~~~~~~
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:181:48: error: expected constructor, destructor, or type conversion before ‘;’ token
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
                                                ^
In file included from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/hls_fpo.h:186,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/hls_half.h:44,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/etc/ap_private.h:90,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/ap_common.h:641,
                 from /mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/ap_int.h:54,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/firmware/mnist_test.h:23,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/firmware/mnist_test_axi.h:5,
                 from /home/henning/Schreibtisch/Masterarbeit/Vitis/Vitis-AI/Masterarbeit/build/HLS/zcu104/mnist_test_test.cpp:28:
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:244:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_version _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:245:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_patches _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:260:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char *
  ^~~~~~~~~~~
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:299:19: error: ‘mpfr_srcptr’ was not declared in this scope
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                   ^~~~~~~~~~~
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:299:42: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                          ^
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:299:54: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                      ^
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:299:66: error: expected primary-expression before ‘,’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                                  ^
/mnt/e41fa463-d10e-4d8d-ae5b-f90fb0242bd6/Xilinx/Vivado/2019.1/include/mpfr.h:299:79: error: expected primary-expression before ‘)’ token
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                                                                               ^
[... cut ...]
ERROR: [COSIM 212-5] *** C/RTL co-simulation file generation failed. ***
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total elapsed time: 305.48 seconds; peak allocated memory: 692.763 MB.
INFO: [Common 17-206] Exiting vivado_hls at Wed Jan 26 14:34:22 2022...
Found 1 solution(s) in build/HLS/zcu104/mnist_test_prj.
Reports for solution "solution1":
```

thesps, 2022-01-26:
This cosimulation issue is related to Vivado and the GMP library, see [here](https://support.xilinx.com/s/question/0D52E00006iHkfp/vivado-20153-hls-bug-gmph?language=en_US) for example.

HenningCode, 2022-01-27:
I got it to work now! Thank you very much! 

I am just wondering why I need the double sized input for my IP to generate an output. I am trying to create an mnist classifier, which uses pictures of size (28,28). My input should than only require 784 input features, but I need to input 1568 features. 

I looked at this problem with the RTL testbench and saw, that after tlast there are getting the same size 0 send again and right in the middle again. Why is that? 

How do I need to Input my Data into the IP to get the correct amount of Output data? 

[image]

thesps, 2022-02-01:
What data type do you have on the interface? I would guess it could be something like a mismatch between the expected and actual data size in bytes. From the screen shot it looks like the data signal is 32 bits but only 16 of them are set. 

When you send two images and then receive one output, does that output make any sense compared to the original NN or is it garbage?

HenningCode, 2022-02-01:
> What data type do you have on the interface? I would guess it could be something like a mismatch between the expected and actual data size in bytes. From the screen shot it looks like the data signal is 32 bits but only 16 of them are set.

Yes. Only the top 16 Bit are set and my Input is expected to be `typedef nnet::array<ap_fixed<16,6>, 1*1> input_t;`

I dont quite know what understand the question about the Interface, because I havent worked much with HLS, but these are my Interface pragmas:
```
    #pragma HLS INTERFACE axis port=in
    #pragma HLS INTERFACE axis port=out

   #pragma HLS INTERFACE axis port=input_1_m,layer11_out 
```
> When you send two images and then receive one output, does that output make any sense compared to the original NN or is it garbage? 

Not quite, but thats also a bit weird. I need to input an array and cant input a single input. So my input has to be: `x_data[0:1]` if I want to input a single picture. `x_data[0]` does not work.

HenningCode, 2022-03-27:
> This cosimulation issue is related to Vivado and the GMP library, see [here](https://support.xilinx.com/s/question/0D52E00006iHkfp/vivado-20153-hls-bug-gmph?language=en_US) for example.

@thesps Is it possible to run the `design.tcl` after I have to manually add this line to my HLS Project ?
```
    #include <gmp.h>
    #define __gmp_const const
```

HenningCode, 2022-03-28:
@thesps With the Issue #513 I finally understood, how I should be working with hls4ml and I can confirm that the changes are working for the `zcu104`. 

> Did you try modifying our block diagram? We use an AXI DMA IP for the AXI4 to AXI-Stream step. Personally, I don't have experience with the AXI-Stream FIFO IP for that purpose.
>
> I made a quick [...]

closed 2022-03-28 by HenningCode

gioregg, 2024-09-25:
Hi @HenningCode ,

Can you please share how to read the data into the ip through the axi stream?

thank you,

Gio