fastmachinelearning/hls4ml #497: BatchNormalization scale and bias array size

jmitrevs, 2022-02-17:
The `normalize` function call in both `nnet_batchnorm.h` and `nnet_batchnorm_stream.h` has an interface of:

```c++
template<class data_T, class res_T, typename CONFIG_T>
void normalize(
    data_T    data[CONFIG_T::n_in],
    res_T     res[CONFIG_T::n_in],
    typename CONFIG_T::scale_t  scale[CONFIG_T::n_in],
    typename CONFIG_T::bias_t   bias[CONFIG_T::n_in]
)
```
However, in the case when `CONFIG_T::n_filt == -1`, only the first `CONFIG_T::n_filt` of the arrays is used. This value is usually much smaller. The code should be updated to reflect this (potentially by me).

Two options that I can think of are to either introduce another constant, say `CONFIG::n_scale_bias`, that is set appropriately to the size of the scale and bias, or to have separate `normailze` variants depending on the value of `CONFIG_T::n_filt`.

vloncar, 2022-02-17:
`CONFIG::n_scale_bias` approach seems far simpler, but are we sure we won't have to split the functions anyway for optimal QoR?

jmitrevs, 2022-07-25:
This was merged on the Vivado side but not on the Quartus side. Leaving it open till it's fixed there.

jmitrevs, 2023-05-12:
This was fixed in #749 for Quartus, so closing.

closed 2023-05-12 by jmitrevs