fastmachinelearning/hls4ml #502: Conv1d compile error. "‘conv_1d_cf’ is not a member of ‘nnet’; did you mean ‘conv_1d_cl’?"

YHRen, 2022-03-10:
TLDR: compile a single layer conv1d model from onnx model produced the following error. 

```
firmware/myproject.cpp: In function ‘void myproject(input_t*, result_t*, short unsigned int&, short unsigned int&)’:
firmware/myproject.cpp:56:11: error: ‘conv_1d_cf’ is not a member of ‘nnet’; did you mean ‘conv_1d_cl’?
   56 |     nnet::conv_1d_cf<input_t, result_t, config2>(input, layer2_out, w2, b2); // Conv_0
      |           ^~~~~~~~~~
      |           conv_1d_cl
```

Here is the code to reproduce.
```python
import torch
import torch.nn as nn
import onnx
import hls4ml

for name in ["torch","onnx","hls4ml"]:
    print(name,": ",eval(name+".__version__"))

x = torch.tensor([0,1,0,0,0]).reshape((1,1,-1)).float()
net = nn.Conv1d(1,2,3) # in=1, out=2, ksz=3
y = net(x) # test

torch.onnx.export(net,x,"conv1d.onnx",input_names=["input", "conv1d"], output_names=["output"])
model = onnx.load('conv1d.onnx')
config = hls4ml.utils.config_from_onnx_model(model)
hls_model = hls4ml.converters.convert_from_onnx_model(model, hls_config=config)
hls_model.compile()
```

Output: 
```
torch: 1.10.2+cpu
onnx: 1.11.0
hls4ml: 0.6.0
Interpreting Model ...
Output layers:  ['Conv_0']
Input shape: [None, 1, 5]
Topology:
Layer name: Conv_0, layer type: Conv1D, current shape: [[None, 1, 5]]
Creating HLS model
Writing HLS project
Done
firmware/myproject.cpp: In function ‘void myproject(input_t*, result_t*, short unsigned int&, short unsigned int&)’:
firmware/myproject.cpp:56:11: error: ‘conv_1d_cf’ is not a member of ‘nnet’; did you mean ‘conv_1d_cl’?
   56 |     nnet::conv_1d_cf<input_t, result_t, config2>(input, layer2_out, w2, b2); // Conv_0
      |           ^~~~~~~~~~
      |           conv_1d_cl
firmware/myproject.cpp:56:29: error: expected primary-expression before ‘,’ token
   56 |     nnet::conv_1d_cf<input_t, result_t, config2>(input, layer2_out, w2, b2); // Conv_0
      |                             ^
firmware/myproject.cpp:56:39: error: expected primary-expression before ‘,’ token
   56 |     nnet::conv_1d_cf<input_t, result_t, config2>(input, layer2_out, w2, b2); // Conv_0
      |                                       ^
firmware/myproject.cpp:56:48: error: expected primary-expression before ‘>’ token
   56 |     nnet::conv_1d_cf<input_t, result_t, config2>(input, layer2_out, w2, b2); // Conv_0
      |                                                ^
/usr/bin/ld: cannot find myproject.o: No such file or directory
collect2: error: ld returned 1 exit status
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
Input In [127], in <cell line: 17>()
     15 config = hls4ml.utils.config_from_onnx_model(model)
     16 hls_model = hls4ml.converters.convert_from_onnx_model(model, hls_config=config)
---> 17 hls_model.compile()

File ~/anaconda3/envs/hls/lib/python3.9/site-packages/hls4ml/model/hls_model.py:546, in HLSModel.compile(self)
    544         dlclose_func.restype = ctypes.c_int
    545         dlclose_func(self._top_function_lib._handle)
--> 546     self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)
    547 finally:
    548     os.chdir(curr_dir)

File ~/anaconda3/envs/hls/lib/python3.9/ctypes/__init__.py:460, in LibraryLoader.LoadLibrary(self, name)
    459 def LoadLibrary(self, name):
--> 460     return self._dlltype(name)

File ~/anaconda3/envs/hls/lib/python3.9/ctypes/__init__.py:382, in CDLL.__init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    379 self._FuncPtr = _FuncPtr
    381 if handle is None:
--> 382     self._handle = _dlopen(self._name, mode)
    383 else:
    384     self._handle = handle

OSError: firmware/myproject-8f439B93.so: cannot open shared object file: No such file or directory
```
Thank you for any suggestions. 

<details><summary> Other things I have tried</summary>

I also tried convert from pytorch model directly, which resulted in a different error. 
I also tried to use an example keras conv1d model, which resulted in the following error.

```
In [118]: keras_config  = hls4ml.utils.fetch_example_model('KERAS_conv1d.json')
     ...:
Downloading example model files ...

In [119]: keras_model = hls4ml.converters.keras_to_hls(config)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [119], in <cell line: 1>()
----> 1 keras_model = hls4ml.converters.keras_to_hls(config)

File ~/anaconda3/envs/hls/lib/python3.9/site-packages/hls4ml/converters/keras_to_hls.py:226, in keras_to_hls(config)
    224     reader = KerasFileReader(config)
    225 else:
--> 226     raise ValueError('No model found in config file.')
    228 #print(model_arch)
    229
    230 #Define layers to skip for conversion to HLS
    231 skip_layers = ['Dropout']

ValueError: No model found in config file.
```
</details>

jmduarte, 2022-06-19:
`conv_1d_cf` (channels first) is currently not supported, only `conv_1d_cl` (channels last) is supported. 

In the future, we have planned an optimization that converts the model from channels first to channels last.

This is in the `qonnx` library: https://github.com/fastmachinelearning/qonnx, which is described here: https://arxiv.org/abs/2206.07527

closed 2022-06-19 by jmduarte

jmitrevs, 2022-10-11:
HI,

Channels-first support (the default in ONNX, the “cf” in the error message) is a bit spotty, and our plan is to transform the representation from channels-first to channels-last before processing. The transformation of cf to cl is implemented in this package:

https://github.com/fastmachinelearning/qonnx <https://github.com/fastmachinelearning/qonnx>

However, this is still a work in progress, and the qonnx parser for this dialect of QONNX is in a feature branch, not  merged into master yet.

Jovan