fastmachinelearning/hls4ml #504: Cannot synthesizing large models

MehdiRh17, 2022-03-12:
Hello. I am trying to use hls4ml to convert the codes into HLS and then synthesize it. Although the hls4ml works for small models, I get this error when I increase the model size for example for mnist:

Stop unrolling loop ........ because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.

I really appreciate if you help me what is the problem. Thanks

TitechTraj, 2022-03-15:
I ran into the same issue. My model still does not synthesize completely but I am not getting that error anymore.
Try to change the Model Strategy in the config file or settings to 'Resource' and increase the ReuseFactor (I went up to 64 to bypass that error for my case)

MehdiRh17, 2022-03-15:
Thank you for your response. Actually I solved the issue and passed the mentioned error. However, the synthesis fails again and shows this error:  line 164: 494 Killed "$RDI_PROG" "$@". I am not sure if HLS4ML can handle big models or there is a problem with my codes. Do you have any idea about this error? Thank you very much

vloncar, 2022-03-15:
hls4ml (currently) stores all weights on-chip in order to minimize the latency of the designs. Naturally, this is limiting the size of the models and as you have observed it is not possible to put any model on the FPGA. How large is a "large model" is also not easy to answer, as it depends on the architecture of the model, the precision/quantization used and pruning (if any). I would suggest you try with the smaller models of the architecture that you want (you don't need to train them), just to get an estimate of the number of parameters that you can fit.

MehdiRh17, 2022-03-16:
Thank you very much for your response. I will try to follow your instruction.

LordScarface, 2022-03-30:
I ran into the same problem numerous times, check your systems memory usage during the HLS compilation, I think the error comes from Vivado HLS running out of RAM. I solved this by incresing the ReuseFactor of the Layer that was being converted when the error occured. 

Alternatively you could try to use fewer parameters in that layer, use a system with more RAM or increse the size of the swap file, but that would probably also increase the runtime by an order of magitude. Alternatively you could try pruning but I have not tried that and do not know if that would fix it.

SteCla0, 2022-06-22:
I have the same issue and will keep an eye on RAM as suggested.
Additionally, is there any report on model size/complexity feasibility using hls4ml with certain reuse factor and for example a KU115 FPGA?

closed 2026-10-22 by github-actions[bot]