fastmachinelearning/hls4ml #512: Different HLS results for a differnet amount of training the network

HenningCode, 2022-03-25:
Hello,

I am experiencing weird results. I got the same phenomenon on different networks, but always the same procedure. If I train a network fast for only 5 epochs I get these results,
 ```
Accuracy qkeras:     0.9861
Accuracy hls4ml:     0.9859
Relative difference: 0.000202819186695039
```
but if I train the same model for 40 epochs the results change a lot:
```
Accuracy qkeras:     0.9892
Accuracy hls4ml:     0.4476
Relative difference: 0.547513141932875
```

This is the model:
```
def create_qcnn_medium(input_shape,output_shape):
    x = x_in = Input(input_shape, name='input_1_m')
    x = QConv2D(64,(3,3),(2,2),
                    kernel_quantizer=quantized_bits(8, 0, alpha=1),
                    bias_quantizer=quantized_bits(8, 0, alpha=1),
                    name='qconv2d_1_m')(x)
    x = QActivation("quantized_relu(8,4)", name='qact_1_m')(x)
    x = QConv2D(64,(3,3),
                    kernel_quantizer=quantized_bits(8, 0, alpha=1),
                    bias_quantizer=quantized_bits(8, 0, alpha=1),
                    name='qconv2d_2_m')(x)
    x = QActivation("quantized_relu(8,4)", name='qact_2_m')(x)
    x = QConv2D(128,(3,3),
                    kernel_quantizer=quantized_bits(8, 0, alpha=1),
                    bias_quantizer=quantized_bits(8, 0, alpha=1),
                    name='qconv2d_3_m')(x)
    x = QActivation("quantized_relu(8,4)", name='qact_3_m')(x)
    x = Flatten(name='flatten_1_m')(x)
    x = QDense(output_shape,
                    kernel_quantizer=quantized_bits(8, 0, alpha=1),
                    bias_quantizer=quantized_bits(8, 0, alpha=1),
                    name='qdense_4_m')(x)
    x = Activation("softmax", name='softmax')(x)
    
    model = Model(inputs=[x_in], outputs=[x])

    return model
```

I am using the latest version from the git repo with a branch for my `zcu104`. The results arent always this drastic, but I often get a decrease of around 10%. The only difference is the amount of training. My profiling results are fine to after optimization. 

I can show you my whole scripts I am using but they are quite long and I dont know if they matter much.

rtspk, 2022-03-25:
@HenningCode You mentioned 'My profiling results are fine to after optimization'. . If you don't mind, May i ask you something related to the profiling results. I have a Qkeras model, but I am not getting plots of all the layers of my Qkeras model ( after optimization), when i am trying to profile the model to drill down the cause of decrease in model's accuracy.

My issue is described here in more detail: #505 

This problem happens in the Qkeras Model. when i profile the usual keras model. it works fine. but since i want to go ahead with some quantization-aware training, i have to use Qkeras. 

So my Question is, Are you able to see the distribution of activations functions after optimization ?

HenningCode, 2022-04-06:
[image]
[image]
I Found the difference between the amount of training applied. The Softmax layer gets deeper as seen in the plots. Is there anyway to fix this?

jmduarte, 2022-06-19:
Hi @HenningCode, you can adjust the precision and look-up table used for softmax layer. See, e.g.: https://github.com/fastmachinelearning/hls4ml-tutorial/blob/master/part4_quantization.ipynb
```python
config = hls4ml.utils.config_from_keras_model(model, granularity='name')
config['LayerName']['softmax']['exp_table_t'] = 'ap_fixed<18,8>'
config['LayerName']['softmax']['inv_table_t'] = 'ap_fixed<18,4>'
config['LayerName']['softmax']['Precision']['default'] = 'ap_fixed<18,8>'
config['LayerName']['softmax']['Precision']['result'] = 'ap_fixed<18,8>'
hls_model = hls4ml.converters.convert_from_keras_model(model, hls_config=config)
```

But a simpler solution may be, just removing the softmax layer entirely. If the only thing you're interested in is the top-1 accuracy, then you can take a simple argmax of the logits (inputs to the softmax).

closed 2022-06-19 by jmduarte