fastmachinelearning/hls4ml #514: AttributeError: module 'tensorflow.keras.layers' has no attribute 'MultiHeadAttention'

GiuseppeDiGuglielmo, 2022-03-27:
Hi all,

I am setting up from scratch and running the [hls4ml tutorial](https://github.com/fastmachinelearning/hls4ml-tutorial) (main branch). I get an error that is a new one: `AttributeError: module 'tensorflow.keras.layers' has no attribute 'MultiHeadAttention'`. The problem seems related to a combination of versions of packages. In particular, I have TensorFlow (in my case, 2.3.1) and QKeras (0.9.0) as requested in [environment.yml](https://github.com/fastmachinelearning/hls4ml-tutorial/blob/master/environment.yml).

Have you seen this problem? Is there a quick workaround?

This is the full log:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/tmp/ipykernel_14566/3564380083.py in <module>
----> 1 import hls4ml
      2 config = hls4ml.utils.config_from_keras_model(model, granularity='model')
      3 print("-----------------------------------")
      4 print("Configuration")
      5 plotting.print_dict(config)

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/__init__.py in <module>
      3 __version__ = '0.6.0'
      4 
----> 5 from hls4ml import converters
      6 from hls4ml import report
      7 from hls4ml import utils

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/converters/__init__.py in <module>
      7 from hls4ml.utils.config import create_config
      8 
----> 9 from hls4ml.converters.keras_to_hls import keras_to_hls, get_supported_keras_layers, register_keras_layer_handler
     10 
     11 #----------Make converters available if the libraries can be imported----------#

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/converters/keras_to_hls.py in <module>
      5 import math
      6 
----> 7 from hls4ml.model import HLSModel
      8 
      9 MAXMULT = 4096

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/model/__init__.py in <module>
      1 from __future__ import absolute_import
      2 
----> 3 from hls4ml.model.hls_model import HLSModel, HLSConfig
      4 
      5 try:

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/model/hls_model.py in <module>
     13 from hls4ml.templates import get_backend
     14 from hls4ml.writer import get_writer
---> 15 from hls4ml.model.optimizer import optimize_model, get_available_passes
     16 from hls4ml.report.vivado_report import parse_vivado_report
     17 

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/model/optimizer/__init__.py in <module>
     18 
     19 try:
---> 20     from hls4ml.model.optimizer.passes.qkeras import OutputRoundingSaturationMode
     21     from hls4ml.model.optimizer.passes.qkeras import QKerasFactorizeAlpha
     22     from hls4ml.model.optimizer.passes.qkeras import FuseConsecutiveBatchNormalization

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/hls4ml/model/optimizer/passes/qkeras.py in <module>
      5 import tensorflow as tf
      6 import numpy as np
----> 7 from qkeras import get_quantizer
      8 
      9 class QKerasPO2Quantizer(object):

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/qkeras/__init__.py in <module>
     21 
     22 from .b2t import *  # pylint: disable=wildcard-import
---> 23 from .estimate import *  # pylint: disable=wildcard-import
     24 from .qlayers import *  # pylint: disable=wildcard-import
     25 from .quantizers import *  # pylint: disable=wildcard-import

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/qkeras/estimate.py in <module>
     39 from tensorflow.keras.models import Model
     40 
---> 41 from .qlayers import QActivation
     42 from .qlayers import QAdaptiveActivation
     43 from .qlayers import QDense

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/qkeras/qlayers.py in <module>
     56 from .quantizers import _get_integer_bits
     57 from .quantizers import get_quantizer
---> 58 from tensorflow_model_optimization.python.core.sparsity.keras.prunable_layer import PrunableLayer
     59 
     60 

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/__init__.py in <module>
     84 from tensorflow_model_optimization.python.core import version
     85 
---> 86 from tensorflow_model_optimization.python.core.api import clustering
     87 from tensorflow_model_optimization.python.core.api import experimental
     88 from tensorflow_model_optimization.python.core.api import quantization

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/api/__init__.py in <module>
     14 # ==============================================================================
     15 """Import API modules for Tensorflow Model Optimization."""
---> 16 from tensorflow_model_optimization.python.core.api import clustering
     17 from tensorflow_model_optimization.python.core.api import experimental
     18 from tensorflow_model_optimization.python.core.api import quantization

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/api/clustering/__init__.py in <module>
     14 # ==============================================================================
     15 """Module containing code for clustering."""
---> 16 from tensorflow_model_optimization.python.core.api.clustering import keras

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/api/clustering/keras/__init__.py in <module>
     17 from tensorflow_model_optimization.python.core.clustering.keras import experimental
     18 
---> 19 from tensorflow_model_optimization.python.core.clustering.keras.cluster import cluster_scope
     20 from tensorflow_model_optimization.python.core.clustering.keras.cluster import cluster_weights
     21 from tensorflow_model_optimization.python.core.clustering.keras.cluster import strip_clustering

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/clustering/keras/cluster.py in <module>
     18 
     19 from tensorflow_model_optimization.python.core.clustering.keras import cluster_config
---> 20 from tensorflow_model_optimization.python.core.clustering.keras import cluster_wrapper
     21 from tensorflow_model_optimization.python.core.clustering.keras import clustering_centroids
     22 

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/clustering/keras/cluster_wrapper.py in <module>
     22 from tensorflow_model_optimization.python.core.clustering.keras import clusterable_layer
     23 from tensorflow_model_optimization.python.core.clustering.keras import clustering_centroids
---> 24 from tensorflow_model_optimization.python.core.clustering.keras import clustering_registry
     25 
     26 attrgetter = operator.attrgetter  # pylint: disable=invalid-name

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/clustering/keras/clustering_registry.py in <module>
     55 
     56 
---> 57 class ClusteringRegistry(object):
     58   """Registry responsible for built-in keras layers."""
     59 

~/miniconda3/envs/hls4ml-tutorial-debug/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/clustering/keras/clustering_registry.py in ClusteringRegistry()
    101 
    102   _SUPPORTED_MHA_LAYERS = {
--> 103       tf.keras.layers.MultiHeadAttention,
    104   }
    105 

AttributeError: module 'tensorflow.keras.layers' has no attribute 'MultiHeadAttention'
```

GiuseppeDiGuglielmo, 2022-03-27:
The quickest solution I could find so far is to bump the TensorFlow version from `2.3.1` to `2.4.0`, however, I am not sure if this breaks anything else. 

```
pip uninstall tensorflow
pip install tensorflow==2.4.0
```

jmduarte, 2022-03-27:
Opened https://github.com/fastmachinelearning/hls4ml-tutorial/pull/27 to update the environment, but I wonder if we should set a minimum tensorflow version also in our [setup.py](https://github.com/fastmachinelearning/hls4ml/blob/master/setup.py)?

@vloncar @thesps

vloncar, 2022-03-28:
I think we should, though I would advocate for the more recent version than 2.4. Since version [2.6](https://github.com/tensorflow/tensorflow/releases/tag/v2.6.0) keras has been split into a separate package and this causes some problems if you tinker with the tf ops at the core, like we do in one of the side projects (not easy to handle imports compatible with both approaches). The upgrade from 2.4 to 2.6 and beyond is not hard for users, since it doesn't require a bump in [CUDA version](https://www.tensorflow.org/install/source#gpu)

closed 2026-10-22 by github-actions[bot]