fastmachinelearning/hls4ml #516: Cant synthesis model with latest master branch

HenningCode, 2022-03-31:
Hello Guys, 

I got some more issues with the latest master branch. I already got the issue resolved in #515, but now I cant synthesis the model.

The first things that was weird is that the HLS Project is written twice, as seen in the following log.
```
Interpreting Model
Topology:
Layer name: input_1_m, layer type: InputLayer, input shapes: [[None, 28, 28, 1]], output shape: [None, 28, 28, 1]
Layer name: flatten_1_m, layer type: Reshape, input shapes: [[None, 28, 28, 1]], output shape: [None, 784]
Layer name: qdense_1_m, layer type: QDense, input shapes: [[None, 784]], output shape: [None, 16]
Layer name: qact_1_m, layer type: Activation, input shapes: [[None, 16]], output shape: [None, 16]
Layer name: qdense_2_m, layer type: QDense, input shapes: [[None, 16]], output shape: [None, 128]
Layer name: qact_2_m, layer type: Activation, input shapes: [[None, 128]], output shape: [None, 128]
Layer name: qdense_3_m, layer type: QDense, input shapes: [[None, 128]], output shape: [None, 10]
Layer name: softmax, layer type: Softmax, input shapes: [[None, 10]], output shape: [None, 10]
Creating HLS model
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_1_m". Using ReuseFactor=12544 instead. Valid ReuseFactor(s): 1,2,4,7,8,14,16,28,49,56,98,112,196,392,784,1568,3136,6272,12544.
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_2_m". Using ReuseFactor=2048 instead. Valid ReuseFactor(s): 1,2,4,8,16,32,64,128,256,512,1024,2048.
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_3_m". Using ReuseFactor=1280 instead. Valid ReuseFactor(s): 1,2,4,8,16,32,64,128,256,640,1280.
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_1_m". Using ReuseFactor=12544 instead. Valid ReuseFactor(s): 1,2,4,7,8,14,16,28,49,56,98,112,196,392,784,1568,3136,6272,12544.
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_2_m". Using ReuseFactor=2048 instead. Valid ReuseFactor(s): 1,2,4,8,16,32,64,128,256,512,1024,2048.
WARNING: Invalid ReuseFactor=2000000 in layer "qdense_3_m". Using ReuseFactor=1280 instead. Valid ReuseFactor(s): 1,2,4,8,16,32,64,128,256,640,1280.
Writing HLS project
Done
Writing HLS project
Done
```

And then if I am trying to build the Project this is the Error I am getting. 

```
ERROR: [HLS 200-70] Part 'None' is not installed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
```

I am targeting a `zcu104` and I know it isnt supported but I used the added board files from @thesps and they worked for me in the past. I also tried to target a `pynq`, but that results into the same error message. 

Any help would be appreciated.

HenningCode, 2022-04-04:
This is my Config:

```
HLSConfig
  LayerName
    input_1_m
      Precision
        result:      ap_fixed<16,6>
    qact_1_m
      Precision
        result:      ap_ufixed<8,4>
      ReuseFactor:   2000000
      Trace:         True
    qact_2_m
      Precision
        result:      ap_ufixed<8,4>
      ReuseFactor:   2000000
      Trace:         True
    qdense_1_m
      Precision
        bias:        ap_fixed<8,3>
        weight:      ap_fixed<8,3>
      ReuseFactor:   2000000
      Trace:         True
    qdense_2_m
      Precision
        bias:        ap_fixed<8,3>
        weight:      ap_fixed<8,3>
      ReuseFactor:   2000000
      Trace:         True
    qdense_3_m
      Precision
        bias:        ap_fixed<8,3>
        weight:      ap_fixed<8,3>
      ReuseFactor:   2000000
      Trace:         True
    softmax
      Precision:     ap_fixed<16,3>
      ReuseFactor:   2000000
      Trace:         True
      exp_table_t:   ap_fixed<18,8,AP_RND,AP_SAT>
      inv_table_t:   ap_fixed<18,8,AP_RND,AP_SAT>
      table_size:    1024
  Model
    Precision:       ap_fixed<16,6>
    ReuseFactor:     2000000
    Strategy:        Resource
Backend:             VivadoAccelerator
ClockPeriod:         10
IOType:              io_stream
OutputDir:           final_training/HLS/fc_small
ProjectName:         fc_small
XilinxPart:          xczu7ev-ffvc1156-2-e
Board:               zcu104
Interface:           axi_stream
InputData:           input_features.dat
OutputPredictions:   output_predictions.dat
```

Do I need to rename anything maybe ?

thesps, 2022-04-04:
I haven't tried out your code, but from your message, I think renaming `XilinxPart` to `Part` should fix it. The master branch is undergoing many changes, including to the API, so just be aware that the situation might change again before the next release.

HenningCode, 2022-04-04:
Thank you. The only thing unclear atm is how do I set the `board` aswell? I saw there is a new AcceleratorConfig? 

And the Problem with double writing the Writing HLS project still exists?