fastmachinelearning/hls4ml #517: Upsampling2D support

ChiRuiChen, 2022-04-02:
Hi all,
So I know that although the UpSampling2D Layer is in the [layer map](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/model/layers.py)

It is not included in the supported layers.
So I went through the nnet::image_stream.h file and realized that the resizing method is not corresponding to the Keras UpSampling2D method.

For example, provided a 2*2 array:
[[1 2 ]
[3 4 ]]
using UpSampling2D((2,2)) should get
[[1 1 2 2 ]
[1 1 2 2 ]
[3 3 4 4 ]
[3 3 4 4 ]]

but in hls4ml it became 
[[1 1 1 1 ]
[2 2 2 2 ]
[3 3 3 3 ]
[4 4 4 4 ]]

So I had rewritten the code to fix this problem.
[nnet_image_stream.h](https://github.com/ChiRuiChen/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_image_stream.h)

Hoping this can be tested and added to the branch.

vloncar, 2022-04-03:
Hi @ChiRuiChen, thanks for the implementation. Can you create a pull request with a test case and some synthesis results?

ChiRuiChen, 2022-04-04:
@vloncar  my pleasure
I had created a pull request and an autoencoder test case.

jmduarte, 2022-05-11:
Closed by #520

closed 2022-05-11 by jmduarte