fastmachinelearning/hls4ml #530: Issue with predict and HLS Compilation

LordScarface, 2022-04-25:
Hi, 

I'm trying to implement [this model](https://github.com/Xilinx/Vitis-AI-Tutorials/blob/master/Design_Tutorials/10-RF_modulation_recognition/files/vai_2018_RadioML_VAI_keras.ipynb) using hls4ml. The model conversion seems to be working but when calling `model.predict()` with the test data I get:
```
python3: firmware/nnet_utils/nnet_conv2d_stream.h:73: void nnet::conv_2d_buffer_cl(hls::stream<srcType>&, 
hls::stream<dstType>&, typename CONFIG_T::weight_t*, typename CONFIG_T::bias_t*) 
[with data_T = nnet::array<ap_fixed<16, 6>, 2>; res_T = nnet::array<ap_fixed<16, 6>, 32>; CONFIG_T = config2; typename CONFIG_T::weight_t = ap_fixed<16, 6>; 
typename CONFIG_T::bias_t = ap_fixed<16, 6>]: 
Assertion `CONFIG_T::pad_top == 0 && CONFIG_T::pad_bottom == 0 && CONFIG_T::pad_left == 0 && CONFIG_T::pad_right == 0' failed.
Aborted (core dumped)
```

When I open the generated HLS Project, it looks fine and when using `backend='VivadoAccelerator'` the HLS Compilation actually completes without any errors, however it only takes a minute, Resource Usage is around 0% and the summary says `?` for Latency and Interval. I assume maybe only the `myproject_axi()` wrapper was processed?

When changing the backend to `backend='Vivado'` and running the HLS Compilation, it stops with the error:
```
ERROR: [HLS 200-474] Empty dataflow region in myproject (it may have been optimized away due to the absence of outputs)
ERROR: [HLS 200-70] Failed building synthesis data model.
command 'ap_source' returned error code
    while executing
"source /home/lukas/Documents/HiWi/tmpp/myproject_prj/solution1/csynth.tcl"
    invoked from within
"hls::main /home/lukas/Documents/HiWi/tmpp/myproject_prj/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.
```

So I assume there is something wrong with the code generated by hls4ml? 

I'm using version 0.6.0 of hls4ml, but I get the same issue with the master branch and Version 0.5.0. 

I hope someone can help me out or point me in the right direction, thank you in advance,
Best Regards

<details>
<summary>My Code used for generating the HLS Project</summary>

```python
from yaml import load_all
import tensorflow as tf
from tensorflow import keras
from tensorflow import Tensor
print("Tensorflow version is ", tf.__version__)
print('Keras version      : ',keras.__version__)
import numpy as np
import os, sys
from tensorflow.keras.models import Model, load_model
import hls4ml
from qkeras.utils import _add_supported_quantized_objects
from sklearn.metrics import accuracy_score

co = {}
_add_supported_quantized_objects(co)

model = load_model('./fp_model/resnet_fp_model.h5', custom_objects=co)

with open('./Dataset/X_train.npy', 'rb') as f:
    X_train = np.load(f)
with open('./Dataset/X_test.npy', 'rb') as f:
    X_test = np.load(f)
with open('./Dataset/Y_test.npy', 'rb') as f:
    Y_test = np.load(f)
with open('./Dataset/Y_train.npy', 'rb') as f:
    Y_train = np.load(f)

[... 262 lines cut ...]
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vivado/2019.1/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vivado/2019.1/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'lukas' on host 'ubuntu' (Linux_x86_64 version 5.13.0-40-generic) on Mon Apr 25 11:03:05 CEST 2022
INFO: [HLS 200-10] On os Ubuntu 20.04.2 LTS
INFO: [HLS 200-10] In directory '/home/lukas/Documents/HiWi/tmppp'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Creating and opening project '/home/lukas/Documents/HiWi/tmppp/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/lukas/Documents/HiWi/tmppp/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-10] Setting target device to 'xczu7ev-ffvc1156-2-e'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [Common 17-206] Exiting vivado_hls at Mon Apr 25 11:03:05 2022...
Synthesis report not found.
python3: firmware/nnet_utils/nnet_conv2d_stream.h:73: void nnet::conv_2d_buffer_cl(hls::stream<srcType>&, hls::stream<dstType>&, typename CONFIG_T::weight_t*, typename CONFIG_T::bias_t*) [with data_T = nnet::array<ap_fixed<16, 6>, 2>; res_T = nnet::array<ap_fixed<16, 6>, 32>; CONFIG_T = config2; typename CONFIG_T::weight_t = ap_fixed<16, 6>; typename CONFIG_T::bias_t = ap_fixed<16, 6>]: Assertion `CONFIG_T::pad_top == 0 && CONFIG_T::pad_bottom == 0 && CONFIG_T::pad_left == 0 && CONFIG_T::pad_right == 0' failed.
Aborted (core dumped)
```

</details>

lloo099, 2022-05-02:
Hi @LordScarface , thanks for your post. Do you have padding in your conv2d in the Resnet model? Would you mind provide your model file also, thanks

LordScarface, 2022-05-02:
Hi and thank you for the reply!

Yes, the Conv2D Layers have the padding set to `'same'`

<details>
  <summary>here is the Code used for generating the model</summary>
  
```python
def resnet_block(input_data, filters, conv_size):
  x = Conv2D(filters, 1, activation=None, padding='same')(input_data)
  x = BatchNormalization()(x)
  x = Activation('relu')(x)
  x = Conv2D(filters, conv_size, activation=None, padding='same')(x)
  x = BatchNormalization()(x)
  x = Activation('relu')(x)
  x = Conv2D(filters, conv_size, activation=None, padding='same')(x)
  x = BatchNormalization()(x)
  x = Add()([x, input_data])
  x = Activation('relu')(x)
    
  y = Conv2D(filters, conv_size, activation=None, padding='same')(x)
  y = BatchNormalization()(y)
  y = Activation('relu')(y)
  y = Conv2D(filters, conv_size, activation=None, padding='same')(y)
  y = BatchNormalization()(y)    
  y = Add()([y, x])
  y = Activation('relu')(y)
  
  z = MaxPooling2D(2, strides = (2,1), padding = 'same') (y)
  return z

num_resnet_blocks = 4
num_filters = 32
kernel_size = 5,1

rf_input = Input(shape=input_shp, name = 'rf_input')

x = Conv2D(num_filters, (kernel_size), activation=None, padding='same')(rf_input)
x = BatchNormalization()(x)
x = Activation('relu')(x)

for i in range(num_resnet_blocks):
    x = resnet_block(x, num_filters, (kernel_size))

x = Conv2D(num_filters, (kernel_size), activation=None, padding = 'same')(x)
x = BatchNormalization()(x)
x = Activation('relu')(x)

# use if number of resnet blocks = 6
#x = Reshape((4,4,num_filters), input_shape = (16,1,num_filters)) (x)

# use if number of resent blocks = 4
x = Reshape((8,8,num_filters), input_shape = (32,1,num_filters)) (x)

x = GlobalAveragePooling2D()(x)
dense_1 = Dense(256, activation='relu')(x)
dropout_1 = Dropout(0.5)(dense_1)
dense_2 = Dense(128, activation='relu')(dropout_1)
dropout_2 = Dropout(0.5)(dense_2)
dense_3 = Dense(num_classes)(dropout_2)          
softmax = Activation('softmax', name = 'softmax')(dense_3)

optimizer= Adam(learning_rate=0.00050)
model = keras.Model(rf_input, softmax)
model.compile(loss='categorical_crossentropy', metrics=["accuracy"])
```

</details>

<details>
[... 200 lines cut ...]
                                                                                                  
 dense (Dense)                  (None, 256)          8448        ['global_average_pooling2d[0][0]'
                                                                 ]                                
                                                                                                  
 dropout (Dropout)              (None, 256)          0           ['dense[0][0]']                  
                                                                                                  
 dense_1 (Dense)                (None, 128)          32896       ['dropout[0][0]']                
                                                                                                  
 dropout_1 (Dropout)            (None, 128)          0           ['dense_1[0][0]']                
                                                                                                  
 dense_2 (Dense)                (None, 22)           2838        ['dropout_1[0][0]']              
                                                                                                  
 softmax (Activation)           (None, 22)           0           ['dense_2[0][0]']                
                                                                                                  
==================================================================================================
Total params: 139,158
Trainable params: 137,750
Non-trainable params: 1,408
__________________________________________________________________________________________________
None
```

</details>

[This](https://github.com/fastmachinelearning/hls4ml/files/8601796/resnet_fp_model.zip) is the trained model (trained only on the first 200k Samples of the Dataset)

lloo099, 2022-05-04:
Thanks, I am not sure the 'same' padding means 0 padding in your models. Normally, conv2d of hls4ml has zero padding but can configure it. I suggest you test a small model at first.

LordScarface, 2022-05-05:
Thank you for the input, I switched the model to use `Conv1D` instead of `Conv2D` but the issue remained. I was able to fix it by removing the `'same'` padding from the `MaxPooling2D` (now `MaxPooling1D`) Layers of the ResNet Blocks. Maybe there is an issue with `'same'` padding for the pooling Layers? 

Anyway, model passes synthesis now

lloo099, 2022-05-06:
> 

As far as I know, the resource strategy is not available now. In this way, it mainly supports the latency strategy. If you want to get a good balance between latency and resources. I suggest that you may improve the coding from the hardware level. Hls4ml is more friendly with lightweight models.

LordScarface, 2022-05-06:
The Resource Strategy has worked for me in the past, I saw that in #534 you were able to get past synthesis with the VGG-16 model, when choosing the Latency Strategy I get issues with Layers containing more that 4096 parameters, how did you adress that? 

My goal right now is to implement the model fully parallel without any optimizations and then to see how much things can be improved by quantization, pruning, etc.

lloo099, 2022-05-06:
> The Resource Strategy has worked for me in the past, I saw that in #534 you were able to get past synthesis with the VGG-16 model, when choosing the Latency Strategy I get issues with Layers [...]

Nice and u can share your resource strategy for ResNet if it's possible. Because it's too much array partition [Here](https://github.com/fastmachinelearning/hls4ml/blob/c96a7bc99b64b3549779c01347b0aa0b5fa7c1f4/hls4ml/templates/vivado/nnet_utils/nnet_dense_latency.h#L53). You may consider to modify it or just comment.

LordScarface, 2022-05-12:
For the Resource Strategy I just replaced this:
```python
hls_config['Model']['Precision'] = 'ap_fixed<16,6>'
hls_config['Model']['ReuseFactor'] = 1

for Layer in hls_config['LayerName'].keys():
    hls_config['LayerName'][Layer]['Strategy'] = 'Latency'
    hls_config['LayerName'][Layer]['ReuseFactor'] = 1
#If you want best numerical performance for high-accuray models, while the default latency strategy is faster but numerically more unstable
hls_config['LayerName']['softmax']['Strategy'] = 'Stable'
```
with this:
```python
hls_config['Model']['Strategy'] = 'Resource'
hls_config['LayerName']['softmax']['Strategy'] = 'Stable'
hls_config['LayerName']['dense_28']['ReuseFactor'] = 16
```

I tried the Latency Strategy with commenting `#pragma HLS ARRAY_PARTITION variable=mult complete` as you suggested, but it did not help me, now I keep running out of memory during synthesis with `$RDI PROG" "@$`, is this something you have experienced when working with larger models? Are 48 GB of RAM not enough?

lloo099, 2022-05-12:
> For the Resource Strategy I just replaced this:
>
> ```python
> hls_config['Model']['Precision'] = 'ap_fixed<16,6>'
> hls_config['Model']['ReuseFactor'] = 1
>
> for Layer in [...]

Ok, thanks. I comment this because the size of layer is over 4096 but not sure your problems. Yes, I try to compile VGG-16 which encounter same memory problem. So I think we should reduce precision to 8bits. In this way, it is possible to deploy.

lloo099, 2022-05-13:
> For the Resource Strategy I just replaced this:
>
> ```python
> hls_config['Model']['Precision'] = 'ap_fixed<16,6>'
> hls_config['Model']['ReuseFactor'] = 1
>
> for Layer in [...]

Btw, you can also try vitis_hls 2020.2.

wilfredkisku, 2022-07-22:
Hi, I am also trying to work with the `ResNet` architecture and `Inception` architectures. While the **unquantized models** synthesize well with the `hls4ml`, there is an accuracy drop for **quantized models** for the above-mentioned architectures. It would be of great help if there be any insights into the issue that I am facing. #587