fastmachinelearning/hls4ml #535: MobileNet V2 Convertion

lloo099, 2022-05-02:
Hi, thanks for your great work on HLS art. Currently, I want to use your lib to implement the MobilnetV2 model. But some problems happened here, would you mind helping me with this? Thanks
Model from [Here](https://github.com/a1pha/smoking_mobilenet)

- My code:
```
from yaml import load_all
import tensorflow as tf
from tensorflow import keras
from tensorflow import Tensor
print("Tensorflow version is ", tf.__version__)
print('Keras version      : ',keras.__version__)
import numpy as np
import os, sys
from tensorflow.keras.models import Model, load_model
import hls4ml
from qkeras.utils import _add_supported_quantized_objects
from sklearn.metrics import accuracy_score

co = {}
_add_supported_quantized_objects(co)

model = load_model('mobilenetv2.h5', custom_objects=co)

hls_config = hls4ml.utils.config_from_keras_model(model, granularity='name')

hls_config['Model']['Precision'] = 'ap_fixed<16,6>'
hls_config['Model']['ReuseFactor'] = 1

for Layer in hls_config['LayerName'].keys():
    hls_config['LayerName'][Layer]['Strategy'] = 'Latency'
    hls_config['LayerName'][Layer]['ReuseFactor'] = 1
#If you want best numerical performance for high-accuray models, while the default latency strategy is faster but numerically more unstable
#hls_config['LayerName']['softmax']['Strategy'] = 'Stable'

hls_model = hls4ml.converters.convert_from_keras_model(model,
                                                       hls_config=hls_config,
                                                       io_type='io_stream',
                                                       backend='Vivado',
                                                       output_dir='tmpp/',
                                                       part='xczu7ev-ffvc1156-2-e')

hls_model.compile()

os.environ['PATH'] = '/tools/Xilinx/Vivado/2019.1/bin:' + os.environ['PATH']
hls_model.build(csim=False, synth=False, vsynth=False)

Y_pred = hls_model.predict(np.ascontiguousarray(X_test))
y_pred = np.argmax(Y_pred, axis = 1)
y_actual = np.argmax(Y_test, axis = 1)

accuracy = accuracy_score(y_actual, y_pred)

print("Accuracy: ", accuracy)

```

```
Tensorflow version is  2.8.0
Keras version      :  2.8.0
2022-05-02 10:57:51.696013: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:936] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2022-05-02 10:57:51.699062: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudnn.so.8'; dlerror: libcudnn.so.8: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /usr/local/cuda/lib64:
2022-05-02 10:57:51.699079: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1850] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.
Skipping registering GPU devices...
2022-05-02 10:57:51.699208: I tensorflow/core/platform/cpu_feature_guard.cc:151] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
Interpreting Model
Topology:
Layer name: input_1, layer type: Input
Layer name: Conv1_pad, layer type: ZeroPadding2D
Layer name: Conv1, layer type: Conv2D
[... 201 lines cut ...]
  -> Activation (linear), layer name: Conv_1
Layer name: Conv_1_bn, layer type: BatchNormalization
Layer name: out_relu, layer type: ReLU
Layer name: global_average_pooling2d, layer type: GlobalAveragePooling2D
Layer name: dense, layer type: Dense
  -> Activation (linear), layer name: dense
Layer name: batch_normalization, layer type: BatchNormalization
Layer name: activation, layer type: Activation
Layer name: dense_1, layer type: Dense
  -> Activation (softmax), layer name: dense_1
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, input shapes: [[None, None, None, 3]], output shape: [None, None, None, 3]
Traceback (most recent call last):
  File "example.py", line 32, in <module>
    hls_model = hls4ml.converters.convert_from_keras_model(model,
  File "/home/jjcc/Desktop/2022_project/hls4ml/hls4ml/converters/__init__.py", line 217, in convert_from_keras_model
    return keras_to_hls(config)
  File "/home/jjcc/Desktop/2022_project/hls4ml/hls4ml/converters/keras_to_hls.py", line 309, in keras_to_hls
    layer, output_shape = layer_handlers[keras_class](keras_layer, input_names, input_shapes, reader, config)
  File "/home/jjcc/Desktop/2022_project/hls4ml/hls4ml/converters/keras/reshaping.py", line 80, in parse_zeropadding2d_layer
    layer['pad_top'] + input_shapes[0][1] + layer['pad_bottom'], # Height
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'

```

wilfredkisku, 2022-05-08:
What I am ending up with is an Unsupported layer type error.

```
Interpreting Model
Topology:
Layer name: input_1, layer type: Input

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Input In [5], in <cell line: 21>()
     15 _add_supported_quantized_objects(co)
     17 model = load_model('mobilenetv2.h5', custom_objects=co)
---> 21 hls_config = hls4ml.utils.config_from_keras_model(model, granularity='name')
     23 hls_config['Model']['Precision'] = 'ap_fixed<16,6>'
     24 hls_config['Model']['ReuseFactor'] = 1

File ~/Vivado/hls4ml-tutorial/venv_hls4ml/lib/python3.8/site-packages/hls4ml/utils/config.py:136, in config_from_keras_model(model, granularity, default_precision, default_reuse_factor)
    134 for keras_layer in keras_layer_config:
    135     if keras_layer['class_name'] not in supported_layers:
--> 136         raise Exception('ERROR: Unsupported layer type: {}'.format(keras_layer['class_name']))
    137     if keras_layer['class_name'] in skip_layers:
    138         continue

Exception: ERROR: Unsupported layer type: ZeroPadding2D
```
Any headway? Also, I wanted to ask how do you handle issue of over utilization of resources for model that are synthesized for ZU104(xczu7ev-ffvc1156-2-e) that you have been using.

jmduarte, 2022-05-11:
Hi @lloo099, what version of hls4ml are you using? Which one of these is `None`? 

```python
layer['pad_top'] + input_shapes[0][1] + layer['pad_bottom']
```

I'd also warn that this model is quite large and may need to be compressed quite a bit to be usable. It may be better for debugging purposes if you could reproduce the error with a minimal example (say 1 or 2 layers).

Hi @wilfredkisku what version of hls4ml are you using? I believe ZeroPadding2D is supported since #480

wilfredkisku, 2022-05-11:
Thank you @jmduarte for the clarification, I am using the current version (v0.6.0). I also wanted to ask regarding the upper bound for a model to be usable for a board such as ZCU104.

Thank for the help.

lloo099, 2022-05-11:
> Here

Hi @jmduarte @wilfredkisku , thanks for your response. I am using hls4ml 0.6.0. I just direct convert [MobilnetV2](https://github.com/a1pha/smoking_mobilenet) this model on hls4ml but not successfully. I think it could be better for a fix precision of 8-bit.