fastmachinelearning/hls4ml #539: Issue while building a model with CNN model using Vivado HLS 2018.2, 2019.2 terminates with ("$RDI PROG" "@$")

wilfredkisku, 2022-05-05:
INFO: [XFORM 203-101] Partitioning array 'shift_buffer.V' (firmware/nnet_utils/nnet_conv_stream.h:217) in dimension 2 completely.
INFO: [XFORM 203-101] Partitioning array 'shift_buffer.V' (firmware/nnet_utils/nnet_conv_stream.h:217) in dimension 2 completely.
INFO: [XFORM 203-101] Partitioning array 'shift_buffer.V' (firmware/nnet_utils/nnet_conv_stream.h:217) in dimension 2 completely.
INFO: [XFORM 203-101] Partitioning array 'shift_buffer.V' (firmware/nnet_utils/nnet_conv_stream.h:217) in dimension 2 completely.
INFO: [XFORM 203-101] Partitioning array 'shift_buffer.V' (firmware/nnet_utils/nnet_conv_stream.h:217) in dimension 2 completely.

/tools/Xilinx/Vivado/2018.2/bin/rdiArgs.sh: line 194:  8288 Killed                  "$RDI_PROG" "$@"
Synthesis report not found.

**The lines above show the gist of the log (final lines) before terminating.
The model has ~14K parameters**

 Model: "keras_baseline"
_________________________________________________________________
 Layer (type)                Output Shape              Param #   
=================================================================
 input_1 (InputLayer)        [(None, 32, 32, 3)]       0                                                                   
 conv_0 (Conv2D)             (None, 30, 30, 16)        432                                                                     
 bn_conv_0 (BatchNormalizati  (None, 30, 30, 16)       64        
 on)                                                                                                                          
 conv_act_0 (Activation)     (None, 30, 30, 16)        0                                                                  
 pool_0 (MaxPooling2D)       (None, 15, 15, 16)        0                                                                 
 conv_1 (Conv2D)             (None, 13, 13, 16)        2304                                                           
 bn_conv_1 (BatchNormalizati  (None, 13, 13, 16)       64        
 on)                                                                                                         
 conv_act_1 (Activation)     (None, 13, 13, 16)        0                                                                          
 pool_1 (MaxPooling2D)       (None, 6, 6, 16)          0                                                                          
 conv_2 (Conv2D)             (None, 4, 4, 24)          3456                                                                       
 bn_conv_2 (BatchNormalizati  (None, 4, 4, 24)         96        
 on)                                                                                                                              
 conv_act_2 (Activation)     (None, 4, 4, 24)          0                                                                          
 pool_2 (MaxPooling2D)       (None, 2, 2, 24)          0                                                                          
 flatten (Flatten)           (None, 96)                0                                                                         
 dense_0 (Dense)             (None, 42)                4032                                                                       
 bn_dense_0 (BatchNormalizat  (None, 42)               168       
 ion)                                                                                                                            
 dense_act_0 (Activation)    (None, 42)                0                                                                         
 dense_1 (Dense)             (None, 64)                2688                                                                       
 bn_dense_1 (BatchNormalizat  (None, 64)               256       
 ion)                                                                                                                             
 dense_act_1 (Activation)    (None, 64)                0                                                                          
 output_dense (Dense)        (None, 10)                650                                                                       
 output_softmax (Activation)  (None, 10)               0                                                                         
=================================================================
Total params: 14,210
Trainable params: 13,886
Non-trainable params: 324
_________________________________________________________________

wilfredkisku, 2022-05-05:
I have worked with Vivado 2020.1 too, which too results in the same error. 

Please help me with a solution to this issue.

jmduarte, 2022-05-05:
Which board are you targeting?

The `Killed` line seems to mean that the Linux kernel killed the process for taking too much memory (out of memory killer). Do you know how much RAM this machine has? Typically you need quite a lot to run synthesis/build firmware with Vivado.

Although this page (https://www.xilinx.com/products/design-tools/vivado/vivado-ml.html#documentation) seems to indicate you only need 2 GB of RAM for the Pynq-Z2 (XC7Z020), it may depends on the number of cores you're using. I would definitely also use a machine with much more RAM (like 32 GB)

wilfredkisku, 2022-05-05:
The board that I am working with is the ZCU104 Ultrascale+ MPSoC, Ubuntu 20.04.3 on a VM. The memory allocated to it is 16 GB.

wilfredkisku, 2022-05-05:
Apparently, the issue was the low memory allocation on the VM. Changed it to 32 GB and it synthesized the design.

Thank you for the helpful insight, if you have information on the parameter limit for a CNN architecture please do share.

closed 2022-05-05 by wilfredkisku