fastmachinelearning/hls4ml #555: Issue with running 3layer.pb example model

rohahann, 2022-05-28:
Hey,

I plan on using hls4ml on tensorflow .pb model. However, when I run the following:

import hls4ml

config = hls4ml.utils.fetch_example_model('3layer.pb')

print(config) #You can print the configuration to see some default parameters

#Convert it to a hls project
hls_model = hls4ml.converters.tf_to_hls(config)

hls_model.compile()
#hls_model.build(csim=False) 

I have the following error: KeyError: 'axis' which seems to be originating from the following file:
python3.8/site-packages/hls4ml/model/hls_layers.py:1341

Could you help me fix this? Thanks!

yrq0208, 2022-06-13:
Hi @rohahann, using hls4ml 0.5.0 help me solve this issue.

jmduarte, 2022-06-19:
hi @rohahann, can you use a keras model instead?

rohahann, 2022-06-21:
> Hi @rohahann, using hls4ml 0.5.0 help me solve this issue.

0.5.0 did work! But I am unable to work with any other .pb model. Looks like a lot of TensorFlow functions are not supported yet.

> hi @rohahann, can you use a keras model instead?

@jmduarte I have to do that. But my keras model has Lambda layers and Conv2DTranspose. Do you plan on supporting these functions any time soon?

Thanks!

yrq0208, 2022-06-21:
@rohahann , yes, I also have some problems with my own CNN and hls4ml. In particular, hls4ml cannot read my custom CNN developed using TF 1 correctly.

rohahann, 2022-06-21:
@yrq0208 Hey! Yes, I know. We might have to request the hls4ml developers for added support.

jmduarte, 2022-07-06:
@rohahann, yes you can fix it with manually changing the pragma to a block (or cyclic) partition to make each part of the array below the 65536 maximum bit width. You can also make the equivalent change to the input variable like so: https://gist.github.com/jmduarte/1a602ac8d69352b1487e15ba49b44c83

Let's use #506 to follow up on this issue

closed 2022-07-06 by jmduarte