fastmachinelearning/hls4ml #563: problem reading in QKeras model

jmitrevs, 2022-06-03:
I haven't gotten a chance to debug much, but reading in a QKeras model fails for me at:

https://github.com/fastmachinelearning/hls4ml/blob/881efeaa84d41f35619080b87a3977b9a73ebc34/hls4ml/converters/keras_to_hls.py#L331

when the layer name  is "q_conv1d_18" and the activation is `{'class_name': 'quantized_relu', 'config': {'bits': 8, 'integer': 0, 'use_sigmoid': 0, 'negative_slope': 0.0, 'use_stochastic_rounding': False, 'relu_upper_bound': None, 'qnoise_factor': 1.0}}`, and you  can't concatenate a string and dictionary. Any recommendations on how to fix this problem? Or does one not use `keras_to_hls` for qkeras models (called from `convert_from_keras_model`)?

jmitrevs, 2022-06-06:
The issue is that hls4ml doesn't seem to handle the case when the QKeras model doesn't explicitly use `QActivation` but instead passes, for example, `activation=quantized_relu(..)` to, for example, `QConv1D`. I will try to add support for this.

jmitrevs, 2022-06-06:
On a somewhat related note, this also cannot be imported to hls4ml:
```python
model = Sequential()
model.add(Dense(64, input_shape=(16,), name='fc1', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation=ReLU(), name='relu1'))
model.add(Dense(32, name='fc2', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation=ReLU(), name='relu2'))
model.add(Dense(32, name='fc3', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation=ReLU(), name='relu3'))
model.add(Dense(5, name='output', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation=Softmax(), name='softmax'))
```
This is from the hls4ml-tutorial, just substituting class instantiations instead of strings for `activation=`.

thesps, 2022-06-07:
Hey, that's actually recently been addressed in #553

thesps, 2022-06-07:
That PR is merged now. Could you try again on master branch?

jmitrevs, 2022-06-07:
The problem is fixed.

closed 2022-06-07 by jmitrevs