fastmachinelearning/hls4ml #566: High Resource usage with Resource Strategy

LordScarface, 2022-06-10:
Hi, I'm trying to implement the same model using different configurations, the baseline model with Strategy Latency works as expected, but changing the Strategy to Resource and setting the Layers ReuseFactors to 2, the model can't get through synthesis as it requires too many resources. Should Strategy Resource not decrease the overall Resource usage?

I'm using Version 0.6.0 with `backend='VivadoAccelerator'`.

Here are the estimated Resource for Strategy Latency on ZCU104 (can be successfully synthesized by Vivado):
[image]

And here the Results with Strategy Resource and increased ReuseFactor to 2 (fails Vivado synthesis):
[image]

LUTs were decreased but DSP and BRAM usage got worse, am I missing something?

Would love some help with this,
Best Regards,
Lukas

thesps, 2022-06-13:
What was the Reuse Factor in the first, Latency Strategy case? Does your NN have many 0 weights, i.e. it's been pruned?

Going from RF=1 to RF=2 often doesn't change the resources very much, or can indeed increase them slightly. Especially if you have a pruned NN, some DSPs could be removed at RF=1 but not at RF=2, which may be why the DSP usage is so much higher in the second case.

I suggest setting the RF to much higher values than 2 to see a resource reduction. Only certain values are valid Reuse Factors, depending on the layer size. Check the conversion print outs for messages about ReuseFactor which reports the allowed values.

thesps, 2022-06-16:
Could you check how many zero weights your model has, even if it's unpruned? What's the LUT usage like in each of those 3 cases? There can be an initial jump in resources from RF=1 to RF=2, although this jump is quite large. But you can see the DSP usage is now smaller again with RF=8, so I'd suggest you can keep increasing RF, and eventually see a DSP usage less than 16%

LordScarface, 2022-06-17:
This is the plot for the weights:
[image]

Lots of small, but not 0 weights, this might be problematic also?

The LUT usage was reduced by increasing the ReuseFactor as intended, just the DSP usage increased significantly

ReuseFactor 1 : 186 882 LUTs (~81%)
ReuseFactor 4 : 153 262 LUTs (~66%)
ReuseFactor 8 :   75 426 LUTs (~32%)

[image]

Thank you for the help, the model is working, I just thaught it was a very significant increase in DSP usage, but when this is expected for small ReuseFactors I'll close this

closed 2022-06-29 by LordScarface