fastmachinelearning/hls4ml #571: Failure to Compile Medium-sized Keras Conv1d Layer with Vivado, Latency Strategy

Jonathan-Shoemaker, 2022-06-16:
On the current version of main, I am not able to synthesize a keras model containing a Conv1D layer with an input size of 8192 x 21. If I reduce the size of the layer to be 1024 x 21, the model has no issues (the only barrier this crosses I have noticed is the max size of a short). This issue also does not exist for when the strategy is set to 'Resource' instead of 'Latency'.

The place where it seems to fail is when the conv_1d_cl function calls conv_1d_latency_cl - specifically, the function conv_1d_latency_cl appears to not begin. 

The vivado_hls output is not particularly helpful here. It prints out the following:
```
@E Simulation failed: SIGSEGV.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
``` 

The code to reproduce a simple network that can't be correctly compiled is below. The code runs to completion, but if one tries to use the hls_model on any data or anything it crashes. Additionally going into the created directory (my-hls-test) and trying to build the project with vivado_hls produces the output I included above. Again, if the below code is just changed such that the input has shape (1024, 21) the error does not occur.
```
import numpy as np
from tensorflow import keras
from tensorflow.keras.models import Sequential, Model
from tensorflow.keras.layers import Conv1D, BatchNormalization, Activation

model = Sequential()

#single layer, with convolution, batch normalizatoin and activation
model.add(Conv1D(
  filters=21,
  kernel_size=7,
  strides=1,
  input_shape=(8192, 21),
  name="input_conv",
  padding="same",
))
model.add(BatchNormalization())
model.add(Activation('tanh'))

# do some random training just to get everything to compile
model.compile(optimizer='adam', loss='mse')
rand_input = np.random.normal(size=(1, 8192, 21))
rand_output = np.random.normal(size=(1, 8192, 21))
model.fit(x=rand_input, y=rand_output, batch_size=1)

import hls4ml

hls_config = hls4ml.utils.config_from_keras_model(model)
hls_model = hls4ml.converters.convert_from_keras_model(model, hls_config=hls_config)
hls_model.compile()
```

jmitrevs, 2022-07-25:
Are there issues if you try doing hls_mode.predict(). I compiled the testbench source code with g++ (not using Vivado_HLS) and it seems to work. But in general, for larger convolution networks like this you probably want to use the streaming interface. Try adding io_type='io_stream' to the convert_from_keras_model function:

Jonathan-Shoemaker, 2022-07-25:
Yeah, this is an older issue. The input size is over the stated limits for io_parallel, which was causing problems compiling that way. This sized network should be (needs to be?) compiled with io_stream, and that is working fine.

closed 2022-07-25 by Jonathan-Shoemaker