fastmachinelearning/hls4ml #572: Number of filters limitation

wilfredkisku, 2022-06-16:
Is there a limitation to the number of filters in a CNN, as a layer with 32 filters tends to be the bottleneck during synthesis. The synthesis is unable to complete and gets stuck at the conv2d layer having 32 filter.

vloncar, 2022-06-17:
Depends on your config. Assuming you use `io_stream`, the limit will be related to the strategy used, since that affects the algorithm used for the CNN kernel. If you use latency strategy (the default), then `filt_height x filt_width x n_channels x n_filters < 4096`. If you use `io_parallel`, well, you shouldn't be using it with large models.

wilfredkisku, 2022-06-18:
asdasd

wilfredkisku, 2022-06-18:
I have used the `io_stream` and the strategy used is `Resource`. Another issue that I am facing is that when I am using the config file to generate the hls4ml model from the quantized models, it results in an accuracy drop from ~75% to ~10%. I check that the baseline quantized model does predict with an accuracy of around ~71% but the hls4ml model gives drops the accuracy after synthesising the model.

jmduarte, 2022-06-19:
Hi @wilfredkisku, can you share your model. You may look into tracing / profiling functionality.

You can make 1D plots of the expected output vs hls4ml output like so: https://github.com/hls4ml-finn-mlperftiny/CIFAR10/blob/main/hls4ml/convert.py#L222-L233

[image]

This can help you pinpoint which layers are causing a mismatch. Then you can increase the precision of those layers. Usually it's required to either increase the precision of the outputs or the accumulators (or both).

liuhao-97, 2022-07-07:
Hi have you solve your problem?

liuhao-97, 2022-07-07:
Hi have you solve your problem? Maybe you can compare the output of the "output_softmax" layer between the hls_model and the keras model. this is my problem. [https://github.com/fastmachinelearning/hls4ml/issues/590](https://github.com/fastmachinelearning/hls4ml/issues/590)

wilfredkisku, 2022-07-07:
@liuhao-97 thank you for the reply. No i could not get it corrected, it's still has the accuracy drop. Is there anything else to rectify the issue that you have pointed out?

liuhao-97, 2022-07-15:
Hi have you tried with full precision model (ap_fix<32,16>)? I mean don't quantize the model and set hls config to ap_fix<32,16>. Maybe you can check the output of the last sofmax layer between the keras model and hls model.

liuhao-97, 2022-07-15:
For me I found there might be some problem with the softmax layer. I print the output of dense layer and it works fine. But for softmax layer, the output is totally different. If you check this link [https://github.com/hls4ml-finn-mlperftiny/CIFAR10/blob/main/hls4ml/convert.py](url) you will find it remove the softmax layer. so I assume there might be some problem with softmax layer.

wilfredkisku, 2022-07-16:
@liuhao-97 I tried to profile the layer but came up with a `graph disconnected error'.

```
ValueError: Graph disconnected: cannot obtain value for tensor KerasTensor(type_spec=TensorSpec(shape=(None, 32, 32, 1), dtype=tf.float32, name='input_1'), name='input_1', description="created by layer 'input_1'") at layer "prune_low_magnitude_conv1". The following previous layers were accessed without issue: []
```

wilfredkisku, 2022-07-16:
> Hi have you tried with full precision model (ap_fix<32,16>)? I mean don't quantize the model and set hls config to ap_fix<32,16>. Maybe you can check the output of the last sofmax layer between the keras model and hls model.

The full precision layer works file, for me the accuracy drops only for the quantized hls model.

liuhao-97, 2022-07-16:
Can you print your output? Is it consisted of some same number and some zeros like [0.25, 0.25, 0.25, 0, 0, 0]?

wilfredkisku, 2022-07-16:
I am not able to print the output yet.

liuhao-97, 2022-07-17:
> @liuhao-97 I tried to profile the layer but came up with a `graph disconnected error'.
>
> ```
> ValueError: Graph disconnected: cannot obtain value for tensor [...]
Did you prune the model? I think quantized pruned model can't work with hls4ml. Maybe you can try quantized model but don't pruned it.

liuhao-97, 2022-07-18:
Besides which hls4ml are you using? hls4ml 0.6.0 or the newest branch?

wilfredkisku, 2022-07-18:
I am using hls4ml 0.6.0. Has this issue been resolved in the new branch?

liuhao-97, 2022-07-19:
> I am using hls4ml 0.6.0. Has this issue been resolved in the new branch?

not sure. You can have a try with new brench. Besides, have you tried with "io_type='io_parallel'"? Maybe it can sove the problem.

liuhao-97, 2022-07-19:
> > I am using hls4ml 0.6.0. Has this issue been resolved in the new branch?
> 
> not sure. You can have a try with new brench. Besides, have you tried with "io_type='io_parallel'"? Maybe it can sove the problem.

https://github.com/fastmachinelearning/hls4ml/pull/448 Maybe you can check this.

wilfredkisku, 2022-07-19:
@liuhao-97 I tried but it still did not work for me. Did you get a workaround to make sure that the accuracy does not drop?

liuhao-97, 2022-07-20:
I think it is because you prune the model. When you prune the model,  somehow the original connection of the model layer by layer goes wrong, which can be seen in your error.  
Can you try with a non-pruning model again to see if there is still accuracy loss?

> @liuhao-97 I tried to profile the layer but came up with a `graph disconnected error'.
>
> ```
> ValueError: Graph disconnected: cannot obtain value for tensor [...]

wilfredkisku, 2022-07-20:
@liuhao-97 yes the error has been removed after I removed pruning the error has also been checked. Thank you.

wilfredkisku, 2022-07-21:
@jmduarte models that have concatenate layer or add layers have a considerable accuracy drop. Might be a bug.