fastmachinelearning/hls4ml #578: Converting custom Keras model leads to 'NoneType' object has no attribute 'shape'

SteCla0, 2022-06-21:
Using hls4ml v0.6.0, Vivado HLS 2020.1, on Ubuntu 20.04

I can convert and synthesize any hls4ml example I tried. Using my own model, from keras v2.3.0, I run into a crash.

At the "Creating HLS model" step, I get " AttributeError: 'NoneType' object has no attribute 'shape' " at "return inp1.shape != inp2.shape" in repack_stream.py in the optimizer.
When I comment out the optimize model step in hls_model.py, the software continues successfully to the "Writing HLS project" step.

Printing inp1 and inp2, tells me inp2 is None.

Printing the inputs list for any Merge layer (Merge class in hls_layer.py) shows correct layers, yet printing the layers in the optimizer shows change of inputs (input[0] = activation layer to batchnormalisation, to conv1d and input[1]=activation), and node.get_input_variable on inputs[1] returns the None that causes the error.

jmduarte, 2022-06-21:
@SteCla0 Can you provide a script or steps to reproduce the error? Do you also see it on the current master branch?

SteCla0, 2022-06-22:
I am using hls4ml v0.6.0
The error seems to be triggered by the optimization of Add (Merge) layers.

When I comment out the optimizer step (self._optimize_model(self.config.optimizers)) OR when I remove all **add** layers from my model (which means I then have a single path through the network), the HLS project writing succeeds.

In attachment some model that can be used to reproduce the error.
[jsons.zip](https://github.com/fastmachinelearning/hls4ml/files/8956527/jsons.zip)

To try with the master, should I pip3 uninstall hls4ml first and then use pip3 install git+https://github.com/fastmachinelearning/hls4ml@master ?

jmduarte, 2023-04-09:
@SteCla0 sorry for the delay. Yes, I would try with the latest `main` branch. I think it should be resolved.

In converting your model, however I noticed an issue with batchnormalization with `center=False` so I fixed that in an upcoming PR

closed 2023-04-13 by jmitrevs