fastmachinelearning/hls4ml #590: random prediction of resnet generated by hls4ml

liuhao-97, 2022-07-06:
Thank you for your great work!
I am trying to generate Resnet20 with hls4ml. I trained Resnet20 with Qkeras and transfer to hls_model. when I ran hls_model.predict, it showed random prediction and the accuracy was low. But the quantized model works well.
I tried with ap_fix<32,16> and ap_fix<16,6> but it still not work.

**This is my model**:

from tensorflow.keras import layers
from qkeras.qlayers import QDense, QActivation
from qkeras.quantizers import quantized_bits, quantized_relu
from qkeras import QActivation
from qkeras import QDense, QConv2DBatchnorm, QConv2D
from tensorflow.keras.layers import Dropout

# Subtracting pixel mean improves accuracy
subtract_pixel_mean = True

# Model parameter
# ----------------------------------------------------------------------------
#           |      | 200-epoch | Orig Paper| 200-epoch | Orig Paper| sec/epoch
# Model     |  n   | ResNet v1 | ResNet v1 | ResNet v2 | ResNet v2 | GTX1080Ti
#           |v1(v2)| %Accuracy | %Accuracy | %Accuracy | %Accuracy | v1 (v2)
# ----------------------------------------------------------------------------
# ResNet20  | 3 (2)| 92.16     | 91.25     | -----     | -----     | 35 (---)
# ResNet32  | 5(NA)| 92.46     | 92.49     | NA        | NA        | 50 ( NA)
# ResNet44  | 7(NA)| 92.50     | 92.83     | NA        | NA        | 70 ( NA)
# ResNet56  | 9 (6)| 92.71     | 93.03     | 93.01     | NA        | 90 (100)
# ResNet110 |18(12)| 92.65     | 93.39+-.16| 93.15     | 93.63     | 165(180)
# ResNet164 |27(18)| -----     | 94.07     | -----     | 94.54     | ---(---)
# ResNet1001| (111)| -----     | 92.39     | -----     | 95.08+-.14| ---(---)
# ---------------------------------------------------------------------------
n = 3

# Model version
# Orig paper: version = 1 (ResNet v1), Improved ResNet: version = 2 (ResNet v2)
version = 1

# Computed depth from supplied model parameter n
if version == 1:
    depth = n * 6 + 2
elif version == 2:
    depth = n * 9 + 2

# Model name, depth and version
model_type = 'ResNet%dv%d' % (depth, version)

def lr_schedule(epoch):
    """Learning Rate Schedule

    Learning rate is scheduled to be reduced after 80, 120, 160, 180 epochs.
    Called automatically every epoch as part of callbacks during training.

    # Arguments
        epoch (int): The number of epochs

    # Returns
        lr (float32): learning rate
    """
    lr = 1e-3
    if epoch > 180:
        lr *= 0.5e-3
    elif epoch > 160:
        lr *= 1e-3
    elif epoch > 120:
        lr *= 1e-2
    elif epoch > 80:
        lr *= 1e-1
    print('Learning rate: ', lr)
    return lr

def resnet_layer(inputs,
[... 296 lines cut ...]
    from matplotlib.legend import Legend
    leg = Legend(ax, lines, labels=['Keras', 'hls4ml'],
                loc='lower right', frameon=False)
    ax.add_artist(leg)
    plt.figtext(0.2, 0.38,label, wrap=True, horizontalalignment='left',verticalalignment='center')
    plt.ylim(0.01,1.)
    plt.xlim(0.7,1.)

X_test_reduced = x_test[:50]
Y_test_reduced = y_test[:50]
start = time.time()
y_predict = model.predict(X_test_reduced)
end = time.time()
print('\n model took {} minutes to run!\n'.format( (end - start)/60.))

start = time.time()
y_predict_hls4ml = hls_model.predict(np.ascontiguousarray(X_test_reduced))
end = time.time()
print('\n hls model took {} minutes to run!\n'.format( (end - start)/60.))

plotROC(Y_test_reduced, y_predict, y_predict_hls4ml, label="Keras") 

[resnet20_1-QKeras-Copy2.ipynb.zip](https://github.com/fastmachinelearning/hls4ml/files/9056866/resnet20_1-QKeras-Copy2.ipynb.zip)

Attached is my code. I will be grateful if you could pls help me with it.

liuhao-97, 2022-07-07:
I think there might be some problem with the last softmax layer. I printed the output of "output_softmax" and it showed like the fig. 
[image]

os-hxfan, 2022-11-04:
Similar issues happened on my side, What is the current solution or workaround?  @liuhao-97

joekang21, 2025-01-24:
@liuhao-97 Have you ever been able to figure out what is going on here? My output softmax layer is causing a similar issue.

closed 2026-10-21 by github-actions[bot]