fastmachinelearning/hls4ml #616: Error when generate .xo file

liuhao-97, 2022-07-22:
Thank you for your work!
When I run the command "hls_model.build(csim=False,synth=True, ecxport=True, bitfile=True)" with the newest branch, I met the error “ERROR: [Common 17-69] Command failed: The '-kernel_xml' or '-kernel_xml_args' must be specified”. 
I think there are some error when it runs the command "package_xo  -force -xo_path xo_files/${myproject}_kernel.xo -kernel_name krnl_rtl -ip_directory hls4ml_IP". 
I am using U280. 
My config is 
'''
cfg = hls4ml.converters.create_config(board='alveo-u280', clock_period= 2, part='xcu280-fsvh2892-2L-e', backend='VivadoAccelerator')
cfg['HLSConfig'] = config

cfg['AcceleratorConfig']['Driver']    = 'python'
cfg['AcceleratorConfig']['Board']     = 'alveo-u280'
cfg['AcceleratorConfig']['Interface'] = 'axi_stream'
#cfg['AcceleratorConfig']['Interface'] = 'axi_lite'
#cfg['AcceleratorConfig']['Interface'] = 'axi_master'
cfg['AcceleratorConfig']['Precision']['Input']  = 'float'
cfg['AcceleratorConfig']['Precision']['Output'] = 'float'
cfg['AcceleratorConfig']['Platform'] = 'xilinx_u280_xdma_201920_3'

cfg['IOType']= 'io_stream'
cfg['KerasModel'] = model
cfg['OutputDir'] = 'test/hls4ml_prj'

print("-----------------------------------")
plotting.print_dict(cfg)
print("-----------------------------------")
hls_model = hls4ml.converters.keras_to_hls(cfg)
hls4ml.model.optimizer.get_optimizer('output_rounding_saturation_mode').configure(layers=[])
'''

thesps, 2022-07-22:
Which Vivado version are you using? This Alveo bitfile generation currently only works with 2020.1

liuhao-97, 2022-07-22:
> Which Vivado version are you using? This Alveo bitfile generation currently only works with 2020.1

Thanks for your reply. I am using 2020.2. I get it.

closed 2022-07-25 by liuhao-97