fastmachinelearning/hls4ml #622: issue with ApplyAlpha for conv2d layers

m-glowacki, 2022-07-27:
I'm attempting a hls synthesis of a simple quantised CNN model, featuring:

- conv_2d layers
- dense layers 
- batch normalisation 

the model is quantised during training, with no alpha scaling (`alpha=1`)

the synthesis breaks with this exception:

```bash
  File "/software/ys20884/miniconda3/envs/ml_env/lib/python3.8/site-packages/hls4ml/model/layers.py", line 105, in _validate_attributes
    raise Exception('Unexpected value of attribute "{}" of layer "{}" ({}). Expected {}, got {} ({})'
Exception: Unexpected value of attribute "n_in" of layer "q_conv2d_alpha" (ApplyAlpha). Expected <class 'int'>, got <class 'NoneType'> (None)
```
this is the layer in question:

`Layer name: q_conv2d, layer type: QConv2D, input shapes: [[None, 20, 12, 1]], output shape: [None, 18, 10, 4]`

I'm on the current [main](https://github.com/fastmachinelearning/hls4ml) branch

I previously had no issues with the same model on release [0.6.0](https://github.com/fastmachinelearning/hls4ml/tree/release-v0.6.0)

thanks for taking a look.

bo3z, 2022-07-27:
Hi @m-glowacki. Thanks for raising this issue. I believe this has been addressed in PR #612 . Could you check out the branch and see if it resolves the problem?

m-glowacki, 2022-07-27:
thanks @bo3z! Indeed this has fixed things, issue closed.

closed 2022-07-27 by m-glowacki