fastmachinelearning/hls4ml #651: hls_model.compile() fails to compile project

tengkianen, 2022-09-10:
## Quick summary
hls_model.compile() runs an exception from tutorial notebook part 1

## Details
```
hls_model.compile()
X_test = np.ascontiguousarray(X_test)
y_hls = hls_model.predict(X_test)
```

### Actual behavior
```
Writing HLS project
Done
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Input In [13], in <cell line: 1>()
----> 1 hls_model.compile()
      2 X_test = np.ascontiguousarray(X_test)
      3 y_hls = hls_model.predict(X_test)

File D:\Anaconda\lib\site-packages\hls4ml\model\hls_model.py:534, in HLSModel.compile(self)
    532 ret_val = os.system('bash build_lib.sh')
    533 if ret_val != 0:
--> 534     raise Exception('Failed to compile project "{}"'.format(self.config.get_project_name()))
    535 lib_name = 'firmware/{}-{}.so'.format(self.config.get_project_name(), self.config.get_config_value('Stamp'))
    536 if self._top_function_lib is not None:

Exception: Failed to compile project "myproject"
```

vloncar, 2022-09-10:
There's a notebook log that should say the reason for the compilation failure, can you share that as well?

tengkianen, 2022-09-11:
Hey I can't seem to locate the log file, do you have an idea of where it's located?

avinashiiitb, 2023-02-04:
Any update on the  above issue

vloncar, 2023-03-22:
No log -> no help. Simple.

LuisSpader, 2023-03-22:
I found this answer in stackoverflow: [https://stackoverflow.com/questions/66919012/how-can-this-issue-with-hls4ml-be-resolved]

> Disclaimer : I asked this question before looking into compilation error details. [This question contains the answer to the problem explained here.](https://stackoverflow.com/questions/12938663/clang-on-cygwin-with-c11) In short, variable type definition errors in exception definitions prevent synthesis from happening properly.
> 
> Unless you are willing to find the correct gcc compiler to run Vivado Synthesis without errors on, I recommend going to Linux for this task.

jmduarte, 2023-03-23:
Thanks @LuisSpader for the additional information. I tried to reproduce this on Linux and I couldn't, so I suppose it's a Windows-specific issue. Our developer/maintainer efforts are focused on Linux so I'm sorry we can't be of more help. 

If somebody has a fix for Windows, we'd be willing to consider it as a pull request, but it won't be a high priority for us to fix at the moment. Thanks!

closed 2023-03-23 by jmduarte