fastmachinelearning/hls4ml #691: building hls_model error

zyt1024, 2022-12-05:
When I use `hls_model.build(csim=False)`,occuring error
[image]
[image]

jmduarte, 2022-12-05:
Hi @zyt1024, that is just a warning, not an error. 

If there is an error, you should share that.

Regarding the warning, we could perhaps rewrite the condition to avoid this tautological comparison (which only happens for certain pad/stride combinations), but I would say it's not worth it (since the code works, it just issues a warning).

zyt1024, 2022-12-05:
> Hi @zyt1024, that is just a warning, not an error.
>
> If there is an error, you should share that.
>
> Regarding the warning, we could perhaps rewrite the condition to avoid this tautological comparison [...]

ok,below is the hls logs，at the end of the log I am getting an error.

```
(ap_q_mode)5, (ap_o_mode)3, 0>, config23>' completely with a factor of 10.
INFO: [HLS 200-489] Unrolling loop 'Result' (firmware/nnet_utils/nnet_dense_latency.h:120) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config23>' completely with a factor of 10.
INFO: [HLS 200-489] Unrolling loop 'Loop-1' (firmware/nnet_utils/nnet_activation.h:76) in function 'nnet::relu<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, relu_config22>' completely with a factor of 256.
INFO: [HLS 200-489] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config21>' completely with a factor of 256.
ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config21>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Sun Dec  4 21:11:59 2022...

```

[image]

jmitrevs, 2022-12-22:
I think you are using too large of a model for the configuration. For larger convolutions you generally have to use io_stream instead of io_parallel (the default). For io_parallel you can sometimes get to bigger values to compile with the Resource strategy, but that's still for fairly small CNNs.

ddddavid-he, 2023-01-26:
> I think you are using too large of a model for the configuration. For larger convolutions you generally have to use io_stream instead of io_parallel (the default). For io_parallel you can sometimes get to bigger values to compile with the Resource strategy, but that's still for fairly small CNNs.

I tried to set `io_type = 'io_stream'`, but this problem still occurs when I attempts to transform a large dense layer (unite = 1250). The error messages are as follows:
```
INFO: [HLS 200-489] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_wrapper<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' completely with a factor of 1250.
ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_wrapper<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
```

Is there any method to transform such a large dense layer?

vloncar, 2023-01-26:
Set `'Strategy': 'Resource'` in the config and increase the `ReuseFactor` for that layer. The log will tell you what the valid values are. Start from the largest (it will have a large latency), then go down.

ddddavid-he, 2023-01-27:
@vloncar Thank you for your kind replying. This does works.

MODISENEISO, 2023-08-27:
[image]

This is the shape of the model I am converting  with this configuration
[image]

The error is as follows:
ERROR: [XFORM 203-504] Stop unrolling loop 'ResizeHeight' (firmware/nnet_utils/nnet_image_stream.h:36) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 64u>, config38>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

MODISENEISO, 2023-08-28:
> Set `'Strategy': 'Resource'` in the config and increase the `ReuseFactor` for that layer. The log will tell you what the valid values are. Start from the largest (it will have a large latency), then go down.
 Hello I have a model with almost 2 000 000 parameters can this approach work, and is the error related to the part number of the bard chosen