fastmachinelearning/hls4ml #728: Keras 1D ConV model build() failed

Yukinosdd, 2023-03-05:
Hi, I am using Keras model, I am using hls4ml to compile and build a model using the compile() and build() methods. However, when running the build() method, there is an error in the log file. Can you please help me to figure out what the problem is? Thank you very much.

the model is:
```
model = tf.keras.models.Sequential()
model.add(tf.keras.layers.Conv1D(32, 3, activation='relu', input_shape=(512, 1))
model.add(tf.keras.layers.MaxPooling1D(2, 2))
model.add(tf.keras.layers.Conv1D(32, 3, activation='relu'))
model.add(tf.keras.layers.MaxPooling1D(2, 2))
model.add(tf.keras.layers.Conv1D(32, 3, activation='relu'))
model.add(tf.keras.layers.MaxPooling1D(2, 2))
model.add(tf.keras.layers.Conv1D(32, 3, activation='relu'))
model.add(tf.keras.layers.MaxPooling1D(2, 2))
model.add(tf.keras.layers.Flatten())
model.add(tf.keras.layers.Dense(960))
model.add(tf.keras.layers.Dense(2))
```

and:
```
model.compile(loss=tf.keras.losses.MeanSquaredError(),
              optimizer=tf.keras.optimizers.Adam(),
              metrics=['accuracy'])

model.fit(X_train, Y_train, epochs=20, batch_size=64)
```
```
config = hls4ml.utils.config.config_from_keras_model(model, granularity='model')
hls_model = hls4ml.converters.convert_from_keras_model(model, hls_config=config, output_dir='test_prj')
hls_model.compile()
hls_model.build()
```
when I running`hls_model.compile()`, display`Writing HLS project Done`

and running`hls_model.build()`, I have
```

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'Yuki' on host 'ubuntu' (Linux_x86_64 version 4.18.0-15-generic) on Sat Mar 04 23:38:13 PST 2023
INFO: [HLS 200-10] On os Ubuntu 18.04.2 LTS
INFO: [HLS 200-10] In directory '/home/Yuki/hls4ml-tutorial/hls4ml-tutorial/test/test_prj'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Creating and opening project '/home/Yuki/hls4ml-tutorial/hls4ml-tutorial/test/test_prj/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/Yuki/hls4ml-tutorial/hls4ml-tutorial/test/test_prj/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-10] Setting target device to 'xcku115-flvb2104-2-i'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 200-40] In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:22:
In file included from firmware/parameters.h:11:
In file included from firmware/nnet_utils/nnet_conv1d.h:24:
firmware/nnet_utils/nnet_conv1d_latency.h:79:55: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
                    if((ii*CONFIG_T::stride_width+jj) < CONFIG_T::pad_left || (ii*CONFIG_T::stride_width+jj) >= (CONFIG_T::pad_left + CONFIG_T::in_width)){
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv1d.h:62:9: note: in instantiation of function template specialization 'nnet::conv_1d_latency_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config2>' requested here
        conv_1d_latency_cl<data_T, res_T, CONFIG_T>(data, res, weights, biases);
[... 93 lines cut ...]
In file included from firmware/parameters.h:11:
In file included from firmware/nnet_utils/nnet_conv1d.h:24:
firmware/nnet_utils/nnet_conv1d_latency.h:24:55: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
                    if((ii*CONFIG_T::stride_width+jj) < CONFIG_T::pad_left || (ii*CONFIG_T::stride_width+jj) >= (CONFIG_T::pad_left + CONFIG_T::in_width)){
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv1d_latency.h:66:31: note: in instantiation of function template specialization 'nnet::compute_multiplier_limit<config11>' requested here
 const int multiplier_limit = compute_multiplier_limit<CONFIG_T>(weights);
                              ^
firmware/nnet_utils/nnet_conv1d.h:62:9: note: in instantiation of function template specialization 'nnet::conv_1d_latency_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config11>' requested here
        conv_1d_latency_cl<data_T, res_T, CONFIG_T>(data, res, weights, biases);
        ^
firmware/myproject.cpp:104:2: note: in instantiation of function template specialization 'nnet::conv_1d_cl<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, config11>' requested here
 nnet::conv_1d_cl<layer10_t, layer11_t, config11>(layer10_out, layer11_out, w11, b11);
 ^
8 warnings generated.
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/nnet_utils/nnet_dense_latency.h:64:9
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/nnet_utils/nnet_dense_latency.h:79:2
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense_latency.h:76:9
WARNING: [HLS 200-471] Dataflow form checks found 3 issue(s) in file firmware/myproject.cpp
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:05:11 ; elapsed = 00:05:24 . Memory (MB): peak = 959.777 ; gain = 526.996 ; free physical = 2383 ; free virtual = 2875
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:05:11 ; elapsed = 00:05:24 . Memory (MB): peak = 959.777 ; gain = 526.996 ; free physical = 2384 ; free virtual = 2876
INFO: [HLS 200-10] Starting code transformations ...
```

Why are these warnings appearing? Is it because there's an error in my model? Please help me.

vloncar, 2023-03-05:
This is not a bug. The messages you see are just warnings, your synthesis is still running. Also, the input to your model is too large to be processed this way, it is unlikely to work. Try with io_stream, and switch to main branch, your hls4ml version is old.

Yukinosdd, 2023-03-05:
> This is not a bug. The messages you see are just warnings, your synthesis is still running. Also, the input to your model is too large to be processed this way, it is unlikely to work. Try with io_stream, and switch to main branch, your hls4ml version is old.

Sure, thank you. How can I tell if my hls4ml version is too old? After using the update command, I got the message `Requirement already satisfied: hls4ml in ./venv hls4ml/lib/python3.6/site-packages (0.6.0)`. Does this mean that I already have the latest version?

vloncar, 2023-03-05:
No, you need to install from git, like `pip install git+https://github.com/fastmachinelearning/hls4ml.git`. I will reiterate that this is unlikely to work even with the latest version, you should try with `io_stream`.

closed 2023-03-19 by jmduarte