fastmachinelearning/hls4ml #745: Code no longer works after updating from pip install hls4ml to main branch

vandenBergArthur, 2023-04-04:
Hi all,

For the past couple of months, I've been working with hls4ml using the conda environment that is created for the tutorial (so the environment.yml file). Everything was fine until yesterday, when I finally noticed that the hls4ml version from pip is very outdated, so I uninstalled hls4ml with `pip uninstall hls4ml` and I installed the main branch like this: `pip install git+https://github.com/fastmachinelearning/hls4ml@main`.

But now my code no longer works:

- The printed output from _hls4ml.converters.keras_to_hls_ is somehow printed twice.
- I cannot compile my Keras models anymore

I have tried the following things to solve the problem:

- Remove the conda environment with `conda env remove --name hls4ml-tutorial`
- Reinstalled the hls4ml-tutorial environment with `conda env create -f environment.yml`. The environment.yml file is downloaded from [here](https://github.com/fastmachinelearning/hls4ml-tutorial).
- Manually created an environment and installed the [dependencies](https://fastmachinelearning.org/hls4ml/setup.html).
But nothings seems to get me back on track.

Any help would be very appreciated because at this moment, I am stuck and I don't know what to do anymore...

Ps: 

- I am running Ubuntu 20.04 and installed Vivado 2019.2
- I'm a beginner

vloncar, 2023-04-04:
Compilation relies on using the host's `g++` compiler. Maybe your conda environment has something extra? I tried creating a new empty environment based on python 3.8 and just installed hls4ml from main there, it pulled it's dependencies and it works. So the issue must be someplace else.

jmduarte, 2023-04-04:
@vandenBergArthur if you post a script that can reproduce your issue (e.g. compiling a simple model), we can try to help resolve it, but otherwise we don't know how to help.

vandenBergArthur, 2023-04-05:
> Compilation relies on using the host's `g++` compiler. Maybe your conda environment has something extra? I tried creating a new empty environment based on python 3.8 and just installed hls4ml from main there, it pulled it's dependencies and it works. So the issue must be someplace else.

I don't know what changed, but I tried this again and now it works after I manually added pydot & graphviz. Thanks!

closed 2023-04-05 by vandenBergArthur