fastmachinelearning/hls4ml #753: Unsupported layer error

sandeep1404, 2023-04-09:
Hi, I am using the latest master branch of hls4ml, I am trying to synthesize  my model which is given below 

'''
Model: "model_4"
__________________________________________________________________________________________________
 Layer (type)                   Output Shape         Param #     Connected to                     
==================================================================================================
 input (InputLayer)             [(None, 128, 128, 3  0           []                               
                                )]                                                                
                                                                                                  
 conv2d_94 (Conv2D)             (None, 128, 128, 16  448         ['input[0][0]']                  
                                )                                                                 
                                                                                                  
 conv2d_95 (Conv2D)             (None, 128, 128, 16  2320        ['conv2d_94[0][0]']              
                                )                                                                 
                                                                                                  
 conv2d_96 (Conv2D)             (None, 128, 128, 16  2320        ['conv2d_95[0][0]']              
                                )                                                                 
                                                                                                  
 conv2d_97 (Conv2D)             (None, 128, 128, 16  2320        ['conv2d_96[0][0]']              
                                )                                                                 
                                                                                                  
 add_37 (Add)                   (None, 128, 128, 16  0           ['conv2d_97[0][0]',              
                                )                                 'conv2d_94[0][0]']              
                                                                                                  
 conv2d_98 (Conv2D)             (None, 128, 128, 16  2320        ['add_37[0][0]']                 
                                )                                                                 
                                                                                                  
 conv2d_99 (Conv2D)             (None, 128, 128, 16  2320        ['conv2d_98[0][0]']              
                                )                                                                 
                                                                                                  
 add_38 (Add)                   (None, 128, 128, 16  0           ['conv2d_99[0][0]',              
                                )                                 'add_37[0][0]']                 
                                                                                                  
 activation_22 (Activation)     (None, 128, 128, 16  0           ['add_38[0][0]']                 
                                )                                                                 
                                                                                                  
 conv2d_100 (Conv2D)            (None, 128, 128, 16  2320        ['activation_22[0][0]']          
                                )                                                                 
                                                                                                  
 conv2d_101 (Conv2D)            (None, 128, 128, 16  272         ['conv2d_100[0][0]']             
                                )                                                                 
                                                                                                  
 add_39 (Add)                   (None, 128, 128, 16  0           ['conv2d_101[0][0]',             
                                )                                 'activation_22[0][0]']          
                                                                                                  
 activation_23 (Activation)     (None, 128, 128, 16  0           ['add_39[0][0]']                 
                                )                                                                 
                                                                                                  
 global_average_pooling2d_19 (G  (None, 16)          0           ['activation_23[0][0]']          
 lobalAveragePooling2D)                                                                           
                                                                                                  
 tf.expand_dims_22 (TFOpLambda)  (None, 1, 16)       0           ['global_average_pooling2d_19[0][
                                                                 0]']                             
                                                                                                  
 tf.expand_dims_23 (TFOpLambda)  (None, 1, 1, 16)    0           ['tf.expand_dims_22[0][0]']      
                                                                                                  
 conv2d_102 (Conv2D)            (None, 1, 1, 2)      290         ['tf.expand_dims_23[0][0]']      
                                                                                                  
 conv2d_103 (Conv2D)            (None, 1, 1, 16)     304         ['conv2d_102[0][0]']             
                                                                                                  
 multiply_17 (Multiply)         (None, 128, 128, 16  0           ['conv2d_103[0][0]',             
                                )                                 'activation_23[0][0]']          
                                                                                                  
 conv2d_104 (Conv2D)            (None, 128, 128, 3)  435         ['multiply_17[0][0]']            
                                                                                                  
 add_40 (Add)                   (None, 128, 128, 3)  0           ['conv2d_104[0][0]',             
                                                                  'input[0][0]']                  
                                                                                                  
==================================================================================================
[... 74 lines cut ...]
hls_config['LayerName']['conv2d_101']['Strategy'] = 'Resource'
hls_config['LayerName']['conv2d_102']['Strategy'] = 'Resource'
hls_config['LayerName']['conv2d_103']['Strategy'] = 'Resource'
hls_config['LayerName']['conv2d_104']['Strategy'] = 'Resource'

###############################################

print("-----------------------------------")
print("Configuration")
plotting.print_dict(hls_config)
print("-----------------------------------")

hls_model = hls4ml.converters.convert_from_keras_model(
        model,
        hls_config=hls_config,
        output_dir="model_pynq_fpn/hls4ml_prj_pynq_fpnrnet_32bit",
        backend="VivadoAccelerator",
        board="pynq-z2",
        io_type='io_stream',
        part='xc7z020-clg400-1'
    )

```

Is there any way to bipass the layers which are not supported by hls4ml and synthesize the model? or how to overcome this issues since my model has expand layer, multiply and add layer are these layers supported for synthesis. Please help me in this regard. Thank you in Advance.

jmduarte, 2023-04-09:
Hi `TFOpLambda` layers are not (and really cannot be) generically supported because they can really represent anything.

I see you're using them for a simple reshape operation, so instead, you should use a standard `Reshape` layer.

If in the future you actually need a custom layer, you can use the Extension API to define the corresponding HLS for a custom layer: https://fastmachinelearning.org/hls4ml/advanced/extension.html

closed 2023-04-09 by jmduarte

sandeep1404, 2023-04-09:
Thank you @jmduarte  for the reply.