fastmachinelearning/hls4ml #790: Update hls4ml recipe in conda-forge

jmduarte, 2023-05-12:
Currently the hls4ml conda-forge feedstock (https://github.com/conda-forge/hls4ml-feedstock) recipe is broken because some of our dependencies (and their dependencies are missing from conda-forge. The full list is of missing libraries are:

  - calmjs.parse
  - pydigitalwavetools ==1.1
  - qkeras
     - pyparser (but may not actually need it? pyparsing is on conda-forge)
     - tensorflow-model-optimization >=0.2.1

There is an example of someone uploading tensorflow-model-optimization on a separate channel: https://anaconda.org/HCC/tensorflow-model-optimization

vloncar, 2023-05-12:
One design goal that we mostly abandoned but can come back to is to completely modularize hls4ml such that only the core internals are available by default and everything else is an option. The options would have their own dependencies. These options would for example be keras converter (depends on `h5py`, `qkeras` and maybe `tensorflow`), pytorch converter (`pytorch` and ultimately `pytorch_geometric`), onnx converter (`onnx` and `qonnx`), quartus backend (`calmjs.parse`, `tabulate`), vivado (`pydigitalwavetools`), vitis (same as vivado)... Because I see this project growing and these dependencies are not something that  the users will care about having in their environments. Apart from the few of us, users of Vivado don't care about Quartus (and having the environment with `calmjs.parse` and `tabulate`). Catapult is incoming, `pylog` code generation is coming, functional verification with `pyverilator` will come at some point etc. These will all include extra packages. Perhaps that can be a way we can offer a simplified core that with pip can be installed with `pip install hls4ml[option1,option2...]` and for conda we simply say "have these dependencies for this option" and be done with it. I think conda also has optional features support, but if we can't find the dependencies for all optional features, the previous instruction should work.

jmduarte, 2023-05-12:
@vloncar I agree having these libaries as "optional/extra" dependencies is a good idea moving forward and would resolve to 1st order the conda-forge issues because we could just say people would need to use pip install to get those optional dependencies.