fastmachinelearning/hls4ml #859: Add Resource, io_parallel, depth_multiplier>1 and Quartus support for (Q)DepthwiseConv

jmitrevs, 2023-08-29:
## Details
As discussed in #834, (Q)DepthwiseConv2D and DepthwiseConv1D only exist for `io_stream` and `Latency` strategy for Vivado and Vitis. Also only `depth_multiplier=1` is supported.

### New behavior

We should add support for `Resource` strategy, `io_parallel`, `depth_multiplier>1` and Quartus support for (Q)DepthwiseConv.

### Motivation
This will provide better support for depthwise convolution.

### Parts of hls4ml being affected
The separable and depthwise convolution implementations.

steltze, 2024-04-12:
hi @jmitrevs @vloncar I am interested in the `Resource` strategy for DepthwiseConv2D as I want to use it for Separable convolution layers. Anything in mind on how to implement it?

vloncar, 2024-04-12:
You would need to implement a `depthwise_product(...)` for resource strategy in `nnet_sepconv_stream.h` that currently is hardcoded to fail an assertion. This product is not the same as matrix-vector multiplication that is at the core of most layers, rather, it's a simpler elementwise product.

The current, latency-oriented implementation of `depthwise_product` was based on `dense_latency` so you can compare the two to see how it was changed. You would essentially have to do the same to produce an elementwise product that has a similar flow as `dense_resource`. This should be simpler than current `dense_resource` implementation which has three possible scenarios that may not be needed here.