fastmachinelearning/hls4ml #878: HLS4ML IN QUARTUS SUPPORT FOR CONV2D

MODISENEISO, 2023-09-28:
Quick Summary 
Attempting to convert a trained Keras Model with Quartus as Backend. but get this error
[image]

Detail
hls4ml v 0.7.1
1)The 3 layer hls4ml tutorial is able to convert to RTL with Quartus Backend.
2)The model I want to convert has Conv2D layers and Upsampling layers.
3)Vivado HLS is able to convert the same model without any errors.
4)When converting the same model with Quatus 20.1 backend the following error comes during the build p project -s step.
5)hl:~$ hls4ml build -p bratsquartus2 -s
Loading configuration from bratsquartus2/hls4ml_config.yml
Loading configuration from bratsquartus2/hls4ml_config.yml
i++   -march=Cyclone10GX --time quartus-hlssynt.log brats_test.cpp firmware/brats.cpp   -o brats-fpga
In file included from firmware/brats.cpp:2:
In file included from firmware/parameters.h:11:
firmware/nnet_utils/nnet_conv2d_stream.h:135:5: error: no matching function for call to 'shift_line_buffer_2d'
    nnet::shift_line_buffer_2d<data_T, CONFIG_T>(in_elem, line_buffer, shift_buffer);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv2d_stream.h:199:13: note: in instantiation of function template specialization 'nnet::compute_output_buffer_2d<nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 4>, config11>' requested here
            compute_output_buffer_2d<data_T, res_T, CONFIG_T>(padds, res, line_buffer, kernel_window, weights, biases);
            ^
firmware/brats.cpp:103:11: note: in instantiation of function template specialization 'nnet::conv_2d_cl<nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 4>, config11>' requested here
    nnet::conv_2d_cl<layer10_t, layer11_t, config11>(layer10_out, layer11_out, w11, b11);
          ^
firmware/nnet_utils/nnet_conv2d_stream.h:69:6: note: candidate template ignored: substitution failure [with data_T = nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, CONFIG_T = config11]: zero-length arrays are not permitted in C++
void shift_line_buffer_2d(
     ^
1 error generated.
HLS Testbench parse FAILED.
make: *** [Makefile:30: brats-fpga] Error 1

Expected behavior
Expect to get the HDL representation of the C++ model.

Current behavior

firmware/nnet_utils/nnet_conv2d_stream.h:135:5: error: no matching function for call to 'shift_line_buffer_2d'
    nnet::shift_line_buffer_2d<data_T, CONFIG_T>(in_elem, line_buffer, shift_buffer);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_conv2d_stream.h:199:13: note: in instantiation of function template specialization 'nnet::compute_output_buffer_2d<nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 4>, config11>' requested here
            compute_output_buffer_2d<data_T, res_T, CONFIG_T>(padds, res, line_buffer, kernel_window, weights, biases);
            ^
firmware/brats.cpp:103:11: note: in instantiation of function template specialization 'nnet::conv_2d_cl<nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 4>, config11>' requested here
    nnet::conv_2d_cl<layer10_t, layer11_t, config11>(layer10_out, layer11_out, w11, b11);
          ^
firmware/nnet_utils/nnet_conv2d_stream.h:69:6: note: candidate template ignored: substitution failure [with data_T = nnet::array<ac_fixed<16, 6, true, AC_TRN, AC_WRAP>, 8>, CONFIG_T = config11]: zero-length arrays are not permitted in C++
void shift_line_buffer_2d(

Possible Fix 
nnet_conv2d_stream.h template investigation for Quartus when converting conv2d layers

MODISENEISO, 2023-10-01:
I have attached project file to sysnthesize which will reproduce the error

MODISENEISO, 2023-10-09:
@vloncar can you help check

jmitrevs, 2023-10-11:
The quartus implementation seems to have issues for fliter sizes of 1.  So I confirm this issue. We will try to fix it soon.

jmitrevs, 2023-10-11:
I think we need to copy the logic like `[MAX(CONFIG_T::filt_height - 1, 1)]` from the Vivado backend:

https://github.com/fastmachinelearning/hls4ml/blob/main/hls4ml/templates/vivado/nnet_utils/nnet_conv2d_stream.h#L74

to the Quartus backend. I can take a look later today.

closed 2026-10-18 by github-actions[bot]