fastmachinelearning/hls4ml #897: SeparableConv1D fail CSynth with Vitis backend

qberthet, 2023-10-25:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.

## Quick summary
Model using SeparableConv1D fail CSynth when using Vitis backend.

### Steps to Reproduce
1. Clone the hls4ml repository
2. Checkout the main branch, with commit hash: d03573888227da9b7527ce7a2d25403640ed4637
3. Run the following script:
```python
from keras.layers import Input
from keras.models import Model
from qkeras import *
import hls4ml

def get_model():
    # Define a dummy model with only one QSeparableConv1D layer
    input_layer = Input(shape=(32, 3))
    layer = QSeparableConv1D(
                filters=16,
                kernel_size=3,
                depthwise_quantizer=quantized_bits(16, 6, alpha=1),
                pointwise_quantizer=quantized_bits(16, 6, alpha=1),
                bias_quantizer=quantized_bits(16, 6, alpha=1)
            )(input_layer)
    model = Model(inputs=input_layer, outputs=layer)
    return model

model = get_model()

model.summary()

config = hls4ml.utils.config_from_keras_model(model, granularity="name")

# Configure the project to be SeparableConv1D compatible
config['Model']['Precision'] = 'ap_fixed<16,6>'
config['Model']['ReuseFactor'] = 1
config['Model']['Strategy'] = 'Latency'

# Use the Vivado backend (2020.1)
cfg = hls4ml.converters.create_config(backend='Vitis')
cfg['IOType'] = 'io_stream'
cfg['HLSConfig'] = config
cfg['KerasModel'] = model
cfg['OutputDir'] = 'hls4ml_prj'
cfg['Part'] = 'xcku115-flvb2104-2-i'

hls_model = hls4ml.converters.keras_to_hls(cfg)

hls_model.compile()

hls_model.build(reset=True, csim=False, synth=True)
```
### Expected behavior
Successful CSynt.

### Actual behavior
CSytnh fail with the following error:
`ERROR: [HLS 214-272] In function 'void nnet::depthwise_product<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2_depthwise>(ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>*, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>*, config2_depthwise::weight_t*, config2_depthwise::bias_t*)', Pragma conflict happens on 'INLINE' and 'PIPELINE'FUNCTION_INSTANTIATE' pragmas: same function (firmware/nnet_utils/nnet_sepconv_stream.h:13:0)`

qberthet, 2024-11-12:
Vitis 2022.2 is the only version supporting the VCK5000 board and is still listed as supported by the current documentation (but I understand that this might change with next hls4ml release).
I wouldn't re-open this ticket just for this usecase tho.

JanFSchulte, 2024-11-12:
Ah, my bad, I somehow misread this and thought your issue was with 2022.1, which we are having a lot of problems with. 2022.2 seems to work with most things, but 2023.2 and 2024.1 are the versions we have tested much more. I still think in this situation we would probably not want to adopt your fix as a general solution as it is not necessary for the newer versions.