fastmachinelearning/hls4ml #900: hls_model.build() cannot not work in Jupyter

jsncepu, 2023-10-27:
Hello, I followed hls4ml tutorial to take execises. However, when I run the cell "hls_model.build(csim='false')", It reported that project myproject_prj does not exist. Rerun hls4ml build -p model1/hls4ml_prj. I try to run this command in the terminal and it works. However, for the part4 and part7a scripts, when I run this command, it reported "project configuration file not found in model_3/hls4ml_prj" or "model_3/hls4ml_prj_pynq". But I can find the hls4ml_config.yml right there. So how can I fix this problem? Thank you very much!

freaksie, 2023-10-27:
@jsncepu  May I ask which HLS backend are you using? I would suggest checking the path for that HLS backend

jsncepu, 2023-10-28:
> @jsncepu May I ask which HLS backend are you using? I would suggest checking the path for that HLS backend

I use vivado 2020.1 as backend. I can complete build process for part1 and part2 tutorials by inputing the command for example "hls4ml build -p model_1/hls4ml/prj" in the terminal. The project can be successfully built and report the csim, cosim and synth results. However, for part4 and part7a, the command doesn't work and remind me that configuration file is not found. It's weird enough.

freaksie, 2023-10-30:
```
import os
os.environ['PATH']+= os.pathsep + '/opt/Xilinx/Vivado/2020.1/bin'
```
@jsncepu  Can you try this snippet in your ipynb file. If it doesn't work, please print the value of ```os.environ['PATH']``` here.
I hope this helps.
Thank you.

jsncepu, 2023-11-04:
> ```
> import os
> os.environ['PATH']+= os.pathsep + '/opt/Xilinx/Vivado/2020.1/bin'
> ```
> 
> @jsncepu Can you try this snippet in your ipynb file, changing the path according to your vivado path? If it doesn't work, please print the value of `os.environ['PATH']` here. I hope this helps. Thank you.

Hi, I can build successfully by inputing the command vivado_hls build -f build_prj.tcl "csim=1 synth=1 cosim=1 vsynth=1" first and inputing the command vivado -stack 2000 -mode batch -source design.tcl to obtain the bitstream and hwh files etc. But I still cannot build the model from jupyter directly. But now, it doesn't matter for me.

closed 2023-11-04 by jsncepu