fastmachinelearning/hls4ml #910: hls4ml building project

abd0smaali, 2023-11-05:
Dear community ,

i am writing to ask assistance with a problem I've encountered while trying to build a project using 'hls4ml'. I am using Ubuntu 20.04 and Vivado 2019.2, and I am facing the following issue:

When I run the command:
`hls4ml build -p my-hls-test -a`
I encounter an error during the build process. Specifically, the error message :
`(base) abdo@abdo:~/hls4ml/example-models$ hls4ml build -p my-hls-test -a
2023-11-05 11:55:13.476159: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-11-05 11:55:13.581885: E tensorflow/compiler/xla/stream_executor/cuda/cuda_dnn.cc:9342] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2023-11-05 11:55:13.581942: E tensorflow/compiler/xla/stream_executor/cuda/cuda_fft.cc:609] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2023-11-05 11:55:13.581997: E tensorflow/compiler/xla/stream_executor/cuda/cuda_blas.cc:1518] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2023-11-05 11:55:13.591529: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-11-05 11:55:13.591796: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-11-05 11:55:15.192767: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Loading configuration from my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /home/abdo/Documents/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/home/abdo/Documents/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'abdo' on host 'abdo' (Linux_x86_64 version 6.2.0-36-generic) on Sun Nov 05 11:55:28 GMT 2023
INFO: [HLS 200-10] On os Ubuntu 22.04.2 LTS
INFO: [HLS 200-10] In directory '/home/abdo/hls4ml/example-models/my-hls-test'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/home/abdo/hls4ml/example-models/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/abdo/hls4ml/example-models/my-hls-test/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 80.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 80.
ERROR: [HLS 200-70] Part 'None' is not installed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Sun Nov  5 11:55:32 2023...`

I have checked my Vivado HLS installation and the project settings, but I am unable to resolve this issue on my own. I would greatly appreciate your help in identifying the cause of the problem and providing guidance on how to resolve it.

Please let me know if you require any additional information or logs to better understand the issue. I am eager to get this project built successfully, and your assistance would be invaluable.

Thank you for your time and support. I look forward to your response.

Sincerely,
abderrahmane smaali

calad0i, 2023-11-05:
` ERROR: [HLS 200-70] Part 'None' is not installed.`
Check  `project.tcl` under your project directory. There should be a line like `set part "xcku115-flvb2104-2-i"` to specify the part number. If not, you probably didn't specify a part number when creating the project.

abd0smaali, 2023-11-06:
thank you  @calad0i for helping me with my previous issue; it worked for me. However, I have encountered a new problem related to the synthesis of my model.

It appears that my model is not synthesizable. If you are aware of any references or resources that I can use to build a model that is compatible with 'hls4ml,' I would greatly appreciate your guidance.

Thank you for your time and support. I look forward to your response.

below the files gerate by hls4ml framework:[lenet5-hls-test.tar.gz](https://github.com/fastmachinelearning/hls4ml/files/13270247/lenet5-hls-test.tar.gz)

calad0i, 2023-11-06:
Dense layer of kernel size `25088` exceeds the unroll limit for vivado (hard coded 4096). Reference to #904.

closed 2023-11-12 by abd0smaali