fastmachinelearning/hls4ml #922: [XFORM 203-502]

abd0smaali, 2023-11-13:
hello , i want to test  hls4ml  with lenet5 model but  i m faced with some problem ,my model respect the number of parameter by layer <4096, now when i run  `hls4ml build -p lenet5-hls-test -a`  the c sim run the terminal show this message:
```
e)5, (ap_o_mode)3, 0>, 10u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 10u>, softmax_config28>' (firmware/nnet_utils/nnet_activation_stream.h:235) automatically.
INFO: [HLS 200-111] Finished Checking Synthesizability Time (s): cpu = 00:00:37 ; elapsed = 00:00:41 . Memory (MB): peak = 957.395 ; gain = 585.035 ; free physical = 125 ; free virtual = 4608
INFO: [XFORM 203-1101] Packing variable 'out_pack.data.V' (firmware/nnet_utils/nnet_activation_stream.h:237) into a 160-bit variable.
INFO: [XFORM 203-1101] Packing variable 'out_data.data.V' (firmware/nnet_utils/nnet_activation_stream.h:45) into a 1024-bit variable.
INFO: [XFORM 203-1101] Packing variable 'out_data.data.V' (firmware/nnet_utils/nnet_activation_stream.h:45) into a 672-bit variable.
INFO: [XFORM 203-1101] Packing variable 'out_data.data.V' (firmware/nnet_utils/nnet_activation_stream.h:45) into a 384-bit variable.
INFO: [XFORM 203-1101] Packing variable 'out_data.data.V' (firmware/nnet_utils/nnet_activation_stream.h:45) into a 256-bit variable.
INFO: [XFORM 203-1101] Packing variable 'out_data.data.V' (firmware/nnet_utils/nnet_activation_stream.h:45) into a 256-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_pooling_stream.h:184) into a 384-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_pooling_stream.h:184) into a 256-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_pooling_stream.h:184) into a 256-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_dense_stream.h:55) into a 160-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_dense_stream.h:55) into a 1024-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_dense_stream.h:55) into a 672-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_conv_stream.h:282) into a 384-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_conv_stream.h:282) into a 256-bit variable.
INFO: [XFORM 203-1101] Packing variable 'res_pack.data.V' (firmware/nnet_utils/nnet_conv_stream.h:282) into a 256-bit variable.
WARNING: [XFORM 203-505] Ignore pipeline pragma in Loop whose tripcount is only 1 (firmware/nnet_utils/nnet_conv_stream.h:195) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, config16>'.
WARNING: [XFORM 203-505] Ignore pipeline pragma in Loop whose tripcount is only 1 (firmware/nnet_utils/nnet_conv_stream.h:195) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, config6>'.
WARNING: [XFORM 203-505] Ignore pipeline pragma in Loop whose tripcount is only 1 (firmware/nnet_utils/nnet_conv_stream.h:195) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, config11>'.
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::softmax_stable<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 10u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 10u>, softmax_config28>' (firmware/nnet_utils/nnet_activation_stream.h:193:47).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::relu<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 64u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 64u>, relu_config25>' (firmware/nnet_utils/nnet_activation_stream.h:41:42).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::relu<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 42u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 42u>, relu_config21>' (firmware/nnet_utils/nnet_activation_stream.h:41:42).
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReLUActLoop' (firmware/nnet_utils/nnet_activation_stream.h:41) in function 'nnet::relu<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, relu_config15>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReLUActLoop' (firmware/nnet_utils/nnet_activation_stream.h:41) in function 'nnet::relu<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, relu_config5>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReLUActLoop' (firmware/nnet_utils/nnet_activation_stream.h:41) in function 'nnet::relu<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, relu_config10>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReadInputWidth' (firmware/nnet_utils/nnet_pooling_stream.h:243) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, config16>' for pipelining.
WARNING: [XFORM 203-505] Ignored pipeline directive for loop 'FiltLoop' (firmware/nnet_utils/nnet_pooling_stream.h:193) because its parent loop or function is pipelined.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReadInputWidth' (firmware/nnet_utils/nnet_pooling_stream.h:243) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, config6>' for pipelining.
WARNING: [XFORM 203-505] Ignored pipeline directive for loop 'FiltLoop' (firmware/nnet_utils/nnet_pooling_stream.h:193) because its parent loop or function is pipelined.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReadInputWidth' (firmware/nnet_utils/nnet_pooling_stream.h:243) in function 'nnet::pooling2d_cl<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, config11>' for pipelining.
WARNING: [XFORM 203-505] Ignored pipeline directive for loop 'FiltLoop' (firmware/nnet_utils/nnet_pooling_stream.h:193) because its parent loop or function is pipelined.
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::dense_wrapper<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config26>' (firmware/nnet_utils/nnet_dense_latency.h:17:48).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::dense_wrapper<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config22>' (firmware/nnet_utils/nnet_dense_latency.h:17:48).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::dense_wrapper<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config18>' (firmware/nnet_utils/nnet_dense_latency.h:17:48).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::compute_output_buffer_2d<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 24u>, config12>' (firmware/nnet_utils/nnet_conv_stream.h:263:5).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config12_mult>' (firmware/nnet_utils/nnet_dense_latency.h:17:48).
INFO: [XFORM 203-502] Unrolling all loops for pipelining in function 'nnet::shift_line_buffer<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 16u>, config12>' (firmware/nnet_utils/nnet_conv_stream.h:194:62).
WARNING: [XFORM 203-505] Ignored pipeline directive for loop 'KernelShiftWidth' (firmware/nnet_utils/nnet_conv_stream.h:194) because its parent loop or function is pipelined.
[... cut ...]
_________________________________________________________________
output_softmax (Activation)  (None, 10)                0
=================================================================
Total params: 11,848
Trainable params: 11,560
Non-trainable params: 288
_________________________________________________________________
```

the cmd used in terminal :
1:
/hls4ml convert -c co_nf.yml
2:
/hls4ml build -p lenet5-hls-test -a

[lenet5-hls-test.tar.gz](https://github.com/fastmachinelearning/hls4ml/files/13329415/lenet5-hls-test.tar.gz)

vloncar, 2023-11-13:
`ReuseFactor: 1` -> everything will be unrolled, every multiplication will use a separate DSP. You don't have that many DSPs on your board. Not a bug.

abd0smaali, 2023-11-15:
ther is any method hel to chose the best ReuseFactor

abd0smaali, 2023-11-19:
@vloncar i solve the problem when i fixed  the `ReuseFactor: 1`  and i add the  `Strategy: Resource` and this cmd `/hls4ml build -p lenet5-hls-test -a` work fine,
after i see the  [hls4ml-tutorial](https://github.com/fastmachinelearning/hls4ml-tutorial/tree/main)/part7a_bitstream.ipynb i try to
 build the bit stream by changing the config in yaml file 
as shown below :
```
KerasJson: LeNet5_MNIST_model.json
KerasH5:   LeNet5_MNIST_weights.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: lenet5-hls-test1
ProjectName: lenet5
board: pynq-z2
Part: xc7z020clg400-1
ClockPeriod: 5
Backend: VivadoAccelerator
Interface: axi_stream
IOType: io_stream
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 1
    Strategy: Resource
#  LayerType:
#    Dense:
#      ReuseFactor: 2
#      Strategy: Resource
#      Compression: True
```
and this is the  out put of terminal :
```
(base) abdo@abdo:~/PycharmProjects/lenet5/qkeras$ hls4ml convert -c co_nf.yml
2023-11-19 18:44:55.190326: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-11-19 18:44:55.285062: E tensorflow/compiler/xla/stream_executor/cuda/cuda_dnn.cc:9342] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2023-11-19 18:44:55.285098: E tensorflow/compiler/xla/stream_executor/cuda/cuda_fft.cc:609] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2023-11-19 18:44:55.285189: E tensorflow/compiler/xla/stream_executor/cuda/cuda_blas.cc:1518] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2023-11-19 18:44:55.308195: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-11-19 18:44:55.308453: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-11-19 18:44:56.599776: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Loading configuration from co_nf.yml
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, input shapes: [[None, 28, 28, 1]], output shape: [None, 28, 28, 1]
Layer name: conv_0, layer type: Conv2D, input shapes: [[None, 28, 28, 1]], output shape: [None, 26, 26, 16]
Layer name: bn_conv_0, layer type: BatchNormalization, input shapes: [[None, 26, 26, 16]], output shape: [None, 26, 26, 16]
Layer name: conv_act_0, layer type: Activation, input shapes: [[None, 26, 26, 16]], output shape: [None, 26, 26, 16]
Layer name: pool_0, layer type: MaxPooling2D, input shapes: [[None, 26, 26, 16]], output shape: [None, 13, 13, 16]
Layer name: conv_1, layer type: Conv2D, input shapes: [[None, 13, 13, 16]], output shape: [None, 11, 11, 16]
Layer name: bn_conv_1, layer type: BatchNormalization, input shapes: [[None, 11, 11, 16]], output shape: [None, 11, 11, 16]
Layer name: conv_act_1, layer type: Activation, input shapes: [[None, 11, 11, 16]], output shape: [None, 11, 11, 16]
Layer name: pool_1, layer type: MaxPooling2D, input shapes: [[None, 11, 11, 16]], output shape: [None, 5, 5, 16]
Layer name: conv_2, layer type: Conv2D, input shapes: [[None, 5, 5, 16]], output shape: [None, 3, 3, 24]
Layer name: bn_conv_2, layer type: BatchNormalization, input shapes: [[None, 3, 3, 24]], output shape: [None, 3, 3, 24]
Layer name: conv_act_2, layer type: Activation, input shapes: [[None, 3, 3, 24]], output shape: [None, 3, 3, 24]
Layer name: pool_2, layer type: MaxPooling2D, input shapes: [[None, 3, 3, 24]], output shape: [None, 1, 1, 24]
Layer name: flatten, layer type: Reshape, input shapes: [[None, 1, 1, 24]], output shape: [None, 24]
Layer name: dense_0, layer type: Dense, input shapes: [[None, 24]], output shape: [None, 42]
Layer name: bn_dense_0, layer type: BatchNormalization, input shapes: [[None, 42]], output shape: [None, 42]
Layer name: dense_act_0, layer type: Activation, input shapes: [[None, 42]], output shape: [None, 42]
Layer name: dense_1, layer type: Dense, input shapes: [[None, 42]], output shape: [None, 64]
Layer name: bn_dense_1, layer type: BatchNormalization, input shapes: [[None, 64]], output shape: [None, 64]
Layer name: dense_act_1, layer type: Activation, input shapes: [[None, 64]], output shape: [None, 64]
Layer name: output_dense, layer type: Dense, input shapes: [[None, 64]], output shape: [None, 10]
Layer name: output_softmax, layer type: Softmax, input shapes: [[None, 10]], output shape: [None, 10]
Creating HLS model
WARNING: Changing pipeline style to "dataflow".
Writing HLS project
Done
(base) abdo@abdo:~/PycharmProjects/lenet5/qkeras$ hls4ml build -p lenet5-hls-test1 -a
2023-11-19 18:45:15.265514: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-11-19 18:45:15.318470: E tensorflow/compiler/xla/stream_executor/cuda/cuda_dnn.cc:9342] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
[... cut ...]
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-11-19 18:45:16.510641: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Loading configuration from lenet5-hls-test1/hls4ml_config.yml
Backend VivadoAccelerator does not support building projects.
(base) abdo@abdo:~/PycharmProjects/lenet5/qkeras$ 

```
i dont now how t fix this issue  : Backend VivadoAccelerator does not support building projects.
to generate the bit stirm for my  pynq z 2

abd0smaali, 2023-11-26:
hi @vloncar  I followed your advice for using the python API instead of hls4ml by terminal to build the IP  for pink z2 using vivado accelerator  back end, but I'm faced with new error messages  as I share below:
```
Creating HLS model
WARNING: Changing pipeline style to "dataflow".
Writing HLS project
/home/abdo/anaconda3/lib/python3.11/site-packages/keras/src/engine/training.py:3079: UserWarning: You are saving your model as an HDF5 file via `model.save()`. This file format is considered legacy. We recommend using instead the native Keras format, e.g. `model.save('my_model.keras')`.
  saving_api.save_model(
Done

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /home/abdo/Documents/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/home/abdo/Documents/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'abdo' on host 'abdo' (Linux_x86_64 version 6.2.0-37-generic) on Sun Nov 26 22:25:37 GMT 2023
INFO: [HLS 200-10] On os Ubuntu 22.04.2 LTS
INFO: [HLS 200-10] In directory '/home/abdo/PycharmProjects/lenet5/new test/model_hls4ml'
Sourcing Tcl script 'build_prj.tcl'
ERROR: [HLS 200-70] Project/solution path '/home/abdo/PycharmProjects/lenet5/new test/model_hls4ml/myproject_prj' contains illegal character ' '.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Sun Nov 26 22:25:37 2023...
Project myproject_prj does not exist. Rerun "hls4ml build -p model_hls4ml".

****** Vivado v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.
```
I flowed the tutorial in part6  for building a quantized CNN model the 
Total parameters in the model: 9214
```
+---------------------+------------+---------+--------+
|        Layer        | Parameters | Weights | Biases |
+---------------------+------------+---------+--------+
|       input_1       |      0     |    0    |   0    |
|   fused_convbn_0    |     208    |   200   |   8    |
|       pool_0        |      0     |    0    |   0    |
|   fused_convbn_1    |    2944    |  2880   |   64   |
|       pool_1        |      0     |    0    |   0    |
|   fused_convbn_2    |    4032    |  4000   |   32   |
|       pool_2        |      0     |    0    |   0    |
|      flatten        |      0     |    0    |   0    |
|       dense_0       |     650    |   640   |   10   |
|      bn_dense_0     |      20    |    0    |   20   |
|     dense_act_0     |      0     |    0    |   0    |
|       dense_1       |     178    |   160   |   18   |
|      bn_dense_1     |      32    |    0    |   32   |
|     dense_act_1     |      0     |    0    |   0    |
|     output_dense    |      90    |   80    |   10   |
+---------------------+------------+---------+--------+

```

[model_hls4ml.tar.gz](https://github.com/fastmachinelearning/hls4ml/files/13468311/model_hls4ml.tar.gz)

vloncar, 2023-11-27:
`Project/solution path '/home/abdo/PycharmProjects/lenet5/new test/model_hls4ml/myproject_prj' contains illegal character ' '.` explains itself. Try changing your output path to not include spaces.

abd0smaali, 2023-12-10:
ihad an new problem :
```

***** EXPORT IP COMPLETED IN 0h0m23s *****
INFO: [HLS 200-112] Total elapsed time: 1634.17 seconds; peak allocated memory: 2.233 GB.
INFO: [Common 17-206] Exiting vivado_hls at Sat Dec  2 23:28:50 2023...
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.
```
after this  i add the file board  to xilinx installation 
but  the process freez in this step :

`[Sat Dec  9 23:40:56 2023] Launched impl_1...
Run output will be captured here: /home/abdo/PycharmProjects/lenet5/qmodel/model_hls4ml/myproject_vivado_accelerator/project_1.runs/impl_1/runme.log
launch_runs: Time (s): cpu = 00:00:20 ; elapsed = 00:00:22 . Memory (MB): peak = 2101.559 ; gain = 220.098 ; free physical = 143 ; free virtual = 2401
# wait_on_run -timeout 360 impl_1
[Sat Dec  9 23:40:56 2023] Waiting for impl_1 to finish (timeout in 360 minutes)...`

closed 2026-10-17 by github-actions[bot]