fastmachinelearning/hls4ml #933: vivadoaccelerator backend : bit file note generated

abd0smaali, 2023-12-03:
## Quick summary
after i reduce my model developed based on hls4ml tutorial part 6 concerning the functions to build the CNN  and to had  the maximum  for each layer is 4096 , when I flow the part7  for configuration for pynqz2, i didn't  find  the generated bit file

## Details
configuration :
```

import hls4ml
import tensorflow as tf
from tensorflow.keras.models import load_model
from qkeras.utils import _add_supported_quantized_objects

# Load the model
co = {}
_add_supported_quantized_objects(co)
model = load_model('LeNet5_MNIST_model_n.h5', custom_objects=co)

# Convert the model to HLS using hls4ml
config = hls4ml.utils.config_from_keras_model(model, granularity='name')

config['Model']['ReuseFactor'] = 1
config['Model']['Strategy'] = 'Resource'
config['Model']['Precision'] = 'ap_fixed<16,6>'

#print("-----------------------------------")
#plotting.print_dict(config)
#print("-----------------------------------")

hls_model = hls4ml.converters.convert_from_keras_model(
    model, hls_config=config, output_dir='model_hls4ml', backend='VivadoAccelerator', board='pynq-z2',io_type='io_stream',
)

#hls_model.compile()
hls_model.build(csim=False, export=True, bitfile=True)
```
the code that i used to generate the model :
```
input_shape = (28, 28, 1)

x = x_in = Input(input_shape)

for i, f in enumerate(filters_per_conv_layer):
    print(('Adding convolutional block {} with N={} filters').format(i, f))
    x = Conv2D(
        int(f),
        kernel_size=(3, 3),
        strides=(1, 1),
        kernel_initializer='lecun_uniform',
        kernel_regularizer=l1(0.0001),
        use_bias=False,
        name='conv_{}'.format(i),
    )(x)
    x = BatchNormalization(name='bn_conv_{}'.format(i))(x)
    x = Activation('relu', name='conv_act_%i' % i)(x)
    x = MaxPooling2D(pool_size=(2, 2), name='pool_{}'.format(i))(x)
x = Flatten()(x)

for i, n in enumerate(neurons_per_dense_layer):
    print(('Adding dense block {} with N={} neurons').format(i, n))
    x = Dense(n, kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='dense_%i' % i, use_bias=False)(x)
    x = BatchNormalization(name='bn_dense_{}'.format(i))(x)
    x = Activation('relu', name='dense_act_%i' % i)(x)
x = Dense(10, name='output_dense')(x)
x_out = Activation('softmax', name='output_softmax')(x)

model = Model(inputs=[x_in], outputs=[x_out], name='LeNet5_MNIST')

# Print model summary
model.summary()
```
[... 174 lines cut ...]
## variable backend
## set backend "vivadoaccelerator"
## variable part
## set part "xc7z020clg400-1"
## variable clock_period
## set clock_period 5
## variable clock_uncertainty
## set clock_uncertainty 12.5%
## variable version
## set version "1.0.0"
## set bit_width_hls_output 32
## set bit_width_hls_input 32
# create_project project_1 ${project_name}_vivado_accelerator -part xc7z020clg400-1 -force
# set_property board_part tul.com.tw:pynq-z2:part0:1.0 [current_project]
ERROR: [Board 49-71] The board_part definition was not found for tul.com.tw:pynq-z2:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7z020clg400-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Common 17-206] Exiting Vivado at Sat Dec  2 23:29:02 2023...
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.
(base) abdo@abdo:~/PycharmProjects/lenet5/qdense$ 

```

ziped file contain the log and the model 
[log+model.zip](https://github.com/fastmachinelearning/hls4ml/files/13537707/log%2Bmodel.zip)

abd0smaali, 2023-12-03:
@calad0i Thank you for your comments, after adding the board file of the pynqz2 for the vivado installation, the warning is displayed more the process continues but it now displays a new issue  and the terminal freez . 
Here is the new problem:
```

#-----------------------------------------------------------
# Vivado v2019.2 (64-bit)
# SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
# IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
# Start of session at: Sun Dec  3 23:11:08 2023
# Process ID: 8307
# Current directory: /home/abdo/PycharmProjects/lenet5/qdense/model_hls4ml
# Command line: vivado -mode batch -source design.tcl
# Log file: /home/abdo/PycharmProjects/lenet5/qdense/model_hls4ml/vivado.log
# Journal file: /home/abdo/PycharmProjects/lenet5/qdense/model_hls4ml/vivado.jou
#-----------------------------------------------------------
WARNING: [Runs 36-547] Tool Strategy 'Vivado Synthesis Defaults' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_AreaOptimized_high' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_AreaOptimized_medium' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_AreaMultThresholdDSP' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_AlternateRoutability' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_PerfOptimized_high' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_PerfThresholdCarry' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Flow_RuntimeOptimized' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDS2019.psg'
WARNING: [Runs 36-547] Tool Strategy 'Vivado Implementation Defaults' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_Explore' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_ExplorePostRoutePhysOpt' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_RefinePlacement' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_WLBlockPlacement' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_WLBlockPlacementFanoutOpt' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_NetDelay_high' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_NetDelay_medium' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_NetDelay_low' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
WARNING: [Runs 36-547] Tool Strategy 'Performance_ExploreSLLs' from file '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg' discarded because strategy with same name already parsed from '/home/abdo/Documents/Vivado/2019.2/strategies/VDI2015.psg'
[... cut ...]
launch_runs: Time (s): cpu = 00:00:22 ; elapsed = 00:00:24 . Memory (MB): peak = 2101.559 ; gain = 220.098 ; free physical = 1730 ; free virtual = 4014
# wait_on_run -timeout 360 impl_1
[Sun Dec  3 23:11:57 2023] Waiting for impl_1 to finish (timeout in 360 minutes)...
```
also here is the vivado  and vivado hls log in the ziped file also the vivadoaccelerator project gneretaed  files :
[myproject_vivado_accelerator11.zip](https://github.com/fastmachinelearning/hls4ml/files/13540597/myproject_vivado_accelerator11.zip)

[vivado.zip](https://github.com/fastmachinelearning/hls4ml/files/13540598/vivado.zip)

abd0smaali, 2023-12-11:
even i change the model with new one nothing change stil the same  issue : wait_on_run -timeout 360 impl_1

lp247, 2024-01-20:
I pretty much get the same issue except that the process does not freeze but is actually killed in between at the step where it is freezing for you.

Seems like there is some problem with the way python is invoking the vivado command. If I execute the command manually, the one which is executed from within `hls_model.build`, everything runs fine and the bitfile will be generated.

vloncar, 2024-01-20:
If the process died -> you ran out of memory (physical + swap)
If the process is seemingly stuck/frozen or takes unreasonably long time -> you ran out of physical memory (and are using swap, which is slow and will probably die eventually)
If the command works when executed manually -> you're close to using all memory, therefore the process dies if the hls4ml environment is taking up memory (dependencies such as tensorflow can be hungry)

The solution, as you've guessed while reading the above, is of course to [download more RAM](https://downloadmoreram.com/) :wink:

yassine-safraoui, 2025-01-19:
I had the same issue of Vivado not finding the board_part I gave, but I didn't encounter any freezing( running on a 32Gb Ubuntu machine). I solved this by downloading the Pynq board files from [here](https://pynq.readthedocs.io/en/latest/overlay_design_methodology/board_settings.html?highlight=vivado#vivado-board-files) and placing them in the board_files path as mentionned in this link, that is in `<Xilinx installation directory>/Vivado/<version>/data/boards/board_files`