fastmachinelearning/hls4ml #983: Issue with ModelGraph function during ONNX model synthesis

sei-rquartiano, 2024-03-18:
On up-to-date main branch, running from within this docker [image](https://github.com/cmu-sei/hls4ml-docker/blob/main/Dockerfile)

Attempting to synthesize hls from ONNX model. First, I ran python code from README.md to convert Keras model without error. Then I modified the script to convert from ONNX instead (see below)

```
import hls4ml

# Fetch a keras model from our example repository
# This will download our example model to your working directory and return an example configuration file
config = hls4ml.utils.fetch_example_model('three_layer_keras.onnx')

# You can print the configuration to see some default parameters
print(config)

# Convert it to a hls project
hls_model = hls4ml.converters.onnx_to_hls(config)

#hls_model.build(csim=False)

# Print full list of example models if you want to explore more
hls4ml.utils.fetch_example_list()

```

When I run the above code, I get the following output and subsequent error message during synthesis 

```
2024-03-18 17:48:46.474669: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2024-03-18 17:48:47.242971: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2024-03-18 17:48:47.243534: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-03-18 17:48:49.843563: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Downloading example model files ...
{'OutputDir': 'my-hls-test', 'ProjectName': 'myproject', 'Backend': 'Vivado', 'Version': '1.0.0', 'Part': 'xcvu13p-flga2577-2-e', 'ClockPeriod': 5, 'IOType': 'io_parallel', 'HLSConfig': {'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': 1}}, 'OnnxModel': 'three_layer_keras.onnx'}
Interpreting Model ...
Output layers:  ['Softmax']
Input shape: [None, 16]
Topology:
Layer name: _class__keras_layers_core_Dense__, layer type: Dense, current shape: [[None, 16]]
Layer name: Add, layer type: Add, current shape: [[None, 64]]
Layer name: Relu, layer type: Activation, current shape: [[None, 64]]
Layer name: _class__keras_layers_core_Dense__1, layer type: Dense, current shape: [[None, 64]]
Layer name: Add1, layer type: Add, current shape: [[None, 32]]
Layer name: Relu1, layer type: Activation, current shape: [[None, 32]]
Layer name: _class__keras_layers_core_Dense__2, layer type: Dense, current shape: [[None, 32]]
Layer name: Add2, layer type: Add, current shape: [[None, 32]]
Layer name: Relu2, layer type: Activation, current shape: [[None, 32]]
Layer name: _class__keras_layers_core_Dense__3, layer type: Dense, current shape: [[None, 32]]
Layer name: Add3, layer type: Add, current shape: [[None, 5]]
Layer name: Softmax, layer type: Softmax, current shape: [[None, 5]]
Creating HLS model
Traceback (most recent call last):
  File "/home/hls4ml-user/work/hls4ml/example_script.py", line 11, in <module>
    hls_model = hls4ml.converters.onnx_to_hls(config)
  File "/home/hls4ml-user/work/hls4ml/hls4ml/converters/onnx_to_hls.py", line 318, in onnx_to_hls
    hls_model = ModelGraph(config, reader, layer_list, input_layers, output_layers)
TypeError: ModelGraph.__init__() takes from 3 to 5 positional arguments but 6 were given

```

I originally encountered this error in a different and more complex project, but reproduced it in hls4ml's example code to facilitate debugging

Thank you!

vloncar, 2024-03-18:
The current ONNX parser is broken in many ways, it is going to be updated soon. You can try out the development version in #979

sei-jgwohlbier, 2024-04-19:
@vloncar what is the expectation on the tests in https://github.com/fastmachinelearning/hls4ml/pull/979 ? I get
```
= 58 failed, 1280 passed, 6 skipped, 649 warnings, 8 errors in 3587.97s (0:59:47) =
```
In particular `test_qonnx.py` does not pass.

```
(hls4ml) hls4ml-user@lx-pinkbird:~/hls4ml/test/pytest$ python -m pytest test_qonnx.py 
============================= test session starts ==============================
platform linux -- Python 3.10.14, pytest-8.1.1, pluggy-1.4.0
rootdir: /home/hls4ml-user/hls4ml
configfile: pyproject.toml
collected 9 items                                                              

test_qonnx.py EEEEEEFFF                                                  [100%]

==================================== ERRORS ====================================
___________________ ERROR at setup of test_tfc_2w2a[Vivado] ____________________

    @pytest.fixture(scope='module')
    def tfc_2w2a_model():
        '''
        Load the tiny fully-connected model
        '''
        dl_dir = test_root_path
        dl_file = str(dl_dir / "qonnx-tfc-2w2a.onnx")
        tfc_w2a2_qonnx_url = (
            "https://raw.githubusercontent.com/fastmachinelearning/"
            "QONNX_model_zoo/main/models/MNIST/Brevitas_FINN_TFC/TFC/TFC_2W2A.onnx"
        )
        urllib.request.urlretrieve(tfc_w2a2_qonnx_url, dl_file)
        assert os.path.isfile(dl_file)
        out_file = str(dl_dir / "qonnx-tfc-2w2a-clean.onnx")
    
        # cleanup
>       qonnx.util.cleanup.cleanup(dl_file, out_file=out_file)

test_qonnx.py:35: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/util/cleanup.py:66: in cleanup
    model = cleanup_model(model, preserve_qnt_ops=preserve_qnt_ops, override_batchsize=override_batchsize)
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/util/cleanup.py:46: in cleanup_model
    model = model.transform(t)
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/core/modelwrapper.py:140: in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/transformation/fold_constants.py:108: in apply
    oxe.execute_node(n, execution_context, model.graph, opset_version=opset_version)
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/core/onnx_exec.py:87: in execute_node
    sess = rt.InferenceSession(node_model.SerializeToString())
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py:419: in __init__
    self._create_inference_session(providers, provider_options, disabled_optimizers)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <onnxruntime.capi.onnxruntime_inference_collection.InferenceSession object at 0x7ff2b2cd78b0>
providers = [], provider_options = [], disabled_optimizers = None

    def _create_inference_session(self, providers, provider_options, disabled_optimizers=None):
        available_providers = C.get_available_providers()
    
        # Tensorrt can fall back to CUDA if it's explicitly assigned. All others fall back to CPU.
        if "TensorrtExecutionProvider" in available_providers:
            if providers and any(
                provider == "CUDAExecutionProvider"
                or (isinstance(provider, tuple) and provider[0] == "CUDAExecutionProvider")
                for provider in providers
            ):
                self._fallback_providers = ["CUDAExecutionProvider", "CPUExecutionProvider"]
            else:
                self._fallback_providers = ["CPUExecutionProvider"]
        # MIGraphX can fall back to ROCM if it's explicitly assigned. All others fall back to CPU.
[... 671 lines cut ...]
  Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
    declare_namespace(pkg)

test/pytest/test_qonnx.py: 498 warnings
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/util/onnx.py:40: DeprecationWarning: `mapping.TENSOR_TYPE_TO_NP_TYPE` is now deprecated and will be removed in a future release.To silence this warning, please use `helper.tensor_dtype_to_np_dtype` instead.
    return np.zeros(dims, dtype=onnx.mapping.TENSOR_TYPE_TO_NP_TYPE[vi.type.tensor_type.elem_type])

test/pytest/test_qonnx.py::test_jet_tagging[Vivado]
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/core/modelwrapper.py:93: UserWarning: Some old-style domain attributes were automatically converted to new-style,
                  i.e. domain=finn to domain=qonnx.custom_op.<general|fpgadataflow|...>
    warnings.warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=========================== short test summary info ============================
FAILED test_qonnx.py::test_jet_tagging[Vivado] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
FAILED test_qonnx.py::test_jet_tagging[Vitis] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
FAILED test_qonnx.py::test_jet_tagging[Quartus] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
[... cut ...]
ERROR test_qonnx.py::test_tfc_2w2a[Quartus] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
ERROR test_qonnx.py::test_cnv_2w2a[Vivado] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
ERROR test_qonnx.py::test_cnv_2w2a[Vitis] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
ERROR test_qonnx.py::test_cnv_2w2a[Quartus] - onnxruntime.capi.onnxruntime_pybind11_state.InvalidArgument: [ONNXRuntimeEr...
================== 3 failed, 503 warnings, 6 errors in 5.61s ===================
```

sei-jgwohlbier, 2024-04-19:
Installing `onnx==1.15` and `onnxruntime==1.17` seems to have cleared those `onnxruntime` errors.

sei-jgwohlbier, 2024-04-19:
After clearing that issue, I do get one failure in `test_qonnx.py`, seen below.

```
(hls4ml) hls4ml-user@lx-pinkbird:~/hls4ml/test/pytest$ python -m pytest test_qonnx.py 
============================= test session starts ==============================
platform linux -- Python 3.10.14, pytest-8.1.1, pluggy-1.4.0
rootdir: /home/hls4ml-user/hls4ml
configfile: pyproject.toml
collected 9 items                                                              

test_qonnx.py .....F...                                                  [100%]

=================================== FAILURES ===================================
____________________________ test_cnv_2w2a[Quartus] ____________________________

cnv_2w2a_model = <qonnx.core.modelwrapper.ModelWrapper object at 0x7fce0ad28c70>
backend = 'Quartus'

    @pytest.mark.parametrize('backend', ['Vivado', 'Vitis', 'Quartus'])
    def test_cnv_2w2a(cnv_2w2a_model, backend):
        """
        This tests a convolution model. Note:  the batch normalizations weights not quantized, so it
        is difficult to make this match perfectly. It is also a slow test.
        """
        model = cnv_2w2a_model
    
        ishape = (1, 32, 32, 3)
        X = np.random.uniform(low=-1, high=+1, size=np.prod(ishape)).reshape(ishape)
        X = (np.round(X * 2**6) * 2**-6).astype(np.float32)
        idict = {model.graph.input[0].name: X}
        y_qonnx = oxe.execute_onnx(model, idict)[model.graph.output[0].name]
    
        # Convert QONNX model, compile, and run inference
        config = hls4ml.utils.config_from_onnx_model(model, backend=backend, default_precision='fixed<32,6>')
        hls_model = hls4ml.converters.convert_from_onnx_model(
            model,
            output_dir=str(test_root_path / f'hls4mlprj_qonnx_cnv-2w2a_{backend}'),
            io_type='io_stream',
            backend=backend,
            hls_config=config,
        )
>       hls_model.compile()

test_qonnx.py:131: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/hls4ml/model/graph.py:672: in compile
    self._compile()
../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/hls4ml/model/graph.py:691: in _compile
    self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)
../../../miniconda3/envs/hls4ml/lib/python3.10/ctypes/__init__.py:452: in LoadLibrary
    return self._dlltype(name)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <CDLL '/home/hls4ml-user/hls4ml/test/pytest/hls4mlprj_qonnx_cnv-2w2a_Quartus/firmware/myproject-D9B7bA7c.so', handle 0 at 0x7fce0a534f40>
name = '/home/hls4ml-user/hls4ml/test/pytest/hls4mlprj_qonnx_cnv-2w2a_Quartus/firmware/myproject-D9B7bA7c.so'
mode = 0, handle = None, use_errno = False, use_last_error = False
winmode = None

    def __init__(self, name, mode=DEFAULT_MODE, handle=None,
                 use_errno=False,
                 use_last_error=False,
                 winmode=None):
        self._name = name
        flags = self._func_flags_
        if use_errno:
            flags |= _FUNCFLAG_USE_ERRNO
        if use_last_error:
            flags |= _FUNCFLAG_USE_LASTERROR
        if _sys.platform.startswith("aix"):
            """When the name contains ".a(" and ends with ")",
[... 138 lines cut ...]
    distutils.version.LooseVersion(required_tensorflow_version)):

../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/calmjs/parse/utils.py:13
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/calmjs/parse/utils.py:13: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
    from pkg_resources import working_set

../../../miniconda3/envs/hls4ml/lib/python3.10/site-packages/pkg_resources/__init__.py:2871
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('calmjs')`.
  Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
    declare_namespace(pkg)

test/pytest/test_qonnx.py: 2101 warnings
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/util/onnx.py:40: DeprecationWarning: `mapping.TENSOR_TYPE_TO_NP_TYPE` is now deprecated and will be removed in a future release.To silence this warning, please use `helper.tensor_dtype_to_np_dtype` instead.
    return np.zeros(dims, dtype=onnx.mapping.TENSOR_TYPE_TO_NP_TYPE[vi.type.tensor_type.elem_type])

test/pytest/test_qonnx.py::test_jet_tagging[Vivado]
  /home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/qonnx/core/modelwrapper.py:93: UserWarning: Some old-style domain attributes were automatically converted to new-style,
                  i.e. domain=finn to domain=qonnx.custom_op.<general|fpgadataflow|...>
    warnings.warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=========================== short test summary info ============================
FAILED test_qonnx.py::test_cnv_2w2a[Quartus] - OSError: /home/hls4ml-user/hls4ml/test/pytest/hls4mlprj_qonnx_cnv-2w2a_Quar...
=========== 1 failed, 8 passed, 2106 warnings in 3564.98s (0:59:24) ============
```