fastmachinelearning/hls4ml #1013: CNN Synthesis fail : ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:37) in function 'nnet::dense_latency<ap_fixed<2, 1, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<2, 1, (ap_q_mode)5, (ap_o_mode)3, 0>, config6>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body. ERROR: [HLS 200-70] Pre-synthesis failed.

ZWhimsi, 2024-05-14:
## Environment

- **OS**: Ubuntu 16.04.07
- **Python**: 3.9.12
- **hls4ml**: 0.8.1

For all the pip dependencies, refer to the env.txt file.
[env.txt](https://github.com/fastmachinelearning/hls4ml/files/15305807/env.txt)

## Quick Summary

I attempted to synthesize some CNN models and all attempts failed. After removing layers one by one, I identified Conv2D as the problematic layer due to excessive memory usage and large runtime. MLPs, however, work perfectly. I attempted synthesis with all NN configurations using the following scripts.
[Test_script.txt](https://github.com/fastmachinelearning/hls4ml/files/15305842/Test_script.txt)
[example.py.txt](https://github.com/fastmachinelearning/hls4ml/files/15306103/example.py.txt)

## Details

### Steps to Reproduce

**Warning:** All provided files are in .txt format; you will need to change the extension to .py or .yml for the process to work. Also, remember to modify the path to your own file locations.

To replicate the bug, follow these steps:

1. Clone the hls4ml repository from GitHub.
2. Checkout the master branch with the specified commit hash.
3. Install all dependencies from `env.txt`.
4. Rename `Test_script.txt` to `Test_script.py` and execute it to create the models to be synthesized.
5. Modify the `yaml.txt` configuration file to include the CNN model you wish to synthesize.
6. Run the `example.py` script to attempt synthesis.

# Step 1: Install dependencies from env.txt
```shell
pip install -r env.txt
```

# Step 2: Execute Test_script to create the models
```shell
python Test_script.py
```
# Step 3: Update yml configuration to include the CNN model

Update this yaml to try the desired NN ( H5 + JSON) : 
[yaml.txt](https://github.com/fastmachinelearning/hls4ml/files/15306049/yaml.txt)

# Step 4: Run the example script
```shell

python example.py
```

### Expected behavior

It should synthetize the CNN.

### Actual behavior
```shell
(hls4ml_env) pierric@pierric-X542UAR:~/Bureau/Stage_2024/Scripts_python$ python example.py
2024-05-14 09:41:33.489814: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-05-14 09:41:34.586838: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
{'Backend': 'Vivado', 'ClockPeriod': 5, 'HLSConfig': {'Model': {'Precision': 'ap_fixed<2,1>', 'ReuseFactor': 1000000, 'Strategy': 'Resources'}}, 'IOType': 'io_parallel', 'KerasJSON': '../models_test/model_simple_cnn_cifar10.json', 'KerasH5': '../models_test/model_simple_cnn_cifar10.h5', 'OutputDir': '../my-hls-test', 'Part': 'xczu9eg-ffvb1156-2-e', 'ProjectName': 'myproject', 'Stamp': 'eAAc27c6', 'Version': '1.0.0'}
{'Backend': 'Vivado', 'ClockPeriod': 5, 'HLSConfig': {'Model': {'Precision': 'ap_fixed<2,1>', 'ReuseFactor': 1000000, 'Strategy': 'Resources'}}, 'IOType': 'io_parallel', 'KerasJSON': '../models_test/model_simple_cnn_cifar10.json', 'KerasH5': '../models_test/model_simple_cnn_cifar10.h5', 'OutputDir': '../my-hls-test', 'Part': 'xczu9eg-ffvb1156-2-e', 'ProjectName': 'myproject', 'Stamp': 'eAAc27c6', 'Version': '1.0.0'}
Interpreting Sequential
Topology:
Layer name: input_layer_6, layer type: InputLayer, input shapes: [None], output shape: [None, 32, 32, 3]
Layer name: conv2d_8, layer type: Conv2D, input shapes: [[None, 32, 32, 3]], output shape: [None, 30, 30, 32]
Layer name: max_pooling2d_8, layer type: MaxPooling2D, input shapes: [[None, 30, 30, 32]], output shape: [None, 15, 15, 32]
Layer name: flatten_6, layer type: Reshape, input shapes: [[None, 15, 15, 32]], output shape: [None, 7200]
Layer name: dense_11, layer type: Dense, input shapes: [[None, 7200]], output shape: [None, 10]
Creating HLS model
[... 120 lines cut ...]
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.
Found 1 solution(s) in ../my-hls-test/myproject_prj.
Reports for solution "solution1":

C SIMULATION RESULT:
INFO: [SIM 2] *************** CSIM start ***************
INFO: [SIM 4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
INFO: Unable to open input/predictions file, using default input.
-0.5 -0.5 -0.5 -0.5 -0.5 -0.5 -0.5 -0.5 -0.5 -0.5
INFO: Saved inference results to file: tb_data/csim_results.log
INFO: [SIM 1] CSim done with 0 errors.
INFO: [SIM 3] *************** CSIM finish ***************

Synthesis report not found.
Co-simulation report not found.
```

### Possible fix

I saw someone talking about hls4ml version 0.3.0 who claimed to solve a similar problem but didn't try it because i expect this repository to be stable.

bo3z, 2024-05-14:
This is not a bug - when loops are unrolled, operations are parallelised. However, there is a limit to how much can be executed in parallel (due to available resources, critical path and sometimes compiler issues with scheduling). 

To address this problem, use the Resource strategy and consider increasing to reuse factor (corresponding to the less parallelism). For more details on the reuse factor see the tutorial: https://github.com/fastmachinelearning/hls4ml-tutorial/blob/main/part2_advanced_config.ipynb

ZWhimsi, 2024-05-14:
Hi, thank you for your response.

However, I encountered  is this issue with a reuse factor of 1,000,000, using the strategy "Resources" and an approximately 10,000-parameter CNN. If these settings are not viable, what can I do to resolve this issue?

bo3z, 2024-05-14:
Make sure to use `io_stream` as the `io_type` and inspect the config YAML and parameters.h/defines.h files to ensure that the properties you set (reuse factor, strategy) propagated and were set. A 10,000 parameter model should work.

bo3z, 2024-05-14:
Furthermore, there is a typo - the strategy is `Resource` not `Resources` :)

ZWhimsi, 2024-05-14:
Yes, my bad. I corrected it. Is there a way to estimate the complexity of the process for a network to avoid attempting 'Latency' mode when it is clearly infeasible? A 10K-parameter CNN is relatively small, and I'd like to understand what kind of network would require caution when unrolling and optimizing for performance.

bo3z, 2024-05-14:
The HLS will fail automatically when there are more than 4,096 unrolls (i.e. parallel multiplications). This parameter is hard-coded and generally not exposed in hls4ml. It was determined heuristically - the conclusion was that after 4,096 the HLS compiler has issue with scheduling and completing the compilation. Now you can modify this variable in build_project.tcl. There are cases where 8,192 or 16,384 could compile (but last longer). 

So as long as every layer has less multiplications than the threshold, you will not get this error. Looking at the error you posted, the last dense layer has 7,200 parameters. In this case you can either:

1. Try increasing the factor from 4,096 to 8,192 in build_project.tcl OR
2. Keep `Latency` for all layers except for the dense layer, `dense_11`, where you can use `Resource` and set the reuse factor to 2 for example (this should already be okay, but keep in mind the number of DSPs is limited, if using these). For more information on variable tuning of layers see:  https://github.com/fastmachinelearning/hls4ml-tutorial/blob/main/part7a_bitstream.ipynb. You can modify the strategy, precision and reuse factor for each layer individually.

ZWhimsi, 2024-05-14:
Thanks for your help !

ZWhimsi, 2024-05-14:
The reuse factor is capped at the number of parameters for each layer, correct? I have valid reuse factors ranging from 1, 2, ..., to 18432 for a conv2D layer (15*15*32 in 13*13*64 out). However, why is 1 considered a valid reuse factor if the number of parameters in the layer divided by the reuse factor is greater than 4,096?

ZWhimsi, 2024-05-14:
Additionally, I'm curious why even with a large reuse factor like 1000000, the synthesis fails when in "Latency" mode. Wouldn't it be possible to select the maximum reuse factor for each layer instead?"

bo3z, 2024-05-14:
> The reuse factor is capped at the number of parameters for each layer, correct? I have valid reuse factors ranging from 1, 2, ..., to 18432 for a conv2D layer (15,15,32) in (13,13,64) out. However, [...]

'valid' means it has to be divide the number of parameters/multiplications in the layer - this is to ensure QoR (otherwise the last clock cycle will have less multiplications than others and this will lead to unequal load, unused hardware, scheduling overhead etc. The idea is that in every clock cycle you do the same number of multiplications as total_layer_mult / layer_reuse_factor). 

On the other hand, 4,096 is a Vivado HLS concept (not hls4ml, so if you use a different backend, e.g. Quartus targeting Intel boards it might have a different limit). So when you unroll more than 4,096 it is still valid it just means the HLS compiler (not hls4ml) has issues scheduling such high parallelism. 

> Additionally, I'm curious why even with a large reuse factor like 1000000, the synthesis fails when in "Latency" mode. Wouldn't it be possible to select the maximum reuse factor for each layer instead?"

Latency has a different implementation. All the loops are unrolled but the multiplications are limited using another pragma. Therefore, the design will be scheduled such that it completes in reuse_factor clock cycles (plus some constant offset), but the loops are treated as unrolled in the HLS code.

ZWhimsi, 2024-05-14:
Thank you! Can you provide more details on how the Latency mode works? I read the paper and the documentation but didn't find the information I was looking for. Is there a place where I can find answers without having to delve deeply into the code?

bo3z, 2024-05-14:
Unfortunately, there is no more documentation than the tutorials, website and papers. 

Please have a look here: https://github.com/fastmachinelearning/hls4ml/blob/main/hls4ml/templates/vivado/nnet_utils/nnet_dense_latency.h. The code is short and the most important part of it is the pragma: `pragma HLS allocation` (https://docs.amd.com/r/en-US/ug1399-vitis-hls/pragma-HLS-allocation). 

In the Latency strategy, all model weights are stored in registers and all the loops (there are only 2) are fully unrolled. However, the implementation still takes more than one clock cycle, determined by the reuse factor. Typically the latency is reuse_factor clock cycles + some overhead (invoking the layer, adding bias, typically 2 clock cycles)

In this case, the number of parallel multiplications is limited through the `pragma HLS allocation`, so that at any time, at most `n_in * n_out / reuse_factor` multiplications occur in parallel. However, since the loops are unrolled and the weight are in registers, the HLS compiler will aim to balance expressions, to minimise latency/resources. Therefore, there is no guarantee of the order in which the weights will be processed.

Importantly: Both convolutional and recurrent layers are implemented using (require) matrix multiplication, so they build on top of the dense layers (which are essentially matrix multiplication)

ZWhimsi, 2024-05-16:
I am working on a script designed to forecast FPGA resource usage (specifically DSPs and BRAMs) for HLS4ML. The goal is to estimate resource usage and throughput without synthesizing every model in HLS. My script also checks each layer to see if the number of parameters and multiply-accumulate operations (MACCs) divided by the reuse factor exceeds 4096, issuing a warning if this threshold is surpassed.

However, when I run this script on CNNs with 10k, 100k, and 1M parameters that fail to synthesize due to extensive unrolling, I do not receive any warnings. The only successful synthesis method I've found is using IO streams and resource constraints.

Can you help me understand what I might be missing in my script that causes it to miss these warnings? Additionally, I would appreciate any suggestions or improvements to enhance my script.

``` python 
import argparse
import json
import numpy as np
import torch
from tensorflow.keras.models import model_from_json, load_model
from tensorflow.keras.layers import Dense, Conv2D, Flatten, LSTM, BatchNormalization, GRU, Activation, MaxPooling2D
import math
from tabulate import tabulate
from colorama import Fore, Style

def format_table_with_colors(data):
    """
    Formats and colors the data table for better visibility when displayed.
    Uses Colorama library colors to highlight keys.
    """
    formatted_data = []
    for key, value in data.items():
        formatted_data.append([Fore.GREEN + key + Style.RESET_ALL, value])
    return formatted_data

# Configuration dictionary to handle all global settings
config = {
    "freq_mhz": 250,
    "DSP_max": 2520,
    "DSP_used": 0,
    "reuse_factor": 2,
    "mem_bandwidth_gbps": 10,
    "BRAM_capacity_bits": 1024 * 36,
    "BRAM_used": 0,
    "BRAM_max": 32.1 * 10**6,
    "weight_bits": 32,
    "info_layer": {},
    "throughput": float('inf'),
    "batch_size": 32,
    "io_type": "stream"
}

def estimations_keras(model):
    """
    Estimates FPGA resource needs and performance metrics for each layer in a Keras model.
    Calculates DSP and BRAM requirements and adjusts throughput based on resource utilization.
    """
    total_macc = sum(macc_per_layer_keras(layer) for layer in model.layers)
    for layer in model.layers:
        if config["DSP_used"] < config["DSP_max"]:
            DSP_needed = math.ceil(config["info_layer"][layer.name]['macc'] / config["reuse_factor"])
            config["DSP_used"] += DSP_needed
            if config["info_layer"][layer.name]['macc'] != 0:
                new_throughput = math.floor(DSP_needed * config["freq_mhz"] * 10**6 / config["info_layer"][layer.name]['macc'])
                if config["throughput"] > new_throughput:
                    config["throughput"] = new_throughput

        if config["BRAM_used"] < config["BRAM_max"] / config["BRAM_capacity_bits"]:
            config["BRAM_used"] += math.ceil(config["info_layer"][layer.name]['params'] * config["weight_bits"] / config["BRAM_capacity_bits"])
            if config["io_type"] == "stream":
                config["BRAM_used"] += math.ceil(np.prod(layer.output.shape[1:3]) * config["weight_bits"] / (config["BRAM_capacity_bits"] * config["reuse_factor"]))

    formatted_table = format_table_with_colors({'Throughput': config["throughput"], 'DSP': config["DSP_used"], 'BRAM': config["BRAM_used"]})
    print(tabulate(formatted_table))

def macc_per_layer_keras(layer):
    """
    Calculates the multiply-accumulate operations (MACCs) for each type of layer in a Keras model.
[... 54 lines cut ...]
    parser.add_argument("--reuse_factor", type=int, default=2, help="Number of DSP reuses")
    parser.add_argument("--frequency", type=int, default=250, help="Operating frequency in MHz")
    parser.add_argument("--io_type", default="stream", help="IO type: stream or parallel")
    args = parser.parse_args()

    # Example of how to load and use the model depending on the framework
    if args.framework == "keras":
        model = model_from_json(open(args.model_file + ".json").read())
        try:
            model.load_weights(args.model_file + ".h5")
        except OSError:
            model.load_weights(args.model_file + "_weights.h5")
        model.summary()
        config.update({
            "batch_size": args.batch_size,
            "reuse_factor": args.reuse_factor,
            "freq_mhz": args.frequency,
            "weight_bits": args.quantization,
            "io_type": args.io_type
        })
        estimations_keras(model)
    else:
        # Placeholder for PyTorch functionality
        print("PyTorch estimation functionality not implemented.")
```

sei-jgwohlbier, 2024-07-09:
I am also hitting this problem. My model is PyTorch with a single `Conv2d` layer. I can provide more code if needed.

Output from `hls4ml` when setting `ReuseFactor=6` and `Strategy=Resource`:
```
ResNet(
  (convIn): Conv2d(2, 16, kernel_size=(3, 1), stride=(1, 1), padding=(1, 0))
)
{'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': 4, 'InputsChannelLast': False, 'TransposeOutputs': False, 'Strategy': 'Resource'}, 'LayerName': {'conv_in': {'ReuseFactor': 6, 'pipeline': 'dataflow', 'Strategy': 'Resource'}}}
Interpreting Model ...
Topology:
Layer name: conv_in, layer type: Conv2D, input shape: [[None, 2, 1024, 1]]
Creating HLS model
WARNING: Strategy for layer conv_in set to "Resource", while pipeline style set to "pipeline".
WARNING: Changing pipeline style to "dataflow".
[<hls4ml.backends.fpga.fpga_backend.VivadoInput object at 0x7f404455ce80>, <hls4ml.backends.fpga.fpga_backend.VivadoTranspose object at 0x7f404455e7a0>, <hls4ml.backends.fpga.fpga_backend.VivadoConv2D object at 0x7f404455d060>]
Writing HLS project
Done
```

Error during synthesis. It doesn't seem like the `ReuseFactor` is being applied?
```
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:37:75
WARNING: [HLS 214-113] Either use an argument of the function or declare the variable inside the dataflow loop body: firmware/myproject.cpp:37:79
WARNING: [HLS 200-471] Dataflow form checks found 2 issue(s) in file firmware/myproject.cpp
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:00:28 ; elapsed = 00:00:30 . Memory (MB): peak = 959.941 ; gain = 530.133 ; free physical = 141217 ; free virtual = 166796
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:00:28 ; elapsed = 00:00:30 . Memory (MB): peak = 959.941 ; gain = 530.133 ; free physical = 141217 ; free virtual = 166796
INFO: [HLS 200-10] Starting code transformations ...
INFO: [HLS 200-489] Unrolling loop 'PixelInitAccumLoop' (firmware/nnet_utils/nnet_conv2d_resource.h:42) in function 'void nnet::conv_2d_resource_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>(FORWARD_REFERENCE*, FORWARD_REFERENCE*, FORWARD_REFERENCE::weight_t*, FORWARD_REFERENCE::bias_t*)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'PixelMultLoop' (firmware/nnet_utils/nnet_conv2d_resource.h:66) in function 'void nnet::conv_2d_resource_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>(FORWARD_REFERENCE*, FORWARD_REFERENCE*, FORWARD_REFERENCE::weight_t*, FORWARD_REFERENCE::bias_t*)' completely with a factor of 1.
INFO: [HLS 200-489] Unrolling loop 'PixelResultLoop' (firmware/nnet_utils/nnet_conv2d_resource.h:92) in function 'void nnet::conv_2d_resource_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>(FORWARD_REFERENCE*, FORWARD_REFERENCE*, FORWARD_REFERENCE::weight_t*, FORWARD_REFERENCE::bias_t*)' completely with a factor of 1.
INFO: [XFORM 203-603] Inlining function 'nnet::product::mult<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0> >::product' into 'nnet::conv_2d_resource_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' (firmware/nnet_utils/nnet_conv2d_resource.h:70).
INFO: [XFORM 203-603] Inlining function 'nnet::conv_2d_resource_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' into 'nnet::conv_2d_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' (firmware/nnet_utils/nnet_conv2d.h:52).
INFO: [HLS 200-111] Finished Standard Transforms Time (s): cpu = 00:00:36 ; elapsed = 00:00:38 . Memory (MB): peak = 959.941 ; gain = 530.133 ; free physical = 141108 ; free virtual = 166687
INFO: [HLS 200-10] Checking synthesizability ...
INFO: [XFORM 203-602] Inlining function 'nnet::cast<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2_mult>' into 'nnet::conv_2d_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' (firmware/nnet_utils/nnet_conv2d_resource.h:98->firmware/nnet_utils/nnet_conv2d.h:52) automatically.
INFO: [HLS 200-111] Finished Checking Synthesizability Time (s): cpu = 00:00:56 ; elapsed = 00:00:59 . Memory (MB): peak = 961.207 ; gain = 531.398 ; free physical = 141088 ; free virtual = 166667
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'ReuseLoop' (firmware/nnet_utils/nnet_conv2d_resource.h:53) in function 'nnet::conv_2d_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' for pipelining.
INFO: [XFORM 203-502] Unrolling small iteration loop 'Loop-2.1.1' (firmware/nnet_utils/nnet_array.h:38) in function 'nnet::transpose_3d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' automatically.
[... cut ...]
 void transpose_3d(data_T data[CONFIG_T::depth * CONFIG_T::height * CONFIG_T::width],
      ^~~~~~~~~~~~
firmware/nnet_utils/nnet_array.h:27:6: note:   template argument deduction/substitution failed:
firmware/myproject.cpp:33:52: note:   cannot convert ‘x’ (type ‘hls::stream<nnet::array<ap_fixed<16, 6>, 1> >’) to type ‘nnet::array<ap_fixed<16, 6>, 1>*’
     nnet::transpose_3d<input_t, layer3_t, config3>(x, layer3_out); // transpose_input_for_x
                                                    ^
g++: error: myproject.o: No such file or directory
```

vloncar, 2024-07-10:
In io_parallel, each element of the tensor is assigned a separate "port" and intermdiate tensors are partitioned into separate registers, so your model will need 16k ports to implement this -> unlikely to work, as you have seen.

For io_stream, the data is packed along the last dimension (in your case only 1) and streamed one "pack" at the time. To transpose large tensors you would need lots of intermediate storage, so we didn't implement it. And you pass the inputs in "channels first" order, so they need to be transposed to "channels last" as that's what the implementation of conv2d expects.

sei-jgwohlbier, 2024-07-10:
Ok, thanks for the explanation. That makes sense. I'm trying to incrementally get something to build, which explains why I would have "needed" 16k ports. I did make the conversion to `io_stream` and got the channels last sorted out. I'll keep trying to move forward.