fastmachinelearning/hls4ml #1048: FPGA Output is Zero with 8,512 parameters.

zsrabbani, 2024-08-08:
I have a CNN model. I used the hls4ml and all file and bitfile generated completely. Now I used the deployment code to implement on FPGA(ZCU104), the prediction output of FPGA is always Zero.
I even checked that I have weights.

I will appericate for helping me.

Here is the Model:
*******

Here is the HLSML code:
hls_config = hls4ml.utils.config_from_keras_model(model, granularity='name')
hls_config['Model']['ReuseFactor'] = 32
hls_config['Model']['Strategy'] = 'Resource'
hls_config['LayerName']['dense_2']['exp_table_t'] = 'ap_fixed<16,6>'
hls_config['LayerName']['dense_2']['inv_table_t'] = 'ap_fixed<16,6>'
hls_config['LayerName']['dense_2']['Strategy'] = 'Stable'

hls_model = hls4ml.converters.convert_from_keras_model(
    model, hls_config=hls_config, output_dir='CNN_16', backend='VivadoAccelerator', part='xczu7ev-ffvc1156-2-e') 

hls_model.compile()

Here s the **deployment** code:
[image]

GeorgeMentzos, 2024-08-08:
I can confirm that I am also encountering similar behaviour where I am using the standard CNN from the hls4ml tutorial quantized at 6-bits and the prediction output I am getting is also random. I would like to add that this only occurs when I am using the Resource Strategy where I am observing considerable accuracy loss (from 84% to 18%) just by switching from Latency to Resource mode.

vloncar, 2024-08-08:
Try to get a better understanding of how configuration of types works from the documentation, what are the effects of using fixed precision and quantization, and ultimately profile your application. See the [tutorial](https://fastmachinelearning.org/hls4ml-tutorial/), especially part 2 and 4.

zsrabbani, 2024-08-09:
As you can see, I used the correct setup but did not get any results.
Could you help me with it?

returnwellbeing, 2024-09-03:
Hi, I've encountered same issue. I am using the example for extension api, kreverse. https://fastmachinelearning.org/hls4ml/advanced/extension.html#
I used vivadoaccelerator and got the final hardware block. but when i deploy the hardware on pynq-z2 board, I got only zero-filled output.

nghielme, 2024-09-03:
Hi, I would suggest to first of all check if `hls_model.predict(x)` (HLS model simulated on CPU) corresponds to `model.predict(x)`(Keras model); they should be at least close each other. If they are not, the problem can be related to accumulators datatype in the network. For that you can try using `auto` so that the size of the accumulator is inferred by the operations that use the accumulator. This [jmitrevs:keras-config-auto](https://github.com/jmitrevs/hls4ml/tree/keras-config-auto) can be helpful in using `auto` for properly handle accumulators datatype.

returnwellbeing, 2024-09-06:
@nghielme Thanks for suggestion. I found that `hls_model.predict(x)` and `model.predict(x)` are different. your advice was a great help in finding the cause.

@zsrabbani In my case, there are som error when generating `{OUTPUT_DIR}/firmware/myproject.cpp`.
There must are some function calls at the end of the `myproject.cpp`. please check yours.

```c
void myproject(
    // Here are some inputs
) {

    // hls-fpga-machine-learning insert IO
    // Here are som pragmas

#ifndef __SYNTHESIS__
    static bool loaded_weights = false;
    if (!loaded_weights) {
        // hls-fpga-machine-learning insert load weights
        loaded_weights = true;
    }
#endif

    // ****************************************
    // NETWORK INSTANTIATION
    // ****************************************

    // hls-fpga-machine-learning insert layers
    // Some function calls should be here. if not, the outputs of hardware block are always ZERO
}
```

zsrabbani, 2024-09-06:
@returnwellbeing 
As I check myproject.cpp, everything looks fine and I didn't get that comment.

Irisaka, 2025-07-14:
> I can confirm that I am also encountering similar behaviour where I am using the standard CNN from the hls4ml tutorial quantized at 6-bits and the prediction output I am getting is also random. I [...]

Hi, I am facing the same problem. When I change the strategy from Latency to Resource, the prediction accuracy drops a lot, and profiling shows that converted layers results are far from the keras model. I am using hls4ml 1.1.0.

closed 2026-10-16 by github-actions[bot]