fastmachinelearning/hls4ml #1065: Wrong bias in rtl

osccircuit, 2024-09-17:
Hello! I used hls4ml to convert my simple "Celsius to Kelvin converter" model. My model ended up with one neuron where the bias is 275.149... After using hls4ml in the generated rtl verilog, the offset is 19608 in 16-bit representation with a precision of fixed<16.6>. When translating this number, we get 306.375. What could be the problem? In the b2.txt file, the value is still correct. So the problem occurs somewhere further

nghielme, 2024-09-17:
Hi, can you provide the output data `y_hls` you obtained by using `y_hls = hls_model.predict(X)`?

vloncar, 2024-09-17:
`ap_fixed<16,6>` is not wide enough to represent 275.149... Your interpretation of the number is wrong too. Try adding this anywhere in the code of the main function to see:
```c++
ap_fixed<16,6> temp = 275.149;
std::cout << "value: " << temp << std::endl; //or printf("value: %f\n", temp.to_double());
```
Read a bit more on how AP datatypes behave [here](https://docs.amd.com/r/en-US/ug1399-vitis-hls/Arbitrary-Precision-AP-Data-Types)

closed 2024-09-17 by vloncar