fastmachinelearning/hls4ml #1083: PyTorch: Unable to Compile Softmax or Sigmoid Layers Correctly

sei-rquartiano, 2024-10-16:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [X] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [X] Check that the issue hasn't already been reported, by checking the currently open issues.
- [X] If there are steps to reproduce the problem, make sure to write them down below.

# Quick summary
Unable to compile a PyTorch softmax or sigmoid layer. Attempted both module and functional calls; got different error messages described in next section

# Details
For the nn.Softmax() call I get a `KeyError: 'axis'` error which stops config generation. For nn.Sigmoid() and functional softmax or sigmoid calls it gets through compilation, however an `AssertionError` shows that the values are out of tolerance. More importantly than the tolerance, they appear to output the same incorrect values every time. 

### Steps to Reproduce
This is the code I've used to investigate errors with pytorch softmax/sigmoid calls. To switch between them comment/uncomment the appropriate lines in the model definition towards the top of the file. I also added a linear layer as an example of something that would normally precede a softmax. Doesn't seem to affect the error.

1. Clone hls4ml repository at current master branch
2. Run this test code
3. Change model definition to a different softmax/sigmoid call, repeat step 2

```
from pathlib import Path

import numpy as np
import pytest
import tensorflow as tf
from sklearn.metrics import accuracy_score
import torch.nn as nn
import torch
#from torchsummary import summary

import hls4ml
import os

from hls4ml.converters import convert_from_pytorch_model
from hls4ml.utils.config import config_from_pytorch_model

test_root_path = Path(__file__).parent

class SoftmaxModel(nn.Module):
    def __init__(self):
        super().__init__()
        #self.linear = nn.Linear(10,4)
        self.softmax = nn.Softmax(dim=-1)
        #self.softmax = nn.Sigmoid()
    def forward(self, x):
        #x = self.linear(x)
        return self.softmax(x)
        #return nn.functional.softmax(x,dim=-1)
        #return nn.functional.sigmoid(x)

if __name__ == "__main__":

    n_in = 2
    size_in = 10
    n_batch = 3

    model = SoftmaxModel()
    model.eval()
    print(model)

    X_input = np.random.rand(n_batch, n_in, size_in)
    with torch.no_grad():
        pytorch_prediction = model(torch.Tensor(X_input))

    config = config_from_pytorch_model(model,
                                       (None, n_in, size_in),
                                       channels_last_conversion='internal',
                                       transpose_outputs=False)
    config['Model']['Strategy'] = 'Resource'
    config['Model']['Precision'] = 'ap_fixed<32,12>'
[... 186 lines cut ...]
  File "/home/hls4ml-user/miniconda3/envs/hls4ml/lib/python3.10/site-packages/numpy/testing/_private/utils.py", line 862, in assert_array_compare
    raise AssertionError(msg)
AssertionError: 
Not equal to tolerance rtol=1e-05, atol=0.05

Mismatched elements: 19 / 20 (95%)
Max absolute difference: 1024.13953553
Max relative difference: 5.673692
 x: array([[0.063111, 0.081679, 0.126688, 0.107046, 0.073098, 0.123828,
        0.063607, 0.139536, 0.160859, 0.060548],
       [0.074005, 0.095292, 0.134147, 0.062772, 0.104638, 0.135005,
        0.074982, 0.124149, 0.077979, 0.117031]], dtype=float32)
 y: array([[ 1.855469e-02,  1.900000e+01, -1.024000e+03,  0.000000e+00,
         0.000000e+00,  1.855469e-02,  1.900000e+01, -1.024000e+03,
         0.000000e+00,  0.000000e+00],...
```
Note: the output values between nn.Sigmoid() and nn.functional.softmax/sigmoid are different, but they're the same for each function and are always outside of tolerance. 
## Optional

## Possible Fix
Isn't a fix as much as it's something I noticed. When I originally encountered the module call error I wondered whether it had to do with the specific keyword 'axis' since it's that in Keras but 'dim' in PyTorch. However when I looked in `converters/core.py: parse_activation_layer()` I notice two things
1. The axis vs dim keyword discrepancy seems to be handled in both module and functional calls, however since I'm getting a KeyError maybe its not functioning properly?
2. Softmax is missing from the `['layer_name']` if statements inside the function (it is in the global `activation_layers` list outside it), however Sigmoid is there and adding softmax to the list doesn't seem to do anything

Thanks for looking into this

returnwellbeing, 2024-10-21:
Hi, I encountered this issue before. I added two lines below https://github.com/fastmachinelearning/hls4ml/blob/main/hls4ml/converters/pytorch/core.py#L66-L67

```py
if hasattr(node, 'dim'):
            layer['axis'] = class_object.dim
if hasattr(class_object, 'dim'): # temperal solution
            layer['axis'] = class_object.dim
```

when  parsing the pytorch's softmax, there is no `dim` in `node`. I found `dim` is in `class_object`. I hope this helps you:)

JanFSchulte, 2024-10-22:
Hi! Thanks for reporting these issues. The softmax issues are indeed a bug that we overlooked because softmax activation was missing from our tests. This is fixed in https://github.com/fastmachinelearning/hls4ml/pull/1086. 

For the issues with numerical correctness in sigmoid and softmax, this seems to be related to the io_type setting. If I run your setup with io_parallel, everything seems to work. The issues with io_stream are not reproduced in our tests, so I'm not quite sure yet what's going wrong there.

sei-rquartiano, 2024-10-24:
Thank you both for the input! I will switch over to that PR branch until it's merged and change my 'io_type' to 'io_parallel.' If I manage to figure out what's going on with 'io_stream' on my end I'll be sure to let you know. Thanks again!

sei-rquartiano, 2024-11-12:
Getting correct results after merged PR and recommended changes. Thank you!

closed 2024-11-12 by sei-rquartiano